indigo-astronomy/indigo #7: build error on macOS

umi-neko, 2017-11-23:
Hi, I follow the instruction in README, installed autoconf, automake, and libtool, then run make all. It said cann't find libqhy.a
How can I fix it?
Here is the log.
> -------------------- Darwin build --------------------
> drivers: indigo_ccd_asi indigo_ccd_atik indigo_ccd_dsi indigo_ccd_fli indigo_ccd_iidc indigo_ccd_qhy indigo_ccd_qsi indigo_ccd_sbig indigo_ccd_simulator indigo_ccd_ssag indigo_ccd_sx indigo_wheel_asi indigo_wheel_atik indigo_wheel_fli indigo_wheel_sx indigo_focuser_fcusb indigo_focuser_fli indigo_focuser_usbv3 indigo_mount_lx200 indigo_mount_nexstar indigo_mount_simulator indigo_mount_temma indigo_gps_nmea indigo_gps_simulator indigo_dome_simulator indigo_guider_asi
> install -d build
> install -d build/bin
> install -d build/lib
> install -d build/drivers
> install -d build/include
> install -d build/share/indigo
> cp indigo_libs/indigo_config.h indigo_libs/indigo_config.h.orig
> sed 's/INDIGO_BUILD.*/INDIGO_BUILD 63/' indigo_libs/indigo_config.h.orig >indigo_libs/indigo_config.h
> cp INDIGO\ Server\ for\ macOS/Info.plist INDIGO\ Server\ for\ macOS/Info.plist.orig
> sed '/CFBundleVersion/ { n; s/>.*</>63</; }' INDIGO\ Server\ for\ macOS/Info.plist.orig >INDIGO\ Server\ for\ macOS/Info.plist
> mkdir libqhy_scratch; cd libqhy_scratch; cmake -DCMAKE_INSTALL_PREFIX=../build/ -DSKIP_FIRMWARE_INSTALL=True ../indigo_drivers/ccd_qhy/bin_externals/qhyccd/; make install; cd ..; rm -rf libqhy_scratch
> /bin/sh: cmake: command not found
> make[1]: *** No rule to make target `install'.  Stop.
> cp indigo_drivers/ccd_qhy/bin_externals/qhyccd/include/*.h build/include
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_agent.o indigo_libs/indigo_agent.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_base64.o indigo_libs/indigo_base64.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_bus.o indigo_libs/indigo_bus.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_ccd_driver.o indigo_libs/indigo_ccd_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_client.o indigo_libs/indigo_client.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_client_xml.o indigo_libs/indigo_client_xml.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_dome_azimuth.o indigo_libs/indigo_dome_azimuth.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_dome_driver.o indigo_libs/indigo_dome_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_driver.o indigo_libs/indigo_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_driver_json.o indigo_libs/indigo_driver_json.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_driver_xml.o indigo_libs/indigo_driver_xml.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_focuser_driver.o indigo_libs/indigo_focuser_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_gps_driver.o indigo_libs/indigo_gps_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_guider_driver.o indigo_libs/indigo_guider_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_io.o indigo_libs/indigo_io.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_json.o indigo_libs/indigo_json.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_mount_driver.o indigo_libs/indigo_mount_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_novas.o indigo_libs/indigo_novas.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_server_tcp.o indigo_libs/indigo_server_tcp.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_timer.o indigo_libs/indigo_timer.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_version.o indigo_libs/indigo_version.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_wheel_driver.o indigo_libs/indigo_wheel_driver.c
> clang -g -mmacosx-version-min=10.10 -fPIC -O3 -Iindigo_libs -Iindigo_drivers -Iindigo_mac_drivers -Ibuild/include -std=gnu11 -DINDIGO_MACOS   -c -o indigo_libs/indigo_xml.o indigo_libs/indigo_xml.c
> ar -rv build/lib/libindigo.a indigo_libs/indigo_agent.o indigo_libs/indigo_base64.o indigo_libs/indigo_bus.o indigo_libs/indigo_ccd_driver.o indigo_libs/indigo_client.o indigo_libs/indigo_client_xml.o indigo_libs/indigo_dome_azimuth.o indigo_libs/indigo_dome_driver.o indigo_libs/indigo_driver.o indigo_libs/indigo_driver_json.o indigo_libs/indigo_driver_xml.o indigo_libs/indigo_focuser_driver.o indigo_libs/indigo_gps_driver.o indigo_libs/indigo_guider_driver.o indigo_libs/indigo_io.o indigo_libs/indigo_json.o indigo_libs/indigo_mount_driver.o indigo_libs/indigo_novas.o indigo_libs/indigo_server_tcp.o indigo_libs/indigo_timer.o indigo_libs/indigo_version.o indigo_libs/indigo_wheel_driver.o indigo_libs/indigo_xml.o
> ar: creating archive build/lib/libindigo.a
> a - indigo_libs/indigo_agent.o
> a - indigo_libs/indigo_base64.o
> a - indigo_libs/indigo_bus.o
> a - indigo_libs/indigo_ccd_driver.o
> a - indigo_libs/indigo_client.o
> a - indigo_libs/indigo_client_xml.o
> a - indigo_libs/indigo_dome_azimuth.o
> a - indigo_libs/indigo_dome_driver.o
> a - indigo_libs/indigo_driver.o
> a - indigo_libs/indigo_driver_json.o
> a - indigo_libs/indigo_driver_xml.o
> a - indigo_libs/indigo_focuser_driver.o
> a - indigo_libs/indigo_gps_driver.o
> a - indigo_libs/indigo_guider_driver.o
> a - indigo_libs/indigo_io.o
> a - indigo_libs/indigo_json.o
> a - indigo_libs/indigo_mount_driver.o
> a - indigo_libs/indigo_novas.o
> a - indigo_libs/indigo_server_tcp.o
> a - indigo_libs/indigo_timer.o
> a - indigo_libs/indigo_version.o
> a - indigo_libs/indigo_wheel_driver.o
> a - indigo_libs/indigo_xml.o
> clang -shared -o build/lib/libindigo.dylib indigo_libs/indigo_agent.o indigo_libs/indigo_base64.o indigo_libs/indigo_bus.o indigo_libs/indigo_ccd_driver.o indigo_libs/indigo_client.o indigo_libs/indigo_client_xml.o indigo_libs/indigo_dome_azimuth.o indigo_libs/indigo_dome_driver.o indigo_libs/indigo_driver.o indigo_libs/indigo_driver_json.o indigo_libs/indigo_driver_xml.o indigo_libs/indigo_focuser_driver.o indigo_libs/indigo_gps_driver.o indigo_libs/indig [... cut ...] k IOKit -framework ImageCaptureCore -lobjc  -Lbuild/lib -lusb-1.0 -lindigo
> clang -shared -o build/drivers/indigo_ccd_qhy.dylib indigo_drivers/ccd_qhy/indigo_ccd_qhy.o build/lib/libqhy.a -headerpad_max_install_names -framework Cocoa -mmacosx-version-min=10.10 -framework CoreFoundation -framework IOKit -framework ImageCaptureCore -lobjc  -Lbuild/lib -lusb-1.0 -lstdc++ -lindigo
> clang: error: no such file or directory: 'build/lib/libqhy.a'
> make: *** [build/drivers/indigo_ccd_qhy.dylib] Error 1

rumengb, 2017-11-23:
Hi
it looks like you do not have cmake installed.

Rumen

closed 2017-11-23 by rumengb

umi-neko, 2017-11-23:
After install cmake everything is ok. Thank you very much.