indigo-astronomy/indigo #80: Unable to connect to ASI1600MM-Pro

joecodenoise, 2018-10-29:
I have a ZWO ASI 1600MM-Pro, as well as an ASI 120MC-S, and an ASI EFW. They are connected to a raspberry pi 3 b+ running the latest raspbian stretch lite. I downloaded and installed the latest binary package of indigo in .deb form and installed it as well. When I start indigo_server  via the commandline:

indigo_server -vvv indigo_ccd_asi indigo_wheel_asi, then run kstars on my laptop, I can successfully connect to the indigo server, and it reports all of the connected devices as available. I can successfully connect to the 120MC-S and the filter wheel. When I try to connect to the 1600MM-Pro, it fails, and I get an alert in the trace output.

Attached is a trace file that shows me connecting to the 120MC-S, attempting to connect to the 1600MM-Pro, and then connecting to the EFW. 
[indigo_err.log](https://github.com/indigo-astronomy/indigo/files/2523908/indigo_err.log)

Any further attempts to connect to the 1600MM-Pro result in the following trace output:

04:37:31.902834 indigo_server: 20 → <newSwitchVector
04:37:31.948956 indigo_server: 20 →   device='ZWO ASI1600MM Pro #1'
04:37:31.948956 indigo_server:   name='CONNECTION'>
04:37:31.948956 indigo_server:   <oneSwitch
04:37:31.948956 indigo_server:     name='CONNECT'>
04:37:31.948956 indigo_server:       On
04:37:31.948956 indigo_server:   </oneSwitch>
04:37:31.948956 indigo_server: </newSwitchVector>
04:37:31.949154 indigo_server: version: CONNECTION -> CONNECTION (current)
04:37:31.949214 indigo_server: version: CONNECTION.CONNECT -> CONNECTION.CONNECTED (current)
04:37:31.949272 indigo_server: INDIGO Bus: property change request
04:37:31.949322 indigo_server: 'ZWO ASI1600MM Pro #1'.'CONNECTION' SWITCH UNDEFINED Idle 0.0 UNDEFINED {
04:37:31.949366 indigo_server:   'CONNECTED' = On
04:37:31.949408 indigo_server: }
04:37:31.949463 indigo_server: indigo_ccd_asi[249]: indigo_try_global_lock(): failed to get lock.
04:37:31.949516 indigo_server: INDIGO Bus: property removal
04:37:31.949564 indigo_server: 'ZWO ASI1600MM Pro #1'.'CCD_INFO' NUMBER ro Ok 2.0  {
04:37:31.949608 indigo_server: }
04:37:31.949656 indigo_server: version: CCD_INFO -> CCD_INFO (legacy)
04:37:31.949706 indigo_server: 20 ← <delProperty device='ZWO ASI1600MM Pro #1' name='CCD_INFO'/>
04:37:31.949789 indigo_server: INDIGO Bus: property removal
04:37:31.949844 indigo_server: 'ZWO ASI1600MM Pro #1'.'CCD_UPLOAD_MODE' SWITCH rw Ok 2.0 OneOfMany {
04:37:31.949894 indigo_server: }

Any ideas on how to further troubleshoot this?

rumengb, 2018-10-29:
ASI recently updated their SDK (it is closed sourse and we have no control over it) we have it updated indigo too. Asi reported fixes regarding 1600. We will prepare a build for you with the new SDK. 

In your case ASIOpenCamera()  returns 5 which means Camera is disconnected. Which is obviously not true. I hope it is fixed with the new SDK. If the problem is still present we will report it to ASI

polakovic, 2018-10-29:
In this case it may be also powering issue. Are ASI120 and wheel connected to ASI1600 internal hub?

joecodenoise, 2018-10-29:
Ah, yes, the 120 and wheel are connected to the 1600’s hub. I will connect
the 1600 to a powered USB hub and connect the hub to the pi.

joecodenoise, 2018-10-29:
Unfortunately, even with a powered hub, I get the same results. Trace log shows no difference.

polakovic, 2018-10-29:
ASI1600 hub is probably not powered from USB but from 12V power source. It may make a difference if you will connect ASI120 and/or the wheel directly to RPi. Or if you will try to connect just ASI1600 first.

joecodenoise, 2018-10-29:
I tried disconnecting everything except the ASI1600. The 1600 has its 12V power connected, and is connected to my powered USB hub, but I get the same result. Perhaps I am encountering one of the bugs that ZWO referred to in their closed-source driver update. :(

polakovic, 2018-10-29:
Hm, yes, in this case it is not powering issue. Rumen pls. can you look at it?

polakovic, 2018-10-29:
Sure, every soul counts :) Just checkout the most recent version from git and build it. Prerequisites are mentioned in README.md.

joecodenoise, 2018-10-29:
Will do this afternoon, and I'll report back with findings.

rumengb, 2018-10-29:
> Hm, yes, in this case it is not powering issue. Rumen pls. can you look at it?

I really hope this is the problem. if not I will write to ASI...

joecodenoise, 2018-10-30:
Unfortunately, even after a new build using the latest drivers from the ASI sdk, I get the same results.

rumengb, 2018-10-30:
I will let ASI know!

joecodenoise, 2018-11-01:
For what its worth, I encountered the same behavior with INDIlib as well. I posted the results from that test in ZWO's driver forum.

tstibor, 2018-11-01:
Have you tried the previous ASI library version. 
The following commit allows you to checkout the former version 1.13.0821
[Revert "Update ZWO SDK to libASICamera2 v.1.13.0930" ](https://github.com/indigo-astronomy/indigo/commit/1907b74125ac3773135284deeb0455eba63bbe55)

joecodenoise, 2018-11-01:
I'll try it out and report back.

joecodenoise, 2018-11-01:
Ah yes that's right, and I get the same result as expected.

rumengb, 2018-11-02:
Just got mail from asi. Yang is taking care of the issue. It does not happen on intel he is now testing on Rpi

rumengb, 2018-11-06:
Here is the answer from ASI, It seems to be a powering issue after all, I am closing the issue and if it does not solve it please reopen it :)

Hi Rumen,
Today I was managed to test on raspberry pi, and works normally, I think he can try replacing his power supply of raspberry pi, I have encountered similar issue too,
and it can be resolve by using a more stable 5v/3A power supply.

Best Wishes
Yang Zhou

closed 2018-11-06 by rumengb

joecodenoise, 2018-11-08:
I'm still having the issue, but working with ZWO directly, Leaving this closed, because at this point, I think it is clearly not an Indigo issue.

rumengb, 2018-11-08:
Thank you! If there is an update on the situation please let us know!