indigo-astronomy/indigo #103: Segfault when starting indigo_ccd_gphoto2

japz, 2018-12-17:
I'm trying to hook up my Sony A7r iii to indigo-server, but the driver indigo_ccd_gphoto2 segfaults:

```
root@raspberrypi:~# indigo_server -v -- indigo_ccd_gphoto2
16:19:33.449115 indigo_server: INDIGO server 2.0-76 built on Sun Dec  9 18:41:22 2018
16:19:33.454483 indigo_server: Driver indigo_ccd_gphoto2 2.0.0.3 loaded
16:19:33.454638 indigo_server: Resource /ctrl (4338, text/html) added
16:19:33.454706 indigo_server: Resource /resource/angular.min.js (54437, text/javascript) added
16:19:33.454760 indigo_server: Resource /resource/bootstrap.min.js (9843, text/javascript) added
16:19:33.454811 indigo_server: Resource /resource/bootstrap.css (21056, text/css) added
16:19:33.454860 indigo_server: Resource /resource/jquery.min.js (29969, text/javascript) added
16:19:33.454910 indigo_server: Resource /fonts/glyphicons-halflings-regular.ttf (23509, application/x-font-ttf) added
16:19:33.454960 indigo_server: Resource /resource/logo.png (20757, image/png) added
16:19:33.623769 indigo_server: indigo_ccd_gphoto2: auto-detected device 'Sony Alpha-A7R III (Control) 054c:0c33 (bus 001, device 008)'
16:19:33.723850 indigo_server: indigo_ccd_gphoto2: attach device 'Sony Alpha-A7R III (Control) 054c:0c33 (bus 001, device 008)' in slot '0'
16:19:33.942605 indigo_server: XML Parser: parser finished
16:19:33.942704 indigo_server: Server attached
16:19:33.942825 indigo_server: Server started on 7624
16:19:33.943226 indigo_server: indigo_ccd_gphoto2: mirror lockup not available for camera 'SONY ALPHA-A7R III (CONTROL)'
Segmentation fault
```

At first I tried with the raspbian provided libgphoto2 and the indigo-server prebuilt package. I also tried with latest libgphoto2 stable. Capturing an image using the gphoto2 command line tool works.

I also tried on an amd64 machine to see if the error is exclusive to the arm build. Same problem.

tstibor, 2018-12-17:
Can you run the following tests and post the result, so one can figure out which libgphoto2 properties your camera possesses:

`>gphoto2 --list-all-config`

Moreover, can you run indigo_server and gphoto2 drivers via gdb as follows, so one can trace problem:

```
>gdb --args ./build/bin/indigo_server --do-not-fork -vvv ./build/drivers/indigo_ccd_gphoto2
GNU gdb (Debian 8.2-1) 8.2
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/bin/indigo_server...done.
(gdb) run
Starting program: /home/tstibor/dev/astronomy/indigo/build/bin/indigo_server --do-not-fork -vvv ./build/drivers/indigo_ccd_gphoto2
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff70d1700 (LWP 32399)]
[New Thread 0x7ffff68d0700 (LWP 32400)]
20:10:23.186971 indigo_server: INDIGO server 2.0-77 built on Mon Dec 17 19:51:42 2018
20:10:23.223838 indigo_server: Driver indigo_ccd_gphoto2 2.0.0.3 loaded
20:10:23.223869 indigo_server: Resource /ctrl (4338, text/html) added
20:10:23.223878 indigo_server: Resource /resource/angular.min.js (54437, text/javascript) added
20:10:23.223885 indigo_server: Resource /resource/bootstrap.min.js (9843, text/javascript) added
20:10:23.223893 indigo_server: Resource /resource/bootstrap.css (21056, text/css) added
20:10:23.223904 indigo_server: Resource /resource/jquery.min.js (29969, text/javascript) added
20:10:23.223917 indigo_server: Resource /fonts/glyphicons-halflings-regular.ttf (23509, application/x-font-ttf) added
20:10:23.223930 indigo_server: Resource /resource/logo.png (20757, image/png) added
20:10:23.224171 indigo_server: 9 → <newSwitchVector device='Server' name='DRIVERS'>
20:10:23.224171 indigo_server: <oneSwitch name='Gphoto2 Camera'>On</oneSwitch>
20:10:23.224171 indigo_server: </newSwitchVector>
20:10:23.224218 indigo_server: INDIGO Bus: property change request
20:10:23.224229 indigo_server: 'Server'.'DRIVERS' SWITCH UNDEFINED Idle 0.0 UNDEFINED {
20:10:23.224236 indigo_server:   'Gphoto2 Camera' = On
20:10:23.224244 indigo_server: }
20:10:26.828717 indigo_server: indigo_ccd_gphoto2: auto-detected device 'Huawei P9 Plus 12d1:107e (bus 003, device 008)'
20:10:40.395273 indigo_server: indigo_ccd_gphoto2[746, ctx_error_func, 0x7ffff70d2800]: PTP General Error
20:10:40.423158 indigo_server: indigo_ccd_gphoto2[1933, device_connect, 0x7ffff70d2800]: [rc:-1] gp_camera_init failed
20:10:40.423224 indigo_server: indigo_ccd_gphoto2[2028, device_attach, 0x7ffff70d2800]: connecting failed slot '0' device 'Huawei P9 Plus 12d1:107e (bus 003, device 008)'
20:10:40.722755 indigo_server: indigo_ccd_gphoto2[2049, device_detach, 0x7ffff68d0700]: no slot found for device 12d1:107e (bus 003, device 008) to be detached
20:10:41.974822 indigo_server: indigo_ccd_gphoto2: auto-detected device 'Huawei P9 Plus 12d1:107e (bus 003, device 009)'
[New Thread 0x7ffff5001700 (LWP 32562)]
20:10:43.454055 indigo_server: indigo_ccd_gphoto2: attach device 'Huawei P9 Plus 12d1:107e (bus 003, device 009)' in slot '0'
20:10:43.497769 indigo_server: indigo_ccd_gphoto2: mirror lockup not available for camera 'HUAWEI P9 PLUS'

Thread 4 "indigo_server" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff5001700 (LWP 32562)]
0x00007ffff5ff83cb in ccd_attach (device=0x7ffff066e6c0) at indigo_ccd_gphoto2.c:1516
1516			for (int i = 0; i < DSLR_SHUTTER_PROPERTY->count; i++) {
```
I connected my Smartphone which is recognized by the driver, however, it does not possess
the shutterspeed property which is however, required for setting the shutterspeed. So I believe this causes the problem of your camera. At current the following shutterspeed property names are supported:

#define NIKON_SHUTTERSPEED			"shutterspeed"
#define EOS_SHUTTERSPEED			NIKON_SHUTTERSPEED

If your camera have a different shutterspeed property name, one could add them and hopefully
fix the problem.

japz, 2018-12-17:
Thanks for the quick response!

I attached the output of gphoto2 (bit too long to paste here).
[gphoto2.txt](https://github.com/indigo-astronomy/indigo/files/2687733/gphoto2.txt)

I rebuilt with the patch, but I don't see the assertion. It looks like DSLR_COMPRESSION_PROPERTY is the culprit:

```
root@raspberrypi:~/indigo# gdb --args ./build/bin/indigo_server --do-not-fork -vvv ./build/drivers/indigo_ccd_gphoto2
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/bin/indigo_server...done.
(gdb) run
Starting program: /root/indigo/build/bin/indigo_server --do-not-fork -vvv ./build/drivers/indigo_ccd_gphoto2
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x767fe360 (LWP 2256)]
[New Thread 0x75ffe360 (LWP 2257)]
20:25:57.762232 indigo_server: INDIGO server 2.0-77 built on Mon Dec 17 12:31:26 2018
20:25:57.922055 indigo_server: Driver indigo_ccd_gphoto2 2.0.0.3 loaded
20:25:57.922156 indigo_server: Resource /ctrl (4338, text/html) added
20:25:57.922191 indigo_server: Resource /resource/angular.min.js (54437, text/javascript) added
20:25:57.922215 indigo_server: Resource /resource/bootstrap.min.js (9843, text/javascript) added
20:25:57.922238 indigo_server: Resource /resource/bootstrap.css (21056, text/css) added
20:25:57.922260 indigo_server: Resource /resource/jquery.min.js (29969, text/javascript) added
20:25:57.922285 indigo_server: Resource /fonts/glyphicons-halflings-regular.ttf (23509, application/x-font-ttf) added
20:25:57.922310 indigo_server: Resource /resource/logo.png (20757, image/png) added
20:25:57.922986 indigo_server: 9 → <newSwitchVector device='Server' name='DRIVERS'>
20:25:57.922986 indigo_server: <oneSwitch name='Gphoto2 Camera'>On</oneSwitch>
20:25:57.922986 indigo_server: </newSwitchVector>
20:25:57.923115 indigo_server: INDIGO Bus: property change request
20:25:57.923147 indigo_server: 'Server'.'DRIVERS' SWITCH UNDEFINED Idle 0.0 UNDEFINED {
20:25:57.923171 indigo_server:   'Gphoto2 Camera' = On
20:25:57.923191 indigo_server: }
20:26:02.203865 indigo_server: indigo_ccd_gphoto2: auto-detected device 'Sony Alpha-A7R III (Control) 054c:0c33 (bus 001, device 007)'
[New Thread 0x7531a360 (LWP 2258)]
20:26:04.445209 indigo_server: indigo_ccd_gphoto2: attach device 'Sony Alpha-A7R III (Control) 054c:0c33 (bus 001, device 007)' in slot '0'
20:26:04.654932 indigo_server: indigo_ccd_gphoto2: mirror lockup not available for camera 'SONY ALPHA-A7R III (CONTROL)'

Thread 4 "indigo_server" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7531a360 (LWP 2258)]
0x75727ac8 in ccd_attach (device=0x9ea7c8) at indigo_ccd_gphoto2.c:1548
1548			for (int i = 0; i < DSLR_COMPRESSION_PROPERTY->count; i++) {
```

I'd be happy to have a go at this. I don't know C that well, but it looks like a just few properties need to be added for the "Sony" vendor case, and then make it recognise it's dealing with a Sony camera?

tstibor, 2018-12-17:
You are right, it is the compression property (image format). The fallback for non-EOS/Nikon
is EOS camera, which is: "imageformat". However, Nikon and Sony have name: "imagequality".
So next time it is better to fallback on Nikon cameras :-). I just send a PR which still needs to be merged. You can however, cherry pick the PR and try it now.

japz, 2018-12-18:
I'm now getting stuck on capturing an image, as it again segfaults.

GDB output: 
[indigo_server_gdb.txt](https://github.com/indigo-astronomy/indigo/files/2690838/indigo_server_gdb.txt)

It turns out the `DSLR_SHUTTER_PROPERTY` has length 0, and can't be enumerated.

Looking at gphoto2 seems to confirm this.

root@raspberrypi:~/indigo# gphoto2 --get-config /main/capturesettings/shutterspeed
Label: Shutter Speed
Readonly: 0
Type: RADIO
Current: 130/10
END

This PR will fix the crash, but not make the camera work: https://github.com/indigo-astronomy/indigo/pull/106

The question of course is how to get this camera to work. :) This should be fixed in libgphoto2, and there is an open bug for this: https://github.com/gphoto/libgphoto2/issues/276 and other projects are currently working around it. I looked at the libgphoto2 ptp2 camera library, but it is too complex for me to understand.

Would it be possible to handle this case in the driver for now, and hardcode the possible shutter speeds for Sony (these are all possible values `1/8000,1/6400,1/5000,1/4000,1/3200,1/2500,1/2000,1/1600,1/1250,1/1000,1/800,1/640,1/500,1/400,1/320,1/250,1/200,1/160,1/125,1/100,1/80,1/60,1/50,1/40,1/30,1/25,1/25,1/15,1/13,1/10,1/8,1/6,1/6,1/4,1/3,4/10,5/10,6/10,8/10,10/10,13/10,16/10,20/10,25/10,32/10,40/10,40/10,60/10,80/10,100/10,130/10,150/10,200/10,250/10,300/10`? And if so, what would be the best place in the code to do so?

tstibor, 2018-12-18:
The problem is indeed in the libgphoto2 library which does not report the "choices" of the shutter property. As a consequence ->count=0 and strange behavior starts. I will make the driver more robust against this strange behavior in the future. As a quick hack, one could indeed fill that "by hand"

```
//int count = enumerate_widget(EOS_SHUTTERSPEED, device, NULL);
// count must be the number of elements in list "all possible values"
		DSLR_SHUTTER_PROPERTY = indigo_init_switch_property(NULL,
								    device->name,
								    DSLR_SHUTTER_PROPERTY_NAME,
								    GPHOTO2_NAME_DSLR,
								    GPHOTO2_NAME_SHUTTER,
								    INDIGO_OK_STATE,
								    INDIGO_RW_PERM,
								    INDIGO_ONE_OF_MANY_RULE,
								    count);
```
and then adding the values one by one here:
```
	indigo_init_switch_item(DSLR_SHUTTER_PROPERTY + i,
					widget_choice,
					"1/8000",
					parse_shutterspeed("1/8000"))
		i++;
```
If I find some time, I could make such a failover hack, however, it would be much better to solve that
in libghoto2.

japz, 2018-12-18:
The hint helps, I will give it a go. 👍

tstibor, 2018-12-18:
I just created a [PR](https://github.com/indigo-astronomy/indigo/pull/107) where the shutterspeed is set as a fixed string:

1/8000,1/6400,1/5000,1/4000,1/3200,1/2500,1/2000,1/1600,1/1250,1/1000,1/800,1/640,1/500,1/400,1/320,1/250,1/200,1/160,1/125,1/100,1/80,1/60,1/50,1/40,1/30,1/25,1/25,1/15,1/13,1/10,1/8,1/6,1/6,1/4,1/3,4/10,5/10,6/10,8/10,10/10,13/10,16/10,20/10,25/10,32/10,40/10,40/10,60/10,80/10,100/10,130/10,150/10,200/10,250/10,300/10

when camera Sony Alpha-A7R III is detected. Let me know whether is works.

japz, 2018-12-18:
Thanks! I will try it if I can un-break my setup.

I spent some time trying to figure this out myself as well (admittedly most of the time was spent figuring out the intricacies of C), getting closer to a solution, but then the connection to the camera from indigo-server stopped working altogether.

In my trial-and-error I crashed indigo-server numerous times, so perhaps there is some lingering data on connections that were never properly closed, but I couldn't find anything and a reboot also didn't help.
It still works with the gphoto2 CLI, but indigo-server no longer sees it nor handles hotplug events. Do you have any idea what might be the problem?

For reference, this was the direction I was working in (I got weird malloc issues with it if I ran the full list of shutterspeeds though - any chance that might have broken something?): https://github.com/japz/indigo/commit/7dc8dbb631dad0d78777d5054162dbe52fe17fa6

If all else fails I can revert to a fresh RPi image. Until I fix this I don't think I can test your patch.

polakovic, 2020-08-10:
gphoto2 driver is replaced by crossplatform indigo_ccd_ptp driver.

closed 2020-08-10 by polakovic