indigo-astronomy/indigo #114: QSI 500 driver image assembly error

Chandler, 2018-12-23:
When using Astroimager on macOS 10.13.6 with a QSI 583 camera, the image is not correctly reassembled. It is displayed with the left half on the right and vice versa. 

The driver in use is INDIGO QSI 500 2.0.0.2

[image]

Chandler, 2018-12-23:
I'm not sure how to update Astroimager to get the lastest driver or an easy way to test this out using the INDIGO driver directly.

polakovic, 2018-12-23:
I'll look at it. Does it happen only with the preview or also with the capture?

This is the most recent AstroImager: 
http://download.cloudmakers.eu/AstroImager_3.9_420.dmg

And you can also build and run INDIGO from source and run indigo_server with indigo_ccd_qsi driver, AstroImage will find it.

Chandler, 2018-12-23:
It also happens with the capture, I tried the updated Astroimager and am seeing the same issue. I will try playing with the INDIGO source build (I see a few changes were made in the last month or so to the QSI driver.

https://github.com/indigo-astronomy/indigo/commit/945b9de9647b78c51435beff5c6f81f61cac70bc

polakovic, 2018-12-23:
The only important change is line 194. If you can build it from source, pls. can you try to change first true to false?

polakovic, 2018-12-23:
Sorry, this is not the issue, keep it. The problem is, that PRIVATE_DATA->buffer is pointer to short, while FITS_HEADER_SIZE is in bytes. I'll fix it and build new AstroImager in a few minutes...

Chandler, 2018-12-23:
thanks for the quick response!

polakovic, 2018-12-23:
Pls. can you try this one?

http://download.cloudmakers.eu/AstroImager_3.9_421.dmg

Chandler, 2018-12-23:
It works!! thank you!

[image]

polakovic, 2018-12-23:
Great! Thank you very much, you helped us a lot.

closed 2018-12-23 by polakovic

polakovic, 2018-12-23:
P.S. I see you used AstroImager from AppStore. In this case you need to "register" it first (in AstroImager > Register) to create permanent license allowing you to run copies downloaded directly from our web.