indigo-astronomy/indigo #122: Mount_synscan joystick control not properly function

JohanBakk, 2018-12-29:
When mount_synscan is controlled from an aux_joystick driver, a small stick input leads to mount motion over a long period of time, which is processed in slow motion, one small step at a time.

jurgenfd, 2021-01-12:
What kind of physical device did you use @JohanBakk ? I have an Xbox One controller that could be used for that with my "Skywatcher EQ6R Pro"? 

[image]
[image]

When I attach and setup the AstroTelescope4 simply hangs. As per log:
[log2.txt](https://github.com/indigo-astronomy/indigo/files/5804945/log2.txt)

A second try reproduces my problem making it impossible to reproduce yours unfortunately.

JohanBakk, 2021-01-12:
I'm using a Logitech F710 (Cordless Rumblepad). I think this bug was fixed a long time ago; it is working fine now.

jurgenfd, 2021-01-12:
Can you double check yours is working 'cause mine isn't. Maybe you have another operator tip?... ;-)

JohanBakk, 2021-01-12:
Used mine yesterday; works fine :-) Did you set-up the joy-stick in Indigo Control panel? Connect and select the joystick in AstroTelescope (AT) (as you did). Next goto Control panel (right most button in the tool bar in AT). Select the buttons and axis to be used and (important!) save the config (Joystick -> "main" section -> Save Config). If you can't get it to work the first round, it sometimes helps to just save the config (in any state). Disconnect, reconnect the device, and the config file is loaded. Next configure and save again. Not sure if the particular Xbox controller is supposed to work though, but it could... :-)

jurgenfd, 2021-01-12:
Thanks! I can get the controller to unpark but that's it. As soon as the right stick is touched the AT4 crashes consistently.
Crash log attached. 
[log4.txt](https://github.com/indigo-astronomy/indigo/files/5805261/log4.txt)

Still this is a low priority but just a cool gadget. My Synscan hand controller is still working (although it needs USB powering; silly).
Thanks JohanBakk! Nighty night.

polakovic, 2026-10-02:
Confirmed as working by the original reporter. Closing.

closed 2026-10-02 by polakovic