indigo-astronomy/indigo #127: Alignment point precision - using single point alignment with mount_synscan

JohanBakk, 2018-12-30:
When using AT and mount_synscan, i noticed that the RA/DEC values of alignment points are two decimals, where the raw values they are derived from are 4 decimals. Using 4 decimals for alignment points as well would lead to better alignment/pointing precision.

polakovic, 2019-01-01:
Alignment points coordinates are actually "double" numbers. What you see is just "label" of the property item, but it will be better to show it in full sexagesimal format instead.

JohanBakk, 2019-01-02:
Maybe it is better to show the label in h:mm:ss format then, so it is easily compared to all other coordinates. The same goes for the RAW Coordinates property.

polakovic, 2019-01-03:
It will be even more complicated, e.g. side of pier is missing.

JohanBakk, 2019-01-03:
David added side of pier yesterday. Tested it, works fine.

polakovic, 2019-01-03:
Yes, I know. It is also added to the alignment point structure, so it needs to be added to the label.

polakovic, 2019-01-04:
changed to format "h:mm:ss h:mm:ss E/W".

closed 2019-01-04 by polakovic