indigo-astronomy/indigo #129: Mount_synscan; deleting alignment points during active slew

JohanBakk, 2018-12-30:
If during an active slew alignment points are deleted, the mount end position is calculated wrong in the second phase of the slew, causing an erratic end position. Either should the driver calculate the proper position in the second phase of the slew (without alignment removed) or deletion of alignment points during a slew should be prevented with some mutex.

jurgenfd, 2021-01-12:
@JohanBakk what are alignment points for a concept? I'll try to reproduce the bug if still relevant to you.

JohanBakk, 2021-01-12:
When the mount agent is synced (manually with the sync button in AstroTelescope, or using the plate solve panel in AstroImager), an "alignment point" is stored with the mount agent, detailing the position change. These alignment points can be selected and deleted with Indigo Control panel or Indigo Dashboard. The bug is not really serious; when you deleted alignment points during a slew, the coordination system of the mount get a bit upset (which is understandable :)

jurgenfd, 2021-01-12:
Yeah, too advanced for me currently. Maybe later.

polakovic, 2021-01-13:
Actually, this can be reproduced only on more advanced drivers with Alignment > Alignment mode property, e.g. SynScan.

earthonlinegame, 2026-02-04:
I was wondering why we delete alignment points during a slew.

polakovic, 2026-10-02:
Fixed in 952ce51eb. Alignment points can no longer be deleted while the mount is slewing: MOUNT_ALIGNMENT_DELETE_POINTS, MOUNT_ALIGNMENT_RESET, MOUNT_ALIGNMENT_SELECT_POINTS and MOUNT_ALIGNMENT_MODE are now refused with an alert while MOUNT_EQUATORIAL_COORDINATES is busy, because each of them changes the model the running slew uses to compute its end position. The same request is accepted again as soon as the slew has finished. The fix is in the common mount code, so it applies to every driver that uses the INDIGO alignment model, including SynScan. It is covered by a regression test on the mount simulator; it has not been re-tested on a physical SynScan mount, so please reopen if you still see it there.

closed 2026-10-02 by polakovic