indigo-astronomy/indigo #144: Abort when mount is either parked or tracking - upsets state machine

JohanBakk, 2019-01-07:
When the mount is either parked or tracking a target, and abort is pressed (repeatedly), there is a message “Aborted” and the abort state machine runs into trouble. The next park stalls after “Tracking stopped” or the next slew is aborted, without abort being pressed. It would be better to make abort conditional on whether there is something to be aborted or else log a message “Nothing to abort”.

ozbandit, 2019-01-19:
Code has been changed to implement the state machine slightly differently such that only a single flag is used to abort ongoing operations, whereas previously there were two flags. These changes are in the latest codebase and I believe this issue is now fixed.

JohanBakk, 2019-01-19:
Aborting a slew or a park works fine after the code change. This issue is from after those code changes.

ozbandit, 2019-01-27:
Pushed new code to hopefully fix this

JohanBakk, 2019-01-27:
Justed tested it; fixed! Can be closed.

closed 2019-01-27 by polakovic