indigo-astronomy/indigo #151: After 1 month & today indigo update got a kernel crash

agnunez, 2019-01-17:
I was running indigo in a NanoPiM4 for 1 month (24x7) in a ZWO skycam getting images every min. Today I receive and update indigo and several hours later got the following kernel error. (perhaps not related!).

Jan 17 19:25:04 localhost kernel: [53594.307674] ------------[ cut here ]------------
Jan 17 19:25:04 localhost kernel: [53594.307700] kernel BUG at mm/slub.c:3744!
Jan 17 19:25:04 localhost kernel: [53594.308072] Internal error: Oops - BUG: 0 [#1] SMP
Jan 17 19:25:04 localhost kernel: [53594.308499] Modules linked in: overlay af_packet rk_vcodec zram lz4_compress ip_tables x_tables autofs4 midgard_kbase
Jan 17 19:25:04 localhost kernel: [53594.309517] CPU: 2 PID: 8366 Comm: indigo_worker Not tainted 4.4.167-rk3399 #9
Jan 17 19:25:04 localhost kernel: [53594.310154] Hardware name: FriendlyElec NanoPi M4 (DT)
Jan 17 19:25:04 localhost kernel: [53594.310613] task: ffffffc06d990000 task.stack: ffffffc077478000
Jan 17 19:25:04 localhost kernel: [53594.311149] PC is at kfree+0xe8/0x1c0
Jan 17 19:25:04 localhost kernel: [53594.311486] LR is at free_async+0xb4/0x128

.....

Jan 17 19:25:04 localhost kernel: [53594.450369] Code: 7212003f 54000101 f9401001 370000c1 (d4210000)
Jan 17 19:25:04 localhost kernel: [53594.451079] ---[ end trace 66ad260a6a708d76 ]---
Jan 17 19:25:24 localhost indigo.sh[1842]: 19:25:24.138607 indigo_server: indigo_ccd_asi[361, asi_read_pixels, 0x7f956fa130]: Exposure failed: dev_id = 0 exposure status = 1
Jan 17 19:25:24 localhost rc.local[1844]: Jan 17 19:25:24 exposure failed
Jan 17 19:25:24 localhost systemd[1]: Stopping indigo.service...
Jan 17 19:25:24 localhost indigo.sh[1842]: 19:25:24.171342 indigo_server: Shutdown initiated (signal 15)...
Jan 17 19:25:24 localhost indigo.sh[1842]: 19:25:24.171631 indigo_server: Shutdown initiated (signal 2)...
Jan 17 19:25:24 localhost indigo.sh[1842]: 19:25:24.171906 indigo_server: Server detached
Jan 17 19:25:24 localhost indigo.sh[1842]: 19:25:24.172075 indigo_server: indigo_ccd_asi: 'ZWO ASI294MC Pro #0' detached
Jan 17 19:25:24 localhost systemd[1]: Stopped indigo.service.
Jan 17 19:25:24 localhost kernel: [53614.361528] usb 4-1.4: USB disconnect, device number 3

Anybody can confirm or diagnose this?
thanks!

tstibor, 2019-01-17:
Which kernel version are you running on your NanoPiM4? (`uname -a`)

polakovic, 2019-01-17:
I don't think that kernel bug is related, but "Exposure failed: dev_id = 0 exposure status = 1" usually means, that USB bandwidth is too high. The value can be set in Advanced property (in Advanced group), try to reduce the value a little bit.

agnunez, 2019-01-18:
$ uname -a
Linux skycam 4.4.167-rk3399 #9 SMP Sat Jan 12 18:01:55 CET 2019 aarch64 aarch64 aarch64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.1 LTS
Release:        18.04
Codename:       bionic
$ systemctl restart indigo.service
Jan 18 09:10:58 localhost indigo.sh[6930]: 09:10:58.344572 indigo_server: INDIGO server 2.0-77 built on Mon Jan 14 20:35:57 2019
Jan 18 09:10:58 localhost indigo.sh[6930]: 09:10:58.347925 indigo_server: Driver indigo_ccd_asi 2.0.0.11 loaded
Jan 18 09:10:58 localhost indigo.sh[6930]: 09:10:58.348029 indigo_server: indigo_ccd_asi: ASI SDK v. 1, 13, 1106

USB bandwidth was an issue until using a very short (20cm) usb3.0 good quality cable. Has been working with this cable and same bandwidth during weeks. I said, might be not related, but new indigo update was just a coincidence, and NanoPiM4 kernel is always my firts choice as cause. Thanks

polakovic, 2019-01-18:
ASI SDK was updated too, so bandwidth may be the issue again, hard to say :(

closed 2020-09-03 by rumengb