indigo-astronomy/indigo #164: indigo_wheel_asi and indigo_ccd_asi error on armv6l build

asantam, 2019-02-13:
The **indigo_wheel_as**i and **indigo_ccd_asi** drivers give an error on the **armv6** built. While the make process works fine and the drivers are compiled, when I try to load them in indigo_server the server just keeps restarting. Moreover, when I run the drivers by themselves (e.g. ./indigo_asi_wheel), they exit with the **"Illegal instruction"** error, something that doesn't happen with other drivers (even If they are not designed to be run by themselves).

Here is the output of the strace: [https://pastebin.com/UEmAQw1M](https://pastebin.com/UEmAQw1M)

Note that this doesn't happen with indigo_focuser_asi which can be loaded fine with indigo_server and doesn't give an error when run by itself. Moreover, both indigo_wheel_asi and indigo_ccd_asi work ok on the x86 build.

polakovic, 2019-02-13:
I suspect we use arm7 binaries. Rumen, any idea?

asantam, 2019-02-13:
You are right, I didn't thought to check.

[image].

Good place to start, will try to fix it and submit a PR.

rumengb, 2019-02-13:
Yes rhis is what we use. We used to use arm6 but you changed it :)

asantam, 2019-02-13:
Hi Rumen, is there any benefit on using the arm7 ones? Maybe the simplest option is just to go back to the arm6 ones.

asantam, 2019-02-14:
I replaced libASICamera2.a and libEFWFilter.a with their arm6 version (taken from the ZWO SDK) and now it works perfectly. 

Let me know if it makes sense to you to go back to the arm6 drivers (as arm7 is backwards compatible) so I can make a simple PR or if you prefer to leave a specific driver for each architecture version.  The rest of the arm7 drivers are:

ccd_sbig/bin_externals/sbigudrv/lib/Linux/arm/libsbigudrv.a
guider_asi/bin_externals/libusb2st4conv/lib/Linux/arm/libUSB2ST4Conv.a
ccd_mi/bin_externals/libgxccd/lib/Linux/arm/libgxccd.a

I think all of them have a arm6 version available (at least in the libindi repository).

polakovic, 2019-02-14:
To be honest, I don't remember why we used arm7 instead of arm6 binaries in some cases, maybe it is my fault :) So pls. make PR if you can test it.

asantam, 2019-02-14:
Ok, I managed to get all 3 arm6 asi drivers working (wheel, ccd, and guider). However, neither ccd_sbig nor ccd_mi work when I replaced the .a files from the ones in indilib. Because of this I'll limited PR #165 to an asi drivers fix.

PR #165 (2019-02-14): Changed ccd, guider, and wheel asi drivers to arm6
The static library files for ccd_asi, guider_asi, and wheel_asi
(libASICamera2.a, libUSB2ST4Conv.a, and libEFWFilter.a respectively)
were generated for arm7 architecture. This made the compiled drivers
unusable on arm6 devices (i.e. Raspberry Pi 0/1). Since the arm7
architecture is mostly backwards compatible, the static libraries were
changed to their arm6 version. The libASICamera2.a and libEFWFilter.a
files were obtained from the ZWO [SDK](https://astronomy-imaging-camera.com/software-drivers), and the libUSB2ST4Conv.a was
obtained from the [... cut]
commit 3e37cec4 (2019-02-14): Changed ccd, guider, and wheel asi drivers to arm6

The static library files for ccd_asi, guider_asi, and wheel_asi
(libASICamera2.a, libUSB2ST4Conv.a, and libEFWFilter.a respectively)
were generated for arm7 architecture. This made the compiled drivers
unusable on arm6 devices (i.e. Raspberry Pi 0/1). Since the arm7
architecture is mostly backwards compatible, the static libraries were
changed to their arm6 version. The libASICamera2.a and libEFWFilter.a
files were obtained from the ZWO SDK, and the libUSB2ST4Conv.a was
obtained from the libindi repository.

Resolves: #164
closed 2019-02-14 by rumengb