indigo-astronomy/indigo #168: Windows Client tweaks needed

MasterYoddha, 2019-02-19:
Hello,

I'm trying to compile the windows client. The project file needs to polished just a bit:

In additional include to use (rather than the full path)
.\externals\pthreads4w\include

And in additional library
.\externals\pthreads4w\lib

Fixing the settings moved to compilation. However there are few errors. See the attached image.

Regards,
Ivo

[image]

polakovic, 2019-02-19:
Hm, we replaced gethostbyname() by getaddrinfo() because of better error handling. but getaddrinfo() is not available on windows :( I have to look at it once again, thank you for the report.

polakovic, 2019-02-19:
Pls. can you try now? I fixed absolute search paths and added one include to fix the issue with getaddrinfo().

MasterYoddha, 2019-02-19:
Works perfectly! :)

Many thanks!

closed 2019-02-19 by MasterYoddha