indigo-astronomy/indigo #185: Synscan fails to connect using SW AZGOTO with or without Handset

cstachon, 2019-04-01:
Hi,

Maybe a bug !

Trying to connect above mount to Indigo Sky without success.

Using FDTI cable(TTL/USB) I get this in SYSLOG
Apr  1 15:10:40 indigosky indigo_server[342]: 15:10:40.910130 indigo_server: indigo_mount_synscan[138, synscan_read_response, 0x69ff0360]: SYN
SCAN_TIMEOUT
Apr  1 15:10:40 indigosky indigo_server[342]: 15:10:40.910307 indigo_server: indigo_mount_synscan[46, synscan_configure, 0x69ff0360]: ERROR GE
TTING FIRMWARE
Apr  1 15:12:20 indigosky indigo_server[342]: 15:12:20.590146 indigo_server: indigo_mount_synscan[138, synscan_read_response, 0x69ff0360]: SYN
SCAN_TIMEOUT
Apr  1 15:12:25 indigosky indigo_server[342]: 15:12:25.870250 indigo_server: indigo_mount_synscan[138, synscan_read_response, 0x69ff0360]: SYN
SCAN_TIMEOUT
Apr  1 15:12:25 indigosky indigo_server[342]: 15:12:25.870425 indigo_server: indigo_mount_synscan[46, synscan_configure, 0x69ff0360]: ERROR GE
TTING FIRMWARE
Apr  1 15:12:34 indigosky indigo_server[342]: 15:12:34.910137 indigo_server: indigo_mount_synscan[138, synscan_read_response, 0x69ff0360]: SYN
SCAN_TIMEOUT
Apr  1 15:12:40 indigosky indigo_server[342]: 15:12:40.190145 indigo_server: indigo_mount_synscan[138, synscan_read_response, 0x69ff0360]: SYN
SCAN_TIMEOUT
Apr  1 15:12:40 indigosky indigo_server[342]: 15:12:40.190290 indigo_server: indigo_mount_synscan[46, synscan_configure, 0x69ff0360]: ERROR GE
TTING FIRMWARE
Apr  1 15:28:58 indigosky indigo_server[342]: 15:28:58.748632 indigo_server: indigo_mount_synscan[46, synscan_configure, 0x69ff0360]: ERROR GE
TTING FIRMWARE
Apr  1 15:29:00 indigosky indigo_server[342]: 15:29:00.283545 indigo_server: indigo_mount_synscan[46, synscan_configure, 0x69ff0360]: ERROR GE
TTING FIRMWARE
Apr  1 15:29:02 indigosky indigo_server[342]: 15:29:02.427411 indigo_server: indigo_mount_synscan[46, synscan_configure, 0x69ff0360]: ERROR GE
TTING FIRMWARE

NOTES !
1. Hit Connect greater than 3 times it says its connected(goes Blue ) but mount status is orange.

2. TTYUSB0 is in group Dialout and so is Indigo if i do "id"
uid=1000(indigo) gid=1000(indigo) groups=1000(indigo),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),101(input),108(netdev),997(gpio),998(i2c),999(spi)

3. I have tried without Handset using FDTI/USB TTL and with Handset in Direct mode and not in any mode using FDTI/USB/SERIAL adapter not go.

4. Using LX200 mount connection status goes GREEN but will do nothing - even shows the Planetarium and the mount position.

5. Using the same cables(any of them) and a Windows Laptop running Synscan Pro APP - moun is connected and works - even though Synscan App says its a AZ114GT mount (used to say AZGOTO on earliest release of APP) 

6. When I tried using Handset the handset was using latest firmware - obviously no handset firmware when not using handset except maybe on the motorboard.

7. Question how would I connect to a Networked device - so for example RPI Zero(W) running Ser2net which is on the same network as the RPI3b+ running Indigo Sky in AP mode. Ser2net would be using TCP port XXX --- Is this even possible with IndiGo Sky.

8. FYI - You probably know anyway but Indigo Sky runs without change - just swop SD card from RPI3b+ to RPI Zero(w). Access to 192.168.235.1:7624 works and so did my Canon Dslr. Now if I can get my Mount to connect LOL

cstachon, 2019-04-01:
Hi Peter,
I did use Synscan driver with a EQDIR equiv device (cable from mount to TLL FDTI USB Adapter which , as I say , works wih Synscan App .  So ,to me, the mount cable and USB adapter works its the mount hand shaking that fails IMO. I tried the other set ups as the latter did nothing not even connect. 
The Handet set up was tried in both "direct mode"(failed) and just "powered on" AKA Indi Synscan driver(failed). 
Todate the LX200 is the only set up that at least "connects" and goes green even if it cant do anything else.
OK I will do some more digging.

polakovic, 2019-04-02:
Are you sure you use the correct /dev/tty... device? The only driver which can successfully connect to EQDir device is SynScan, but it has a complicated handshake, so if the device name is not correct or not accessible (e.g. opened by another application), it will fail.

SynScan driver is probably the most popular driver (I use it too with my mounts), so it may certainly have bugs, but generally works :)

Peter

cstachon, 2019-04-02:
As best as I can be sure I was using the correct device /dev/ttyUSB0. The FDTI I use is transparent and you could see the normal led blinking when you pressed the "connect" button. I had to "submit" the USB0 as the default was the Modem type driver AMA. Did this by looking at the "serial ports" and clicking on the USB driver shown which then appeared in the "serial Port" which I then submitted using the "submit" key.     As I say I will do so more digging and use Telnet or similar to talk to the Mount (or loop tx/rx) to see if I get anything.
Just a thought but there arn't any Udev rules in play here are there ?

polakovic, 2019-04-02:
There are some udev rules involved, that's why the port is writable for _dialout_ group.

Pls. can you try it one more time, but add -vvv switch to indigo_server invocation? It will produce a very verbose log, not just errors. What we can see is just that there is no reply to any synscan command.

BTW, what mount is it? Something like this one?

https://www.astroshop.eu/alt-azimuth-with-goto/skywatcher-mount-az-synscan-goto/p,16089

Peter

cstachon, 2019-04-02:
Ok I will try as ASAP.

Yes that's one model - mine is grey and the other is black and the board
version numbers in the Mounts are different by a few points.

Just trying to get a simple "Grab and Go" for DSLR wide field :-) - already
use a UP Core (windows 10) but would like Linux version - especially if I
can get the RPI Zero(w) to work (for now just using RPI3b+ :-)

Regards

Clive Stachon

cstachon, 2019-04-03:
I am closing this as Indigo DSLR is not working they way I want (and already use) so little point for me. Many thanks for taking the time to answer queries. Good Luck

closed 2019-04-03 by cstachon

AlsFatemah, 2022-03-30:
Hello there 
I am a Mac user, my Macbook has been updated to the latest version BigSur. 
I am trying to connect my AZ GTI to INIGO server, although FTDI cable is connected and the FTDI chip 232R has been installed already and I am sure there might be something wrong with the firmware. 
I switched to Windows to update the firmware, so its been updated and I am happy with it, I have tried the SYNSCAN App from my iPhone and the mount slews. However, It is not connected on AstroTelescope software. 
This is the message I am getting: 
22:22:39.599 'Mount SynScan' selected
22:22:44.705 indigo_mount_synscan[synscan_read_response:153, 0x700003f96000]: SYNSCAN_TIMEOUT
22:22:44.705 indigo_mount_synscan[synscan_read_response:164, 0x700003f96000]: Reading response failed
22:22:49.716 indigo_mount_synscan[synscan_read_response:153, 0x700003f96000]: SYNSCAN_TIMEOUT
22:22:49.717 indigo_mount_synscan[synscan_read_response:164, 0x700003f96000]: Reading response failed
22:22:49.717 indigo_mount_synscan[synscan_configure:109, 0x700003f96000]: ERROR GETTING FIRMWARE
22:22:49.728 Trying to change baudrate
22:22:54.745 indigo_mount_synscan[synscan_read_response:153, 0x700003f96000]: SYNSCAN_TIMEOUT
22:22:54.745 indigo_mount_synscan[synscan_read_response:164, 0x700003f96000]: Reading response failed
22:22:59.757 indigo_mount_synscan[synscan_read_response:153, 0x700003f96000]: SYNSCAN_TIMEOUT
22:22:59.758 indigo_mount_synscan[synscan_read_response:164, 0x700003f96000]: Reading response failed
22:22:59.758 indigo_mount_synscan[synscan_configure:109, 0x700003f96000]: ERROR GETTING FIRMWARE
22:22:59.775 Failed to connect to mount
22:22:59.779 Failed to connect

looking forward hearing from you 

Thank you.

rumengb, 2022-03-30:
Are you setting the correct serial port?

AlsFatemah, 2022-03-30:
Could you let me know the correct serial port? 
Let me check.

rumengb, 2022-03-30:
There is no way for me to know which one it is but you will see the
available ones in the indigo control panel:
[image: image.png]
This is the linux version but it is not much different on Mac.  See "Serial
Ports" property.
You can disconnect the serial converter and refresh, then connect it and
refresh again. the one that shows up is the correct one.
Klick on it and it will be loaded in "Serial Post" property as shown above.

then you can use the client to connect the mount.

AlsFatemah, 2022-04-02:
Sorry for the late reply. 
I got what you mean. yes everything is shown in the control panel
QHY cool serial
ZWO serial 
Synscan Mount serial 
It is only shown that the mount is not connect (Red dot) in the connection settings. 
Any solutions what might be wrong?

rumengb, 2022-04-03:
I am not sure I understand! Can you provide a trace log?

AlsFatemah, 2022-04-03:
Oh sorry. I meant that everything is connected from the 'INDIGO Serial Ports', but only the mount is not from the 'Control Panel' (A red dot next to Synscan Mount)
Sure. Give me few minutes. I will screenshot the control panel and the serial ports. 
Thank you again. @rumengb

AlsFatemah, 2022-04-03:
Hello, 
Here I attached the screenshots. 
I also wonder why QHY is not imaging (Main Camera, although it is connected). 
The guide camera ZWO is connected and capturing. 
My problem is why Synscan Mount is not connected although it is linked and you can see from the system information > USB > FT232R. 
Also check INDIGO Ports. 

![Screen Shot 2022-04-04 at 12 27 08 AM](https://user-images.githubusercontent.com/102630537/161450875-d4a40874-903d-4b81-a80d-1
[image]
8d
[image]
270873d41.png)
[image]
[image]

I wish I could get help here. 
Thank you.

rumengb, 2022-04-04:
As I can see uou have 2 serial ports available: /dev/cu.usbserial-AB0NQKXK
and /dev/su.usbserial-0001
And you are trying to connect to /dev/cu.usbserial-AB0NQKXK. Have you tried
/dev/su.usbserial-0001?

Rumen

polakovic, 2022-04-05:
What QHY camera is it? If it is a legacy CCD camera, use "QHY Camera" driver. If it is some newer CMOS camera, use "QHY Camera 2" driver.

AlsFatemah, 2022-04-05:
> 
Thanks for the suggestion. Still not working either. 

[image]

Might it be because I didn't download the Synscan App on my MacBook?

AlsFatemah, 2022-04-05:
> What QHY camera is it? If it is a legacy CCD camera, use "QHY Camera" driver. If it is some newer CMOS camera, use "QHY Camera 2" driver.

It worked now with QHY Camera 2 
I own 183M. Thank you!

rumengb, 2022-04-05:
did you try 9600-8N1 ? I doubt it would work at 115200....

AlsFatemah, 2022-04-06:
> did you try 9600-8N1 ? I doubt it would work at 115200....
> […](#)

When I traced the log, I have seen that the baudrate has  actually changed by itself, from 115200-8N1 to 9600-8N1 or from 9600-8N1 to 115200-8N1. But I think either way is not working. 
I have no idea what is wrong, It was working with me while I using Stellarmate. I will check again tonight, I will screenshot the logs and let you know. 
Shall I target the mount towards Polaris as usual? 
Thank you again @rumengb

polakovic, 2022-04-07:
I'm sorry, I'm late to this discussion :(

Just to make sure - if you connect to SW mount WITH SynScan hand controller in native mode, you have to use NexStar driver. If you use it with SynScan hand controller in PC DIRECT mode, you have to use really SynScan driver and if you use it without hand controller with some kind of EQDir device, you have to use SynScan driver too.  Is it your case?

AlsFatemah, 2022-04-07:
It's alright @polakovic 
How do I know if the mount is on native mode? 
I am using EQDir cable directly from the mount to my MAC without hand controller. 
I will try to connect the mount to Nexstar now to see whether it is connecting or not.  
Thank you.

AlsFatemah, 2022-04-07:
Mount Nexstar:

0:37:48.131 'Mount Nexstar' selected
00:37:53.207 indigo_mount_nexstar[mount_open:109, 0x700005d7d000]: open_telescope(/dev/cu.usbserial-0001) = -1 (Undefined error: 0)
00:37:53.256 Failed to connect

For the Synscan Mount: 
00:47:54.810 'Mount SynScan' selected
00:47:59.856 indigo_mount_synscan[synscan_read_response:154, 0x700005d7d000]: SYNSCAN_TIMEOUT
00:47:59.856 indigo_mount_synscan[synscan_read_response:165, 0x700005d7d000]: Reading response failed
00:48:04.869 indigo_mount_synscan[synscan_read_response:154, 0x700005d7d000]: SYNSCAN_TIMEOUT
00:48:04.870 indigo_mount_synscan[synscan_read_response:165, 0x700005d7d000]: Reading response failed
00:48:04.870 indigo_mount_synscan[synscan_configure:109, 0x700005d7d000]: ERROR GETTING FIRMWARE
00:48:09.902 indigo_mount_synscan[synscan_read_response:154, 0x700005d7d000]: SYNSCAN_TIMEOUT
00:48:09.903 indigo_mount_synscan[synscan_read_response:165, 0x700005d7d000]: Reading response failed
00:48:14.915 indigo_mount_synscan[synscan_read_response:154, 0x700005d7d000]: SYNSCAN_TIMEOUT
00:48:14.915 indigo_mount_synscan[synscan_read_response:165, 0x700005d7d000]: Reading response failed
00:48:14.917 indigo_mount_synscan[synscan_configure:109, 0x700005d7d000]: ERROR GETTING FIRMWARE
00:48:14.931 Failed to connect

I can see the baudrate changes from as I said 9600-8N1 to 11520-8N1
I have also tried both usbserial 

I was searching and I have found out this 
(https://plugable.com/pages/prolific-drivers)
Should I have to download it? 
Although I have the FTDI installation already. 
please see the attached photos and screenshot of the version 
Thank you.

[image]
[image]

[image]

polakovic, 2022-04-08:
> It's alright @polakovic
> How do I know if the mount is on native mode?
> I am using EQDir cable directly from the mount to my MAC without hand controller.
> I will try to connect the mount to Nexstar now to see whether it is connecting or not.
> Thank you.

If you don't do anything, hand controller is in native mode. You have to switch it to PC direct mode to act as a EQDir adapter. But if you use EQDir adapter, it is not relevant and you have to use SynScan driver.

Nevertheless, from what I see you probably use a wrong serial port. Try disconnect it, click refresh, connect and click refresh once again. If there will be a difference, it is the right port. Otherwise something is wrong on hardware or OS level.

AlsFatemah, 2022-04-08:
@polakovic 
Thank you for your reply. I am sure that the EQDir is switched to PC as I have already used the Synscqn App from Windows screen on my MAC (Same laptop). That time I was using another software from windows. 
Actually, I have tried to downlaod Synscan App on Mac but it's not compatible with the new version. I have the latest version of MAC (BigSur). I was thinking that it might be of Synscan driver is not installed on MAC. 
I will check the OS level tonight. Hopefully the issue will be resolved. I will let you know once I get it solved. 
Thank you again.

polakovic, 2022-04-09:
You may need a driver for USB-to-serial adapter inside EQDir. If it is FTDI chipset, try to download a driver from here: https://ftdichip.com/drivers/vcp-drivers/

AlsFatemah, 2022-04-09:
[image]

I think I have installed it already. 
Please see the attached screenshot.