indigo-astronomy/indigo #186: PORTS property on linux doesn't show symlinks

polakovic, 2019-04-10:


rumengb, 2019-04-11:
Will fix it as soon as i can. On vacation now :)

rumengb, 2019-04-11:
fixed with commits: bd522748e1ca62672f790da0527847256f7f7994 and a24432ca1189df12426d8179dff019e60b304800

polakovic, 2019-04-12:
Doesn't work :( Duplicate items...

19:54:10.798241 indigo_server: INDIGO Bus: property definition
19:54:10.798263 indigo_server: 'Mount SynScan'.'DEVICE_PORTS' SWITCH rw Ok 2.0 OneOfMany { // Serial ports
19:54:10.798283 indigo_server:   'REFRESH' = Off // Refresh
19:54:10.798303 indigo_server:   '/dev/ttyUSB1' = Off // /dev/ttyUSB1
19:54:10.798323 indigo_server:   '/dev/serial1' = Off // /dev/serial1
19:54:10.798342 indigo_server:   '/dev/ttyAMA0' = Off // /dev/ttyAMA0
19:54:10.798362 indigo_server:   '/dev/ttyUSB1' = Off // /dev/ttyUSB1
19:54:10.798382 indigo_server:   '/dev/ttyUSB0' = Off // /dev/ttyUSB0
19:54:10.798400 indigo_server: }

polakovic, 2019-04-12:
BTW, symlink (ttyUPB -> ttyUSB0) is missing :)

rumengb, 2019-04-13:
please post output of:
ls -al /dev
and
ls -al /dev/serial/by-id

polakovic, 2019-04-14:
PegasusAstro UBP and EQDir is connected

polakovic@raspi32:~/indigo $ lsusb
Bus 001 Device 047: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 046: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 045: ID 0424:2517 Standard Microsystems Corp. Hub
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

polakovic@raspi32:~/indigo $ ls -la /dev
total 4
drwxr-xr-x 15 root root        5640 Apr 14 06:22 .
drwxr-xr-x 22 root root        4096 Aug 29  2018 ..
crw-r--r--  1 root root     10, 235 Mar 17 19:17 autofs
drwxr-xr-x  2 root root         580 Mar 17 19:17 block
crw-------  1 root root     10, 234 Mar 17 19:17 btrfs-control
drwxr-xr-x  3 root root          60 Jan  1  1970 bus
crw-------  1 root root     10,  63 Mar 17 19:17 cachefiles
drwxr-xr-x  2 root root        4800 Apr 14 06:22 char
crw-------  1 root root      5,   1 Mar 17 19:36 console
crw-------  1 root root     10,  62 Mar 17 19:17 cpu_dma_latency
crw-------  1 root root     10, 203 Mar 17 19:17 cuse
drwxr-xr-x  7 root root         140 Mar 17 19:17 disk
crw-rw----  1 root video    29,   0 Mar 17 19:17 fb0
lrwxrwxrwx  1 root root          13 Nov  3  2016 fd -> /proc/self/fd
crw-rw-rw-  1 root root      1,   7 Mar 17 19:17 full
crw-rw-rw-  1 root root     10, 229 Mar 17 19:17 fuse
crw-rw----  1 root gpio    254,   0 Mar 17 19:17 gpiochip0
crw-rw----  1 root gpio    254,   1 Mar 17 19:17 gpiochip1
crw-rw----  1 root gpio    248,   0 Mar 17 19:17 gpiomem
crw-------  1 root root     10, 183 Mar 17 19:17 hwrng
lrwxrwxrwx  1 root root          25 Nov  3  2016 initctl -> /run/systemd/initctl/fifo
drwxr-xr-x  2 root root          60 Mar 31 17:47 input
crw-r--r--  1 root root      1,  11 Mar 17 19:17 kmsg
lrwxrwxrwx  1 root root          28 Nov  3  2016 log -> /run/systemd/journal/dev-log
crw-rw----  1 root disk     10, 237 Mar 17 19:17 loop-control
brw-rw----  1 root disk      7,   0 Mar 17 19:17 loop0
brw-rw----  1 root disk      7,   1 Mar 17 19:17 loop1
brw-rw----  1 root disk      7,   2 Mar 17 19:17 loop2
brw-rw----  1 root disk      7,   3 Mar 17 19:17 loop3
brw-rw----  1 root disk      7,   4 Mar 17 19:17 loop4
brw-rw----  1 root disk      7,   5 Mar 17 19:17 loop5
brw-rw----  1 root disk      7,   6 Mar 17 19:17 loop6
brw-rw----  1 root disk      7,   7 Mar 17 19:17 loop7
drwxr-xr-x  2 root root          60 Mar 17 19:17 mapper
crw-r-----  1 root kmem      1,   1 Mar 17 19:17 mem
crw-------  1 root root     10,  59 Mar 17 19:17 memory_bandwidth
brw-rw----  1 root disk    179,   0 Mar 17 19:17 mmcblk0
brw-rw----  1 root disk    179,   1 Mar 17 19:17 mmcblk0p1
brw-rw----  1 root disk    179,   2 Mar 17 19:17 mmcblk0p2
drwxrwxrwt  2 root root          40 Jan  1  1970 mqueue
drwxr-xr-x  2 root root          60 Mar 17 19:17 net
crw-------  1 root root     10,  61 Mar 17 19:17 network_latency
crw-------  1 root root     10,  60 Mar 17 19:17 network_throughput
crw-rw-rw-  1 root root      1,   3 Mar 17 19:17 null
crw-------  1 root root    108,   0 Mar 17 19:17 ppp
crw-rw-rw-  1 root tty       5,   2 Apr 14  2019 ptmx
drwxr-xr-x  2 root root           0 Nov  3  2016 pts
brw-rw----  1 root disk      1,   0 Mar 17 19:17 ram0
brw-rw----  1 root disk      1,   1 Mar 17 19:17 ram1
brw-rw----  1 root disk      1,  10 Mar 17 19:17 ram10
brw-rw----  1 root disk      1,  11 Mar 17 19:17 ram11
brw-rw----  1 root disk      1,  12 Mar 17 19:17 ram12
brw-rw----  1 root disk      1,  13 Mar 17 19:17 ram13
brw-rw----  1 root disk      1,  14 Mar 17 19:17 ram14
brw-rw----  1 root disk      1,  15 Mar 17 19:17 ram15
brw-rw----  1 root disk      1,   2 Mar 17 19:17 ram2
brw-rw----  1 root disk      1,   3 Mar 17 19:17 ram3
[... 217 lines cut ...]
crw-rw----  1 root tty       7, 135 Mar 23 16:11 vcsa7
crw-rw----  1 root tty       7, 136 Mar 23 16:11 vcsa8
crw-rw----  1 root tty       7, 137 Mar 23 16:11 vcsa9
crw-rw----  1 root video   246,   0 Mar 17 19:17 vcsm
crw-------  1 root root     10, 137 Mar 17 19:17 vhci
crw-------  1 root root     10, 130 Mar 17 19:17 watchdog
crw-------  1 root root    252,   0 Mar 17 19:17 watchdog0
crw-rw-rw-  1 root root      1,   5 Mar 17 19:17 zero

polakovic@raspi32:~/indigo $ ls -al /dev/serial/by-id
total 0
drwxr-xr-x 2 root root 80 Apr 14 06:22 .
drwxr-xr-x 4 root root 80 Apr 14 06:22 ..
lrwxrwxrwx 1 root root 13 Apr 14 06:22 usb-Pegasus_Astro_UPB_v1.7_PA36T4RB-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root 13 Apr 14 06:22 usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0 -> ../../ttyUSB1

06:24:54.741483 indigo_server: INDIGO Bus: property definition
06:24:54.741675 indigo_server: 'Mount SynScan'.'DEVICE_PORTS' SWITCH rw Ok 2.0 OneOfMany { // Serial ports
06:24:54.741863 indigo_server:   'REFRESH' = Off // Refresh
06:24:54.742048 indigo_server:   '/dev/ttyUSB1' = Off // /dev/ttyUSB1
06:24:54.742136 indigo_server:   '/dev/serial1' = Off // /dev/serial1
06:24:54.742390 indigo_server:   '/dev/ttyAMA0' = Off // /dev/ttyAMA0
06:24:54.742970 indigo_server:   '/dev/ttyUSB1' = Off // /dev/ttyUSB1
06:24:54.743025 indigo_server:   '/dev/ttyUSB0' = Off // /dev/ttyUSB0
06:24:54.743363 indigo_server: }

polakovic, 2019-04-14:
What I suppose to see is /dev/ttyUSB0 and /dev/ttyUSB1, but also /dev/ttyUPB (preselected by UPB driver)!

rumengb, 2019-04-19:
Created workaround, Pegasus is not seen by the OS as a real UART. Confirmed to work.

closed 2019-04-19 by rumengb