indigo-astronomy/indigo #215: Canon EOS650D doesn't work with INDIGO Sky

polakovic, 2019-07-01:
indigo@indigosky:~ $ indigo_server --enable-rpi-management
19:31:18.563726 indigo_server: indigo_ccd_gphoto2[2556, device_connect, 0x76f8a000]: [rc:-7] gp_camera_init failed
19:31:18.563861 indigo_server: indigo_ccd_gphoto2[2656, device_attach, 0x76f8a000]: connecting failed slot '0' device 'Canon EOS 650D 04a9:323b (bus 001, device 004)'

root@indigosky:/home/indigo# sudo /usr/bin/indigo_server --enable-rpi-management
19:25:00.624616 indigo_server: indigo_ccd_gphoto2[2556, device_connect, 0x76f20000]: [rc:-53] gp_camera_init failed
19:25:00.624769 indigo_server: indigo_ccd_gphoto2[2656, device_attach, 0x76f20000]: connecting failed slot '0' device 'Canon EOS 650D 04a9:323b (bus 001, device 004)'

(Reported by Marc on FB)

PatchworkBoy, 2019-07-01:
(‘Marc on FB’ being me... :D)

PatchworkBoy, 2019-07-02:
Scratch that - wonked install of gphoto2. Fixed.

polakovic, 2019-07-02:
OK, great! :)

closed 2019-07-02 by polakovic