indigo-astronomy/indigo #220: Indigo_server DSLR failure when exp time over 55 secs - fails with debayer failed

cstachon, 2019-07-08:
Indigo_server on Raspbian stretch on RPI3b+ Indigo_server version is 2.0-95
When using the Internal Web server on port 7624 it is not possible to image with exp time of over 55 secs - image is taken and displayed on camera LCD but Web Imager fails with "debayer failed". Using 40 secs or less it works.   NOTE I have the Canon 100d set for High ISO NR max and Long Exposure NR is on. 
Started Indigo_server from CL as below
indigo_server
Output was
sh: 1: s_rpi_ctrl.sh: not found
sh: 1: s_rpi_ctrl.sh: not found
14:12:30.869998 indigo_server: indigo_ccd_gphoto2[1430, thread_capture, 0x674f0360]: [rc:-1,camera:0x748004a0,context:0x748008b8] gp_camera_capture
14:12:30.872740 indigo_server: indigo_ccd_gphoto2[525, process_dslr_image_debayer, 0x6b4f8360]: [rc:-2] libraw_open_buffer failed: 'Unsupported file format or not RAW file'
14:18:46.212136 indigo_server: indigo_ccd_gphoto2[1430, thread_capture, 0x674f0360]: [rc:-1,camera:0x748004a0,context:0x748008b8] gp_camera_capture
14:18:46.213823 indigo_server: indigo_ccd_gphoto2[525, process_dslr_image_debayer, 0x67cf1360]: [rc:-2] libraw_open_buffer failed: 'Unsupported file format or not RAW file'

I know NR gave GPHOTO problems in the past so is it just the same issue again?

rumengb, 2019-07-08:
Hi,
Can you tell us what Browser are you using? It is known that MS Edge does not play well with web sockets and we do not recommend using Edge. It is known it resets the connection every minute or so, and the time you reported is close to that.

polakovic, 2019-07-08:
DSLR is not supported by webGUI yet, just astro cameras. It is not so easy, driver should be able to produce low quality JPEG preview from RAW and this is not implemented yet.

cstachon, 2019-07-08:
Just to add its is Long Exp NR thats the problem as High Iso NR on and Long Exp NR off works at 60secs ok

cstachon, 2019-07-08:
Web Browser was Opera on Windows

polakovic, 2019-07-08:
High ISO NR ON probably makes image incompatible with libraw.

cstachon, 2019-07-08:
Tried 120secs with just High ISO on and Long Exp off - works ok . 

p.s. can we have an enhancement in the future to have a button to clear status message area. Last error stays on for ever :-)

cstachon, 2019-07-08:
ok so no using Long Exp On setting :-)

polakovic, 2019-07-08:
Yes, a good point. But it is just a cosmetic issue. The main problem is, that DSLR is not generally supported :) indigo_process_dslr_image() doesn't produce preview image the same way as indigo_process_image().

polakovic, 2026-10-02:
This was reported against the gphoto2 driver, which has since been replaced by the PTP driver. DSLR preview in the web GUI is tracked in #702.

closed 2026-10-02 by polakovic