indigo-astronomy/indigo #223: Focuser support to NexStar driver

polakovic, 2019-07-12:
https://indilib.org/forum/general/4470-celestron-motorised-focuser-is-it-supported-in-ekos-yet.html?start=0

orangemaze, 2019-10-09:
Any updates on this?

polakovic, 2019-10-09:
Yes, standalone Celestron focuser is already supported (and does work) as a side effect of PlaneWave EFA focuser support (which doesn't work yet).

https://github.com/indigo-astronomy/indigo/tree/master/indigo_drivers/focuser_efa

Focuser connected to NexStar mount is not supported yet, but just arrived SE mount for development and testing, so I can finally start to work on it.

orangemaze, 2019-10-09:
I do not see that listed on the INDIGO site

polakovic, 2019-10-09:
It is not listed, because the driver still doesn't work with EFA :(

polakovic, 2020-08-10:
Confirmed it works.

closed 2020-08-10 by polakovic