indigo-astronomy/indigo #235: Slow refresh on Web GUI when a number of devices being used

cstachon, 2019-08-01:
Tried this a number of ways even on the host using 127.0.0.1:7624 but all are slow to refresh to the extent you cant use the Webgui. Even just collapsing items takes 3 mins or so. However changes are made to properties quite fast just not being shown on the Webgui. Tried this on Chrome,Firefox and on Windows,Linux(Ubuntu) all did the same.   

Using the local Control Panel doesn't seem to be effected.

Kit attached was AZEQ6GT (EQ mode Synscan driver) ,Moonlite focuser ,Blox GPS, and Canon DSLR 100d.

polakovic, 2019-08-02:
What build is it? We had this issue some weeks ago but I believe it is fixed already.

cstachon, 2019-08-02:
I will check and let you know

polakovic, 2019-08-02:
Pls. make sure you have the most recent 2.0-100.

cstachon, 2019-08-02:
Ok was on 99 now updated. See how it goes thanks

cstachon, 2019-08-03:
solved by 100 - thanks

closed 2019-08-03 by cstachon