indigo-astronomy/indigo #236: Webgui hint bubbles (Preview,batch,download) do not disappear and block count,timer

cstachon, 2019-08-03:
Any chance that this behaviour being changed so they automatically disappear either after a couple of seconds or when the mouse is not over the icon. As I say they block the Count,Timer etc above. Happens using Firefox,Chrome,IE
:-)

polakovic, 2019-08-03:
I'll check it, but with Safari they disappear as soon as I move mouse cursor away.

We use Bootstrap CSS framework, but on the first sight I don't see any way how to hide tooltips automatically. Well, and as far as it was one of the most wanted feature, we probably can't just remove them :)

cstachon, 2019-08-03:
hi here are some example screen prints attached.

Note - if you dont select(click) an Icon yes the bubbles do go
automatically but if you click on an icon it stays. You can see the arrow
(mouse pointer) which is nowhere near the icon so should have gone. How
about making the "gaps" bigger above the icons to allow for the bubble or a
nice option to disable them :-)