indigo-astronomy/indigo #237: Running 2 or more Indigo_servers on different pc's (RPI's)

cstachon, 2019-08-05:
Hi

Having a play but I would like to do the following:-

1. Run Indigo_server on PC A which has just DSLR connected
2. Run Indigo_server on PC B which will have mount focuser and use the DSLR on PC A
3. Run WebGui connected to PC B and control both PC A and PC B resources.

Have had a play but cannot get it to work.

Attempts have been
1. Start Indigo_server on PC B with just indigo_server
2. Sartt indigo_server on PC A with indigo_server -r ip_addressof_pc_B
3. Connect Webgui to PC B

But although I can control the DSLR on PC B and change properties if I take an image nothing is shown in Webgui or saved any where.
Version 100 on both PC's
settings used :-
Enable Preview disabled
FITS
Debayer set NONE (but have tried "linear"
saved config and clicked load
Upload to client

Note when I connect another WebGUI to PC A I can see the DSLR and any images taken !

Can I do this ?

What is the "Agent" for when server window shows dslr@ip.address ok

Logs all looked ok - no errors

Any indicators would be most helpful

polakovic, 2019-08-05:
It is a limitation of agent acting as backend for webGUI, it can control only "local" camera. The problem is, that webGUI can't display FITS format directly and doing the conversion for remote server is inefficient. But I tried this setup with native control panel and it worked.

Server chaining is in general legacy INDI approach (INDI client can't connect to more than one server), it doesn't have too much sense with INDIGO, maybe for INDI client or for a proxy server on intranet boundary. But if you want to do it this way for some reason, use RPi with DSLR as local one (with webGUI) and RPi with focuser as remote.

cstachon, 2019-08-06:
ok thanks for the info

closed 2019-08-06 by cstachon