indigo-astronomy/indigo #247: drivers list

GuLinux, 2019-09-16:
Hi,
I'm thinking of migrating my project ( [AstroPhoto Plus](https://github.com/GuLinux/AstroPhoto-Plus) ) to INDIGO, it looks very promising, particularly the JSON api would allow me to ditch all the pyindi-client part entirely and directly talk with the INDIGO server.

One thing leaves me puzzled: how to determine dynamically all the available drivers installed in the system?
With INDI, I can look at the `/usr/share/indi/` directory, and get info about drivers by looking at the XML files in that directory.

With INDIGO I noticed there are two similar files: `indigo_drivers` and `indigo_linux_drivers`. There are two problem with this though:
 - It doesn't seem to be part of the "official" installation (I didn't see anything on github, it looks generated for deb/ubuntu packages only)
 - there file format looks like `driver_binary, driver_name, type`, where the type looks like an integer (enum?). How to properly map the type to a named category (CCD, Focuser, etc)?
Cheers,
Marco

polakovic, 2019-09-16:
Hi Marco,

you usually don't need the list of the drivers (and if you do, you can find DRIVERS property on Server device on each server), but the list of devices (as far as some of them are plug & play and one driver can handle multiple physical devices and create multiple logical devices for each physical device).

The easiest way which I use in our applications is to issue an enumerate request and to monitor the response for a definition of INFO properties for all devices.

Peter

GuLinux, 2019-09-16:
Hi,
The idea in AstroPhoto Plus is that you launch a new instance of the INDI server selecting all the driver you want/need. See here: [https://astrophotoplus.gulinux.net/user/pages/03.features/02_INDIServer-Select-Devices_20190120_165429.png](https://astrophotoplus.gulinux.net/user/pages/03.features/02_INDIServer-Select-Devices_20190120_165429.png).
I had a look at the driver property, I guess this would allow me to dynamically start and stop drivers on demand right?
I'm closing the issue anyway, I think I have enough info for now to proceed :)
Thanks,
Marco

closed 2019-09-16 by GuLinux

rumengb, 2019-09-16:
Yes you can load and unload drivers dynamically.