indigo-astronomy/indigo #267: Feature request: gpsd driver

ghulands, 2020-01-07:
Hi,
I have a gps installed via a raspberry pi hat to get the pps from it to have accurate timing on the device. The problem with this is that I can't share the serial port for the generic nmea driver. GPSD is installed and claims the port.

It'd be great if a gpsd driver could be added.

ghulands, 2020-01-07:
I was looking into the source of the nmea driver and saw that you could connect to a tcp host and port. I wrote a [wrapper service](https://gist.github.com/ghulands/56f6d17a424baff602f4c7661e018508) that configures a stream from GPSD in the nmea 0183 format and then I wanted indigo to connect to that.

I came across a bug in the code which i have raised a PR for #269 . I was able to verify my wrapper script as well. Feel free to include it in the indigosky image.

ghulands, 2020-01-07:
Closing since the PR was merged

closed 2020-01-07 by ghulands

rumengb, 2020-01-12:
There is GPSD driver in indigo now!