indigo-astronomy/indigo #271: indigo_ccd_ptp: Need to make multiple attempts to connect to the SONY camera

tail-feather, 2020-01-09:
Connect the SONY α7S and start the INDIGO server.

```
18:39:12.421411 indigo_server: Driver indigo_ccd_ptp 2.0.0.2 loaded
18:39:12.421423 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421427 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421430 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421457 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421462 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421465 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421468 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421470 indigo_server: indigo_ccd_ptp[hotplug_callback:491]: libusb_get_device_descriptor ->  OK
18:39:12.421474 indigo_server: indigo_ccd_ptp[indigo_ccd_ptp:655]: libusb_hotplug_register_callback ->  OK
18:39:12.427414 indigo_server: indigo_ccd_ptp: 'Sony Alpha A7S #0106' attached
```

Attempt to connect as the device appears.

```
18:39:19.419297 indigo_server: indigo_ccd_ptp[ptp_open:783]: libusb_get_device_descriptor() -> OK
18:39:19.419345 indigo_server: indigo_ccd_ptp[ptp_open:786]: libusb_open() -> OK
18:39:19.419355 indigo_server: indigo_ccd_ptp[ptp_open:797]: libusb_get_config_descriptor(0) -> OK
18:39:19.419359 indigo_server: indigo_ccd_ptp[ptp_open:803]: PTP CONFIG = 1 IFACE = 0
18:39:19.428067 indigo_server: indigo_ccd_ptp[ptp_open:816]: libusb_set_configuration(1) -> OK
18:39:19.428108 indigo_server: indigo_ccd_ptp[ptp_open:821]: libusb_claim_interface(0) -> OK
18:39:19.428670 indigo_server: indigo_ccd_ptp[ptp_open:848]: PTP EP OUT = 02 IN = 81 INT = 83
18:39:19.428686 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request OpenSession (1002) 00000000 [00000001]
18:39:19.428759 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(16) -> OK
18:39:19.434382 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 12
18:39:19.434412 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: response OK (2001) 00000000 []
18:39:19.434422 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request GetDeviceInfo (1001) 00000001 []
18:39:19.434481 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(12) -> OK
18:39:19.449445 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 255
18:39:19.449466 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: data 1001 00000001 +243 bytes
18:39:19.449750 indigo_server: indigo_ccd_ptp[ptp_transaction:952]: libusb_bulk_transfer() -> OK, 12
18:39:19.449762 indigo_server: indigo_ccd_ptp[962, ptp_transaction]: response OK (2001) 00000001 []
18:39:19.449779 indigo_server: indigo_ccd_ptp[1151, ptp_initialise]: device info
18:39:19.449785 indigo_server: PTP 1.00 + Nikon (000a), Sony Extension 1.00
18:39:19.449788 indigo_server: ILCE-7S [3.0], Sony Corporation, #00000000000000003282740003023974
18:39:19.449791 indigo_server: operations:
18:39:19.449795 indigo_server:   1002 OpenSession
18:39:19.449798 indigo_server:   1003 CloseSession
18:39:19.449801 indigo_server:   1001 GetDeviceInfo
18:39:19.449804 indigo_server:   1004 GetStorageIDs
18:39:19.449807 indigo_server:   1005 GetStorageInfo
18:39:19.449810 indigo_server:   1006 GetNumObjects
18:39:19.449813 indigo_server:   1007 GetObjectHandles
18:39:19.449816 indigo_server:   1008 GetObjectInfo
18:39:19.449819 indigo_server:   1009 GetObject
18:39:19.449822 indigo_server:   100a GetThumb
18:39:19.449825 indigo_server:   101b GetPartialObject
18:39:19.449828 indigo_server:   9201 SDIOConnect_Sony
18:39:19.449831 indigo_server:   9202 GetSDIOGetExtDeviceInfo_Sony
18:39:19.449834 indigo_server:   9205 SetControlDeviceA_Sony
18:39:19.449837 indigo_server:   9207 SetControlDeviceB_Sony
18:39:19.449840 indigo_server:   9209 GetAllDevicePropData_Sony
18:39:19.449843 indigo_server: events:
18:39:19.449846 indigo_server:   c201 ObjectAdded_Sony
18:39:19.449849 indigo_server:   c202 ObjectRemoved_Sony
18:39:19.449852 indigo_server:   c203 PropertyChanged_Sony
18:39:19.449855 indigo_server: properties:
18:39:19.449859 indigo_server: indigo_ccd_ptp[616, ptp_sony_initialise]: device ext_info
18:39:19.449865 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request SDIOConnect_Sony (9201) 00000002 [00000001, 00000000, 00000000]
18:39:19.449932 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(24) -> OK
18:39:19.452233 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 20
18:39:19.452246 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: data 9201 00000002 +8 bytes
18:39:19.452578 indigo_server: indigo_ccd_ptp[ptp_transaction:952]: libusb_bulk_transfer() -> OK, 12
18:39:19.452594 indigo_server: indigo_ccd_ptp[962, ptp_transaction]: response OK (2001) 00000002 []
18:39:19.452601 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request SDIOConnect_Sony (9201) 00000003 [00000002, 00000000, 00000000]
18:39:19.452693 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(24) -> OK
[... 358 lines cut ...]
18:39:39.446719 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Aspect ratio (d211), type = uint8 -> DSLR_ASPECT_RATIO (1)
18:39:39.446729 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: FocusStatus_Sony (d213), type = uint8 -> NONE (1)
18:39:39.446737 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: ISO (d21e), type = uint32 -> DSLR_ISO (160)
18:39:39.446743 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Picture effect (d21b), type = uint16 -> DSLR_PICTURE_STYLE (32768)
18:39:39.446756 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d21d (d21d), type = uint8 -> NONE (0)
18:39:39.446768 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d21f (d21f), type = uint8 -> NONE (1)
18:39:39.446781 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d217 (d217), type = uint8 -> NONE (1)
18:39:39.446793 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d218 (d218), type = int8 -> NONE (100)
18:39:39.446804 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d219 (d219), type = uint8 -> NONE (1)
18:39:39.446815 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Autofocus_Sony (d2c1), type = uint16 -> NONE (1)
18:39:39.446830 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Capture_Sony (d2c2), type = uint16 -> NONE (1)
18:39:39.446842 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d2c3 (d2c3), type = uint16 -> NONE (1)
18:39:39.446855 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d2c9 (d2c9), type = uint16 -> NONE (1)
18:39:39.446867 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Movie_Sony (d2c8), type = uint16 -> NONE (1)
18:39:39.446879 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d212 (d212), type = uint8 -> NONE (0)
18:39:39.446892 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d221 (d221), type = uint8 -> NONE (1)
18:39:39.446904 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d214 (d214), type = uint32 -> NONE (18324480)
18:39:39.446916 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d215 (d215), type = uint16 -> NONE (0)
18:39:39.446928 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d2c5 (d2c5), type = uint16 -> NONE (1)
18:39:39.446940 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: StillImage_Sony (d2c7), type = uint16 -> NONE (1)
```

Connected to the device and the properties are complete.

Is this a timing issue?

polakovic, 2020-01-09:
"data 9209 00000006 +8 bytes" is definitely not enough data, it should be much more. Pls. can you try to put some delay between lines 655 and 656 in indigo_ptp_sony.c? Thank you!

tail-feather, 2020-01-10:
Waiting one second before connecting, the camera no longer restarts!

```diff
--- a/indigo_drivers/ccd_ptp/indigo_ptp_sony.c
+++ b/indigo_drivers/ccd_ptp/indigo_ptp_sony.c
@@ -652,6 +652,7 @@ bool ptp_sony_initialise(indigo_device *device) {
                        for (uint16_t *property = properties; *property; property++) {
                                INDIGO_LOG(indigo_log("  %04x %s", *property, ptp_property_sony_code_label(*property)));
                        }
+                       sleep(1);
                        ptp_transaction_3_0(device, ptp_operation_sony_SDIOConnect, 3, 0, 0);
                        if (ptp_transaction_0_0_i(device, ptp_operation_sony_GetAllDevicePropData, &buffer, &size)) {
                                uint8_t *source = buffer;
```

But, 2-4 steps still remaining.

GetAllDevicePropData seems complete. ("data 9209 00000006 +1164 bytes", I checked the output matches.)
Oddly, all properties are evaluated as `DSLR_COMPRESSION`.
How do I fix it?

---------

Log:

First time connection:
```
21:45:39.605873 indigo_server: INDIGO Bus: property update
21:45:39.605906 indigo_server: 'Sony Alpha A7S #0105'.'CONNECTION' SWITCH rw Busy 2.0 OneOfMany {
21:45:39.605914 indigo_server:   'CONNECTED' = On 
21:45:39.605921 indigo_server:   'DISCONNECTED' = Off 
21:45:39.605930 indigo_server: }
21:45:39.605987 indigo_server: 20 ← { "setSwitchVector": { "device": "Sony Alpha A7S #0105", "name": "CONNECTION", "state": "Busy", "items": [  { "name": "CONNECTED", "value": true }, { "name": "DISCONNECTED", "value": false } ] } }
21:45:39.606078 indigo_server: indigo_ccd_ptp[ptp_open:783]: libusb_get_device_descriptor() -> OK
21:45:39.606127 indigo_server: indigo_ccd_ptp[ptp_open:786]: libusb_open() -> OK
21:45:39.606150 indigo_server: indigo_ccd_ptp[ptp_open:797]: libusb_get_config_descriptor(0) -> OK
21:45:39.606161 indigo_server: indigo_ccd_ptp[ptp_open:803]: PTP CONFIG = 1 IFACE = 0
21:45:39.614624 indigo_server: indigo_ccd_ptp[ptp_open:816]: libusb_set_configuration(1) -> OK
21:45:39.614721 indigo_server: indigo_ccd_ptp[ptp_open:821]: libusb_claim_interface(0) -> OK
21:45:39.615111 indigo_server: indigo_ccd_ptp[ptp_open:848]: PTP EP OUT = 02 IN = 81 INT = 83
21:45:39.615144 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request OpenSession (1002) 00000000 [00000001]
21:45:39.615246 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(16) -> OK
21:45:39.621129 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 12
21:45:39.621161 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: response OK (2001) 00000000 []
21:45:39.621174 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request GetDeviceInfo (1001) 00000001 []
21:45:39.621268 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(12) -> OK
21:45:39.623211 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 255
21:45:39.623228 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: data 1001 00000001 +243 bytes
21:45:39.623427 indigo_server: indigo_ccd_ptp[ptp_transaction:952]: libusb_bulk_transfer() -> OK, 12
21:45:39.623438 indigo_server: indigo_ccd_ptp[962, ptp_transaction]: response OK (2001) 00000001 []
21:45:39.623458 indigo_server: indigo_ccd_ptp[1151, ptp_initialise]: device info
21:45:39.623467 indigo_server: PTP 1.00 + Nikon (000a), Sony Extension 1.00
21:45:39.623472 indigo_server: ILCE-7S [3.0], Sony Corporation, #00000000000000003282740003023974
21:45:39.623476 indigo_server: operations:
21:45:39.623481 indigo_server:   1002 OpenSession
21:45:39.623485 indigo_server:   1003 CloseSession
21:45:39.623489 indigo_server:   1001 GetDeviceInfo
21:45:39.623494 indigo_server:   1004 GetStorageIDs
21:45:39.623498 indigo_server:   1005 GetStorageInfo
21:45:39.623502 indigo_server:   1006 GetNumObjects
21:45:39.623506 indigo_server:   1007 GetObjectHandles
21:45:39.623511 indigo_server:   1008 GetObjectInfo
21:45:39.623517 indigo_server:   1009 GetObject
21:45:39.623521 indigo_server:   100a GetThumb
21:45:39.623526 indigo_server:   101b GetPartialObject
21:45:39.623531 indigo_server:   9201 SDIOConnect_Sony
21:45:39.623535 indigo_server:   9202 GetSDIOGetExtDeviceInfo_Sony
21:45:39.623540 indigo_server:   9205 SetControlDeviceA_Sony
21:45:39.623545 indigo_server:   9207 SetControlDeviceB_Sony
21:45:39.623549 indigo_server:   9209 GetAllDevicePropData_Sony
[... 799 lines cut ...]
21:45:51.262597 indigo_server:   'IMAGE' // Image data
21:45:51.262599 indigo_server: }
21:45:51.262607 indigo_server: 20 ← { "defBLOBVector": { "version": 512, "device": "Sony Alpha A7S #0105", "name": "CCD_IMAGE", "group": "Image", "label": "Image data", "state": "Ok", "items": [  { "name": "IMAGE", "label": "Image data" } ] } }
21:45:51.262611 indigo_server: INDIGO Bus: property definition
21:45:51.262615 indigo_server: 'Sony Alpha A7S #0105'.'CCD_FITS_HEADERS' TEXT rw Ok 2.0  { // Custom FITS headers
21:45:51.262618 indigo_server:   'HEADER_1' = '' // Custom Header #1
21:45:51.262621 indigo_server:   'HEADER_2' = '' // Custom Header #2
21:45:51.262624 indigo_server:   'HEADER_3' = '' // Custom Header #3
21:45:51.262627 indigo_server:   'HEADER_4' = '' // Custom Header #4
21:45:51.262630 indigo_server:   'HEADER_5' = '' // Custom Header #5
21:45:51.262633 indigo_server:   'HEADER_6' = '' // Custom Header #6
21:45:51.262636 indigo_server:   'HEADER_7' = '' // Custom Header #7
[... cut ...]
21:45:51.262665 indigo_server: 'Sony Alpha A7S #0105'.'CONNECTION' SWITCH rw Ok 2.0 OneOfMany {
21:45:51.262668 indigo_server:   'CONNECTED' = On 
21:45:51.262671 indigo_server:   'DISCONNECTED' = Off 
21:45:51.262674 indigo_server: }
21:45:51.262682 indigo_server: 20 ← { "setSwitchVector": { "device": "Sony Alpha A7S #0105", "name": "CONNECTION", "state": "Ok", "items": [  { "name": "CONNECTED", "value": true }, { "name": "DISCONNECTED", "value": false } ] } }
21:45:51.262686 indigo_server: timer #0 done
```

polakovic, 2020-01-12:
I made a lot of changes this weekend. As it is it works reliable with A7RII and A5000. Pls. can you try it with your camera?

tail-feather, 2020-01-14:
Thanks for changes.

But still not working...

* Need to wait one second before connecting
* All properties are evaluated as `DSLR_COMPRESSION` at first time connection

I noticed `target->properties` is not `NULL` when first time connection.
The appropriate property name has been evaluated when the `dummy` initialize with zero like this:

```diff
@@ -327,7 +328,7 @@ char *ptp_property_sony_value_code_label(indigo_device *device, uint16_t propert
 uint8_t *ptp_sony_decode_property(uint8_t *source, indigo_device *device) {
        uint16_t code;
        source = ptp_decode_uint16(source, &code);
-       ptp_property dummy, *target = &dummy;
+       ptp_property dummy = {0}, *target = &dummy;
        for (int i = 0; PRIVATE_DATA->info_properties_supported[i]; i++) {
                if (PRIVATE_DATA->info_properties_supported[i] == code) {
                        target = PRIVATE_DATA->properties + i;
```

But, the properties are still not enumerated at the first time connection...

polakovic, 2020-01-14:
Sony firmware is a nightmare :( With A7RII I noticed, that to really restart the camera I need to remove the battery, turn it off is not enough. If you'll do it with A7S, is it still the same?

tail-feather, 2020-01-15:
No. It has never occurred to me.

A7S firmware version is 3.20.

By the way, if Err70 occurs with a Canon camera, it may not be possible to restart it until the battery is removed.
But SONY A7S does not.
I tried removing the battery just in case, but there was no change.

tail-feather, 2020-03-04:
Multiple attempts are no longer required by #291 !!

closed 2020-03-04 by tail-feather