indigo-astronomy/indigo #272: indigo_ccd_ptp: Cannot set image quality with RAW (SONY)

tail-feather, 2020-01-09:
`Standard`, `Fine` and `Extra Fine` is working well.
But `RAW` and `RAW + JPEG` is not working.

Standard:
```
20:17:27.400495 indigo_server: 20 → {"newSwitchVector":{"device":"Sony Alpha A7S #0106","name":"DSLR_COMPRESSION","items":[{"name":"2","value":true}]}}
20:17:27.400631 indigo_server: INDIGO Bus: property change request
20:17:27.400657 indigo_server: 'Sony Alpha A7S #0106'.'DSLR_COMPRESSION' SWITCH UNDEFINED Idle 2.0 UNDEFINED {
20:17:27.400678 indigo_server:   '2' = On 
20:17:27.400697 indigo_server: }
20:17:27.400772 indigo_server: timer #1 (of 3) used for 0s
20:17:27.400839 indigo_server: INDIGO Bus: property update
20:17:27.400867 indigo_server: 'Sony Alpha A7S #0106'.'DSLR_COMPRESSION' SWITCH rw Busy 2.0 OneOfMany {
20:17:27.400896 indigo_server:   '2' = On 
20:17:27.400924 indigo_server:   '3' = Off 
20:17:27.400949 indigo_server:   '4' = Off 
20:17:27.400976 indigo_server:   '10' = Off 
20:17:27.401003 indigo_server:   '13' = Off 
20:17:27.401024 indigo_server: }
20:17:27.401153 indigo_server: 20 ← { "setSwitchVector": { "device": "Sony Alpha A7S #0106", "name": "DSLR_COMPRESSION", "state": "Busy", "items": [  { "name": "2", "value": true }, { "name": "3", "value": false }, { "name": "4", "value": false }, { "name": "10", "value": false }, { "name": "13", "value": false } ] } }
20:17:27.401207 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request SetControlDeviceA_Sony (9205) 00000009 [00005004]
20:17:27.401448 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(16) -> OK
20:17:27.401516 indigo_server: indigo_ccd_ptp[890, ptp_transaction]: data 9205 00000009 +1 bytes
20:17:27.406367 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 12
20:17:27.406378 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: response OK (2001) 00000009 []
20:17:27.406382 indigo_server: INDIGO Bus: property update
20:17:27.406385 indigo_server: 'Sony Alpha A7S #0106'.'DSLR_COMPRESSION' SWITCH rw Ok 2.0 OneOfMany {
20:17:27.406388 indigo_server:   '2' = On 
20:17:27.406391 indigo_server:   '3' = Off 
20:17:27.406393 indigo_server:   '4' = Off 
20:17:27.406396 indigo_server:   '10' = Off 
20:17:27.406399 indigo_server:   '13' = Off 
20:17:27.406401 indigo_server: }
20:17:27.406458 indigo_server: 20 ← { "setSwitchVector": { "device": "Sony Alpha A7S #0106", "name": "DSLR_COMPRESSION", "state": "Ok", "items": [  { "name": "2", "value": true }, { "name": "3", "value": false }, { "name": "4", "value": false }, { "name": "10", "value": false }, { "name": "13", "value": false } ] } }
20:17:27.406463 indigo_server: timer #1 done
20:17:27.874543 indigo_server: indigo_ccd_ptp[ptp_get_event:1131]: libusb_bulk_transfer() -> OK, 16
20:17:27.874617 indigo_server: indigo_ccd_ptp[1137, ptp_get_event]: event PropertyChanged_Sony (c203) [0000d21d]
20:17:27.874639 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request GetAllDevicePropData_Sony (9209) 0000000a []
20:17:27.874876 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(12) -> OK
20:17:27.877638 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 1024
20:17:27.877695 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: data 9209 0000000a +1164 bytes
20:17:27.877913 indigo_server: indigo_ccd_ptp[ptp_transaction:937]: libusb_bulk_transfer() -> OK, 152
20:17:27.878253 indigo_server: indigo_ccd_ptp[ptp_transaction:952]: libusb_bulk_transfer() -> OK, 12
20:17:27.878310 indigo_server: indigo_ccd_ptp[962, ptp_transaction]: response OK (2001) 0000000a []
20:17:27.878343 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Compression (5004), type = uint8 -> DSLR_COMPRESSION (4)
20:17:27.878363 indigo_server: INDIGO Bus: property update
20:17:27.878384 indigo_server: 'Sony Alpha A7S #0106'.'DSLR_COMPRESSION' SWITCH rw Ok 2.0 OneOfMany {
20:17:27.878401 indigo_server:   '2' = Off 
20:17:27.878414 indigo_server:   '3' = Off 
20:17:27.878429 indigo_server:   '4' = On 
20:17:27.878447 indigo_server:   '10' = Off 
20:17:27.878464 indigo_server:   '13' = Off 
20:17:27.878479 indigo_server: }
20:17:27.878579 indigo_server: 20 ← { "setSwitchVector": { "device": "Sony Alpha A7S #0106", "name": "DSLR_COMPRESSION", "state": "Ok", "items": [  { "name": "2", "value": false }, { "name": "3", "value": false }, { "name": "4", "value": true }, { "name": "10", "value": false }, { "name": "13", "value": false } ] } }
20:17:27.878623 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: White balance (5005), type = uint16 -> DSLR_WHITE_BALANCE (4)
20:17:27.878656 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Aperture (5007), type = uint16 -> DSLR_APERTURE (2200)
20:17:27.878683 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Focus mode (500a), type = uint16 -> DSLR_FOCUS_MODE (1)
20:17:27.878754 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Metering mode (500b), type = uint16 -> NONE (1)
20:17:27.878787 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Flash mode (500c), type = uint16 -> DSLR_FLASH_MODE (0)
20:17:27.878823 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Program mode (500e), type = uint16 -> DSLR_PROGRAM (1)
20:17:27.878851 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Exposure compensation (5010), type = int16 -> DSLR_EXPOSURE_COMPENSATION (0)
20:17:27.878895 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Capture mode (5013), type = uint16 -> DSLR_CAPTURE_MODE (1)
20:17:27.878988 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: DPCCompensation_Sony (d200), type = int16 -> NONE (0)
20:17:27.879034 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: DRangeOptimize_Sony (d201), type = uint8 -> NONE (1)
20:17:27.879058 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Image size (d203), type = uint8 -> DSLR_IMAGE_SIZE (1)
20:17:27.879085 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: Shutter speed (d20d), type = uint32 -> DSLR_SHUTTER (524298)
20:17:27.879159 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: d20e (d20e), type = uint8 -> NONE (7)
20:17:27.879183 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: ColorTemp_Sony (d20f), type = uint16 -> NONE (0)
20:17:27.879202 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: CCFilter_Sony (d210), type = uint8 -> NONE (128)
20:17:27.879220 indigo_server: indigo_ccd_ptp[ptp_update_property:1113]: ABFilter_Sony (d21c), type = uint8 -> NONE (128)
[... 121 lines cut ...]
20:21:44.868458 indigo_server: INDIGO Bus: property update
20:21:44.868478 indigo_server: 'Sony Alpha A7S #0106'.'DSLR_COMPRESSION' SWITCH rw Busy 2.0 OneOfMany {
20:21:44.868493 indigo_server:   '2' = Off 
20:21:44.868510 indigo_server:   '3' = Off 
20:21:44.868530 indigo_server:   '4' = Off 
20:21:44.868556 indigo_server:   '10' = Off 
20:21:44.868580 indigo_server:   '13' = On 
20:21:44.868605 indigo_server: }
20:21:44.868723 indigo_server: 20 ← { "setSwitchVector": { "device": "Sony Alpha A7S #0106", "name": "DSLR_COMPRESSION", "state": "Busy", "items": [  { "name": "2", "value": false }, { "name": "3", "value": false }, { "name": "4", "value": false }, { "name": "10", "value": false }, { "name": "13", "value": true } ] } }
20:21:44.868780 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request SetControlDeviceA_Sony (9205) 00000016 [00005004]
20:21:44.869020 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(16) -> OK
20:21:44.869077 indigo_server: indigo_ccd_ptp[890, ptp_transaction]: data 9205 00000016 +1 bytes
20:21:44.871850 indigo_server: indigo_ccd_ptp[ptp_transaction:914]: libusb_bulk_transfer() -> OK, 12
20:21:44.871908 indigo_server: indigo_ccd_ptp[924, ptp_transaction]: response OK (2001) 00000016 []
20:21:44.871927 indigo_server: INDIGO Bus: property update
20:21:44.871948 indigo_server: 'Sony Alpha A7S #0106'.'DSLR_COMPRESSION' SWITCH rw Ok 2.0 OneOfMany {
20:21:44.871965 indigo_server:   '2' = Off 
20:21:44.871981 indigo_server:   '3' = Off 
20:21:44.871994 indigo_server:   '4' = Off 
20:21:44.872009 indigo_server:   '10' = Off 
20:21:44.872022 indigo_server:   '13' = On 
20:21:44.872037 indigo_server: }
20:21:44.872116 indigo_server: 20 ← { "setSwitchVector": { "device": "Sony Alpha A7S #0106", "name": "DSLR_COMPRESSION", "state": "Ok", "items": [  { "name": "2", "value": false }, { "name": "3", "value": false }, { "name": "4", "value": false }, { "name": "10", "value": false }, { "name": "13", "value": true } ] } }
20:21:44.872144 indigo_server: timer #1 done
```

polakovic, 2020-01-09:
I'll try if I can reproduce it, I have 7RII.

polakovic, 2020-01-12:
Pls. can you try now?

tail-feather, 2020-01-14:
It worked! Thanks!

closed 2020-01-14 by tail-feather