indigo-astronomy/indigo #273: Why does the mount agent have LX200 stuff in it?

ghulands, 2020-01-10:
Hi,
I found it a little confusing why the mount agent has the two LX200 sections in it?

[image]

As you can see I don't have the LX200 mount selected.
[image]

Is this a bug?

rumengb, 2020-01-10:
Hi, a lot of the existing software can speak Lx200 protocol out of the box.
Got that reason nount agent can accept lx200 and translate it to Indigo and
pass it to the native driver. This way every lx200 capable software can
control every mount.

ghulands, 2020-01-10:
Thanks for the explanation.

closed 2020-01-10 by ghulands