indigo-astronomy/indigo #274: FLI CCD camera can not reconnect after first connection

danibeni, 2020-01-14:
I have been working with FLI CCD driver to include additional keywords in FITS header for a concrete project. After some tests I have realized that I can connect to the FLI CCD camera for the first time but the connection failed in next connections attemps saying that "indigo_try_global_lock(): failed to get lock.".

After some research I have found that in the **ccd_change_property** function the device->is_connected parameter is set to False before releasing the FLI resources using fli_close() function:

https://github.com/indigo-astronomy/indigo/blob/ec3ad95dc22148245dfbf0bbf7612f7f7e7f225d/indigo_drivers/ccd_fli/indigo_ccd_fli.c#L772

I think it is enough to put that line after the doing fli_close(device):

https://github.com/indigo-astronomy/indigo/blob/ec3ad95dc22148245dfbf0bbf7612f7f7e7f225d/indigo_drivers/ccd_fli/indigo_ccd_fli.c#L777

This way the resources are releases and unlocked properly before setting the is_connected parameter to False.

rumengb, 2020-01-14:
Yes you are right! Thanks I will fix it today.

rumengb, 2020-01-14:
Fixed, Can you try it now?

danibeni, 2020-01-14:
Ok, now it seems to work properly. Thank you

closed 2020-01-14 by danibeni

rumengb, 2020-01-14:
What project is that? Can you give us some more information?

danibeni, 2020-01-14:
We are using Indigo running on a Raspberry Pi 3 to control a ProLine FLI camera installed on a Schmidt telescope, but additional FITS keywords for the exposure start and end times are needed, so I have modified the FLI CCD driver to include this times for each image taken. 

We are also working in the integration of the Schmidt based on Indigo to create an Astronomy Observation Suite, including drivers for our Schmiddt Telescope, dome and focus, and clients for the user interaction with the system and a scheduler, If you consider it interesting, I can give you more details when the project is in a more advanced state

rumengb, 2020-01-14:
This is really interesting. I just added Indigo client programming guide to
the repo. You may find it interesting.

 Is this a professional or amateur project? Schmidt telescope sounds
professional :) unless you mean schmidt-cassegrain... What observatory?

rumengb, 2020-01-14:
Is that 0.8m Calar Alto Schmidt?

danibeni, 2020-01-14:
Correct!!! We are renewing the hardware and software related to the 0.8m CAHA Schmidt telescope and we are testing Indigo as base for the software control subsystem.

rumengb, 2020-01-14:
Great! Our 2m RCC telescope at Rozhen works with indigo and several Andor
cameras and with FLI on several 0.6m smaller telescopes... Also Several
other observatories use it. I really want to establish Indigo as a
professional framework. So if you need any support do not hesitate to
contact us. And we really need what you are doing.. Astronomy Observation
Suite...

rumengb, 2020-03-25:
Hi @danibeni ,
I have sent you and answer to your e mail a week ago and now i got a reply that the mail has beed reused by your mail server. Please contact me via some other email. So that we can discuss the things you are interested in.