indigo-astronomy/indigo #276: indigo_ccd_ptp: Cannot connect Nikon D5300

tail-feather, 2020-01-18:
I tried to connect to Nikon D5300 using only ccd_ptp driver.
But, OpenSession has timed out.

If I used only the ccd_gphoto2 driver, it worked.
(Physical connection seems to be fine.)

```
23:00:52.912690 indigo_server: indigo_ccd_ptp: 'Nikon D5300 #0103' attached
23:00:52.913586 indigo_server: Can't open /home/tsunoda/.indigo/Nikon_D5300_#0103.config (No such file or directory)
23:00:59.958720 indigo_server: indigo_ccd_ptp[ptp_open:783]: libusb_get_device_descriptor() -> OK
23:00:59.958792 indigo_server: indigo_ccd_ptp[ptp_open:786]: libusb_open() -> OK
23:00:59.958819 indigo_server: indigo_ccd_ptp[ptp_open:797]: libusb_get_config_descriptor(0) -> OK
23:00:59.958828 indigo_server: indigo_ccd_ptp[ptp_open:803]: PTP CONFIG = 1 IFACE = 0
23:00:59.959171 indigo_server: indigo_ccd_ptp[ptp_open:816]: libusb_set_configuration(1) -> OK
23:00:59.959236 indigo_server: indigo_ccd_ptp[ptp_open:821]: libusb_claim_interface(0) -> OK
23:00:59.959950 indigo_server: indigo_ccd_ptp[ptp_open:848]: PTP EP OUT = 02 IN = 81 INT = 83
23:00:59.959982 indigo_server: indigo_ccd_ptp[872, ptp_transaction]: request OpenSession (1002) 00000000 [00000001]
23:01:09.960116 indigo_server: indigo_ccd_ptp[ptp_transaction:874]: libusb_bulk_transfer(0) -> LIBUSB_ERROR_TIMEOUT
23:01:09.960559 indigo_server: indigo_ccd_ptp[ptp_transaction:877]: libusb_clear_halt() -> OK
23:01:19.960764 indigo_server: indigo_ccd_ptp[ptp_transaction:879, 0x7f8bf37fe700]: libusb_bulk_transfer(0) -> LIBUSB_ERROR_TIMEOUT
23:01:19.960833 indigo_server: indigo_ccd_ptp[ptp_transaction:882, 0x7f8bf37fe700]: Failed to send request -> LIBUSB_ERROR_TIMEOUT
23:01:19.960995 indigo_server: indigo_ccd_ptp[ptp_close:982]: libusb_close()
```

tail-feather, 2020-01-18:
Oops, I found evil gvfsd-gphoto2...
I'll remove and connect again.

polakovic, 2020-01-18:
Yep, you can't mix gphoto and ptp :(

tail-feather, 2020-01-18:
I removed `gvfs-backends` and restarted, but it still happens.
Is there anything else I should do?

rumengb, 2020-01-18:
There is a note in the ptp Readme.

tail-feather, 2020-01-19:
Yeah, I removed `gvfs-backends` according to the Readme note, but it is still happening.

`gvfs-backends` includes `gvfs-gphoto2-volume-monitor` and `gvfsd-gphoto2`.
see also: https://packages.ubuntu.com/ja/bionic-updates/amd64/gvfs-backends/filelist

tail-feather, 2020-01-20:
D5100 connects, but D5300 does not.
If you have a D5300, can you try the connection?

polakovic, 2020-01-20:
I have 5100 and 5600, but will retest. Pls. make sure you turned WiFi off, I'm not quite sure about this camera, but it often prevents from USB connection.

tail-feather, 2020-01-20:
Thanks reply.
I checked the Wi-Fi settings, but it was already off...

I can operate correctly with SDK or gPhoto2, but OpenSession fails only when I operate with ccd_ptp.

polakovic, 2020-01-21:
I tested PTP driver on macOS with D5600, but it worked. I'll check it also on Linux tomorrow.

polakovic, 2020-02-02:
We have another report that D5300 doesn't work the same way on macOS. It looks, that the issue is related to D5300 and independent on operating system.

polakovic, 2020-02-23:
Nothing new on this issue :( I made a lot of experiments with 5600, but it always works. Even if WiFi is turned on, it is automatically disabled when the camera is connected to USB.

Maybe it is less automatic on 5300. Does it have airplane mode?

tail-feather, 2020-02-25:
> Maybe it is less automatic on 5300.

When the D5300 is connected to USB, Wi-Fi is automatically disabled as well as D5600.

> Does it have airplane mode?

No, D5300 does not have airplane mode.

tail-feather, 2020-02-25:
I got the USB log using VMware.

Since there is no difference in OpenSession (`10 00 00 00 01 00 02 10 00 00 00 00 01 00 00 00`), there may be a problem with the communication before that.

----------

ccd_gphoto2:

```
Date:2020-02-25T21:01:28.621+09:00 vmx: I125: USBIO:
	GetDescriptor(device, 0)
Date:2020-02-25T21:01:28.622+09:00 vmx: I125: USBIO:
	Down dev=5 'usb:2' endpt=0 stream=0 datalen=18 numPackets=0 status=0 0
	 000: 80 06 00 01 00 00 12 00                         ........        
Date:2020-02-25T21:01:28.622+09:00 vmx: I125: USBIO:
	Up dev=5 'usb:2' endpt=0 stream=0 datalen=18 numPackets=0 status=0 0
	 000: 80 06 00 01 00 00 12 00                         ........        
	 000: 12 01 00 02 00 00 00 40 b0 04 31 04 00 01 01 02 .......@..1.....
	 010: 03 01                                           ..              
Date:2020-02-25T21:01:28.623+09:00 vmx: I125: USBIO:
	GetDescriptor(config, 0)
Date:2020-02-25T21:01:28.623+09:00 vmx: I125: USBIO:
	Down dev=5 'usb:2' endpt=0 stream=0 datalen=9 numPackets=0 status=0 0
	 000: 80 06 00 02 00 00 09 00                         ........        
Date:2020-02-25T21:01:28.623+09:00 vmx: I125: USBIO:
	Up dev=5 'usb:2' endpt=0 stream=0 datalen=9 numPackets=0 status=0 0
	 000: 80 06 00 02 00 00 09 00                         ........        
	 000: 09 02 27 00 01 01 00 c0 01                      ..'......       
Date:2020-02-25T21:01:28.624+09:00 vmx: I125: USBIO:
	GetDescriptor(config, 0)
Date:2020-02-25T21:01:28.624+09:00 vmx: I125: USBIO:
	Down dev=5 'usb:2' endpt=0 stream=0 datalen=39 numPackets=0 status=0 0
	 000: 80 06 00 02 00 00 27 00                         ......'.        
Date:2020-02-25T21:01:28.624+09:00 vmx: I125: USBIO:
	Up dev=5 'usb:2' endpt=0 stream=0 datalen=39 numPackets=0 status=0 0
	 000: 80 06 00 02 00 00 27 00                         ......'.        
	 000: 09 02 27 00 01 01 00 c0 01 09 04 00 00 03 06 01 ..'.............
	 010: 01 00 07 05 81 02 00 02 00 07 05 02 02 00 02 00 ................
	 020: 07 05 83 03 08 00 09                            .......         
Date:2020-02-25T21:01:28.625+09:00 vmx: I125: USBIO:
	GetDescriptor(string, 0)
Date:2020-02-25T21:01:28.625+09:00 vmx: I125: USBIO:
	Down dev=5 'usb:2' endpt=0 stream=0 datalen=255 numPackets=0 status=0 0
	 000: 80 06 00 03 00 00 ff 00                         ........        
Date:2020-02-25T21:01:28.625+09:00/2020-02-25T21:01:28.626+09:00 vmx: I125: USBIO:
	Up dev=5 'usb:2' endpt=0 stream=0 datalen=4 numPackets=0 status=0 0
	 000: 80 06 00 03 00 00 ff 00                         ........        
	 000: 04 03 09 04                                     ....            
Date:2020-02-25T21:01:28.627+09:00 vmx: I125: USBIO:
	GetDescriptor(string, 2, langId=0x0409)
Date:2020-02-25T21:01:28.627+09:00 vmx: I125: USBIO:
	Down dev=5 'usb:2' endpt=0 stream=0 datalen=255 numPackets=0 status=0 0
	 000: 80 06 02 03 09 04 ff 00                         ........        
Date:2020-02-25T21:01:28.628+09:00 vmx: I125: USBIO:
	Up dev=5 'usb:2' endpt=0 stream=0 datalen=32 numPackets=0 status=0 0
	 000: 80 06 02 03 09 04 ff 00                         ........        
	 000: 20 03 4e 00 49 00 4b 00 4f 00 4e 00 20 00 44 00  .N.I.K.O.N. .D.
	 010: 53 00 43 00 20 00 44 00 35 00 33 00 30 00 30 00 S.C. .D.5.3.0.0.
Date:2020-02-25T21:01:28.630+09:00 vmx: I125: USBIO:
	GetDescriptor(string, 1, langId=0x0409)
Date:2020-02-25T21:01:28.630+09:00 vmx: I125: USBIO:
	Down dev=5 'usb:2' endpt=0 stream=0 datalen=255 numPackets=0 status=0 0
	 000: 80 06 01 03 09 04 ff 00                         ........        
Date:2020-02-25T21:01:28.630+09:00 vmx: I125: USBIO:
	Up dev=5 'usb:2' endpt=0 stream=0 datalen=12 numPackets=0 status=0 0
	 000: 80 06 01 03 09 04 ff 00                         ........        
	 000: 0c 03 4e 00 49 00 4b 00 4f 00 4e 00             ..N.I.K.O.N.    
Date:2020-02-25T21:01:28.632+09:00 vmx: I125: USBIO:
	GetDescriptor(string, 3, langId=0x0409)
Date:2020-02-25T21:01:28.632+09:00 vmx: I125: USBIO:
[... 180 lines cut ...]
	Up dev=4 'usb:2' endpt=0 stream=0 datalen=0 numPackets=0 status=0 0
	 000: 02 01 00 00 02 00 00 00                         ........        
Date:2020-02-25T21:05:35.030+09:00 vmx: I125: USBIO:
	ClearFeature(feature=0, endpoint=131)
Date:2020-02-25T21:05:35.030+09:00 vmx: I125: USBIO:
	Down dev=4 'usb:2' endpt=0 stream=0 datalen=0 numPackets=0 status=0 0
	 000: 02 01 00 00 83 00 00 00                         ........        
Date:2020-02-25T21:05:35.030+09:00 vmx: I125: USBIO:
	Up dev=4 'usb:2' endpt=0 stream=0 datalen=0 numPackets=0 status=0 0
	 000: 02 01 00 00 83 00 00 00                         ........        
Date:2020-02-25T21:05:35.032+09:00 vmx: I125: USBIO:
	Down dev=4 'usb:2' endpt=2 stream=0 datalen=16 numPackets=0 status=0 0
	 000: 10 00 00 00 01 00 02 10 00 00 00 00 01 00 00 00 ................
Date:2020-02-25T21:05:46.374+09:00 vmx: I125: USBIO:
	ClearFeature(feature=0, endpoint=2)
Date:2020-02-25T21:05:46.374+09:00 vmx: I125: USBIO:
	Down dev=4 'usb:2' endpt=0 stream=0 datalen=0 numPackets=0 status=0 0
	 000: 02 01 00 00 02 00 00 00                         ........        
Date:2020-02-25T21:05:46.375+09:00 vmx: I125: USBIO:
	Up dev=4 'usb:2' endpt=0 stream=0 datalen=0 numPackets=0 status=0 0
	 000: 02 01 00 00 02 00 00 00                         ........        
Date:2020-02-25T21:05:46.376+09:00 vmx: I125: USBIO:
	Down dev=4 'usb:2' endpt=2 stream=0 datalen=16 numPackets=0 status=0 0
	 000: 10 00 00 00 01 00 02 10 00 00 00 00 01 00 00 00 ................
```

polakovic, 2020-02-25:
It seems, that the difference is that gphoto doesn't call libusb_set_configuration. It is in indigo_ptp.c at lines 813 - 817. Pls. can you try to comment it out and test again? Thank you!!!

tail-feather, 2020-02-26:
In gPhoto2 logs, I noticed that `ClearFeature` was not called after `SetConfiguration`.
When I commented out `libusb_clear_halt`, I was able to connect to D5300!

polakovic, 2020-02-26:
Great, I think, we can omit it! Thank you!

polakovic, 2020-02-27:
It is confirmed, it works! A good job, thank you!

closed 2020-02-28 by tail-feather