indigo-astronomy/indigo #303: Moonlite Focuser not connecting

ghulands, 2020-03-24:
Hi,
I have a WR35 moonlite and I see in the logs when I attach it, the following:

`06:04:29.997267 indigo_server: indigo_focuser_moonlite[focuser_connection_handler:232, 0xa22f93d0]: MoonLite focuser not detected`
`06:04:29.999514 indigo_server: indigo_focuser_moonlite[focuser_connection_handler:262, 0xa22f93d0]: Failed to connect to /dev/ttyUSB3`
`06:07:38.980925 indigo_server: indigo_aux_usbdp[usbdp_status:260, 0xa10fe3d0]: Error: parsed 15 of 16 values in response "##20.12/19.87/21.48/49.54/10.44/0/0/0/0/0/0/2/2/0/04**"`
`06:11:25.699334 indigo_server: indigo_aux_usbdp[usbdp_status:260, 0xa10fe3d0]: Error: parsed 15 of 16 values in response "##20.18/20.00/21.62/49.37/10.52/00/0/0/0/0/2/2/0/0/4**"`

Any ideas? 

 _____ _   _ _____ _____ _____  ____     indigo@indigosky
|_   _| \ | |  __ \_   _/ ____|/ __ \    ----------------
  | | |  \| | |  | || || |  __| |  | |   INDIGO version: 2.0-118
  | | | . ` | |  | || || | |_ | |  | |   INDIGO running: Yes
 _| |_| |\  | |__| || || |__| | |__| |   OS: Raspbian GNU/Linux 10 (buster) armv7l
|_____|_| \_|_____/_____\_____|\____/    Model: Raspberry Pi 4 Model B Rev 1.1
  _____ _                                Kernel: 4.19.93-v7l+
 / ____| |                               Uptime: 41 mins
| (___ | | ___   _                       Packages: 644 (dpkg)
 \___ \| |/ / | | |                      Shell: bash 5.0.3
 ____) |   <| |_| |                      Terminal: /dev/pts/0
|_____/|_|\_\__, |                       CPU: BCM2835 (4) @ 1.500GHz
              __/ |                      Memory: 169MiB / 3906MiB
             |___/                       Disk (/): 2.0G / 459G (1%)
                                         Local IP: 192.168.11.110

polakovic, 2020-03-24:
Are you sure you used the right serial port for moonlite focuser? This error means, that the device didn't reply to :GV# command.

Rumen, it looks, you have wrong format specifiers in aux_usbdp.

[image]

ghulands, 2020-03-24:
i uplugged it, plugged it back in, ran dmesg and got the /dev/tty device that was setup. I also tried running `screen /dev/ttyUSB0 9600` to run the :GV# command but the device never responded.

polakovic, 2020-03-24:
Can you execute commands

lsusb -v
ls -l /dev/ttyUSB0

and copy & paste the result?

rumengb, 2020-03-24:
> Are you sure you used the right serial port for moonlite focuser? This error means, that the device didn't reply to :GV# command.
> 
> Rumen, it looks, you have wrong format specifiers in aux_usbdp.
> 
> [image]

Fixed!

ghulands, 2020-03-25:
I had to send the unit back as I dropped it when packing up the scope for the night. I'll report back in a couple of days. I asked moonlight to check the USB as well. I tried it on my mac as well and was unable to get it to respond.

polakovic, 2020-03-25:
OK, let's wait, maybe it is a faulty unit.

polakovic, 2020-08-10:
Anything new?

polakovic, 2020-11-17:
I'll close it for now, nobody else reported this kind of issue, maybe really faulty unit...

closed 2020-11-17 by polakovic