indigo-astronomy/indigo #311: Dependency to 'GLIBC_2.28' since 2.0-112

kkretzschmar, 2020-04-02:
Hi,
since version 2.0-112 the libindigo.so library seems to be built with a new compiler which requires GLIBC with version 2.28. This is too new for users of the current ubuntu 18.04 LTS version, which has GLIBC 2.27 installed. I am using the newest ubuntu mate image for my raspberry pi and cannot upgrade to indigo versions >= 2.0-112 anymore. Are you aware of that and was this change to GLIBC_2.28 required on your side?

Thanks,
Klaus

polakovic, 2020-04-02:
Hi Klaus, it is requirement of Atik SDK on ARM :( If you don't need it, you can use older glibc.

kkretzschmar, 2020-04-02:
Hi Peter,
I need the ATIK SDK, so I am stuck on 2.0-112, which is quite bad. I'd suggest to upgrade to the new ATIK SDK, if the LTS OS linux distributions support it. Or ask ATIK to patch the older SDK if there is a major bug in it ... What do you think?
Klaus

polakovic, 2020-04-02:
The recent SDK is buggy too, I reported it to Atik already, but no new build yet :(

INDIGO Sky is buster based, so there is new glibc already.

kkretzschmar, 2020-04-02:
Ah, ok. Maybe can you ask them to provide the SDK on ARM with GLIBC 2.27 as well? Maybe they introduced the GLIBC 2.28 dependency accidently?

polakovic, 2020-04-02:
The problem is that they build on physical RPi4 with buster :(

The older version doesn't work properly on RPi, there is a problem with system buffer size and horizontal lines are visible in the images.

kkretzschmar, 2020-04-02:
Oh no, why don't they cross-compile arm binaries with clang? 

My problem is that my ATIK camera worked well also with the old ATIK SDK since many years. Now I am cut off from any improvements/fixes on indigo which I am not happy about.

Would it be possible to provide the old ATIK SDK so that I (or other users which got stuck) can build my own indigo binaries for my ubuntu lts 18.04?

polakovic, 2020-04-02:
It is not so easy :( Original SDK was developed by us and has quite different API than the new one developed by Atik. So it is actually completely different driver. But I'll try to stress it to Atik guys that we need to build it with GLIBC 2.27.

kkretzschmar, 2020-04-02:
Ok I understand. Thanks for asking them, I think I am not the only one who is running into this problem ...

kkretzschmar, 2020-05-09:
Hi Peter,
did you already get feedback from the ATIK developers?

Thanks,
Klaus

polakovic, 2020-05-10:
Hi Klaus,

no reply yet, but I noticed, there is a new SDK released last week. I'll
try it and if it will work, I'll let you know.

Peter

polakovic, 2020-05-10:
Hi Klaus,

I briefly tested the new SDK, but as far as I can say it doesn't work at all (on macOS). Doesn't capture correctly with Horizon and doesn't even connect with Titan :(

I'll report it to Atik guys...

Peter

kkretzschmar, 2020-05-10:
Ouch ;), I'll stay on the current version, its ok. Thanks for testing ...

closed 2021-11-25 by rumengb