indigo-astronomy/indigo #312: Please add manual filter wheel or filter drawer

tlkw79, 2020-04-05:
Hi,

Please add manual filter wheel or filter drawer. Thanks

rumengb, 2020-04-05:
Hi,
i am not sure I understand, how can this be controlled by the computer if it is manual?

Rumen

polakovic, 2020-04-05:
I do, the reason is to copy filter name metadata to fits files. It is on my TODO list, but not implemented yet.

rumengb, 2020-04-05:
In that case wheel simulator can be used isn't it? at least untill we have dedicated drover for that. It should be easy to implement...

polakovic, 2020-04-05:
Yes, you're right. Simulator can do the job.

rumengb, 2020-04-30:
Created! Will this do the job?

[image]

closed 2020-04-30 by rumengb

tlkw79, 2020-05-01:
how and where I can get it?

rumengb, 2020-05-02:
It will be in the next Indigo release.