indigo-astronomy/indigo #321: INDI Library including all 3rd party drivers

tlkw79, 2020-04-16:
Hi,

I am new in INDIGO server. Just curious how to add INDI Library including all 3rd party drivers into INDIGO device selection list?? I already add all of them using repo via SSH. But none of the  INDI Library including all 3rd party drivers are visible in the list. 

Please advice.

Tom

polakovic, 2020-04-16:
You have to add them on a server command line with "-i driver_executable" switch. Nevertheless, the compatibility is not maintained anymore. What driver do you want to use?

rumengb, 2020-04-16:
I would like to make a clarification here. The drivers will work, but indi
and indigo are using different device properties, we tried to maintain
backward compatibility with INDI, but the problem is that INDI itself does
not maintain compatibility with previous versions, which makes it
impossible for us to track and maintain this so called property mapping.
This will result in applications not finding the properties they relay on.
However with the control panel (property browser) everything will continue
to work... only applications that relay on some properties that INDI
changed recently may not work.

Rumen

tlkw79, 2020-04-16:
> You have to add them on a server command line with "-i driver_executable" switch. Nevertheless, the compatibility is not maintained anymore. What driver do you want to use?

How to use the switch command in Indigo Server? Let say i wanted to try on indi manual filter.

Thanks

rumengb, 2020-04-16:
Hi you should start indigo_server from the command line like this:

Indigo_server -i indi_driver

Just replace indi_driver with the real driver name and make sure no other
Indigo server is running.

But it is better to wait a bit and I will make a manual filter wheel
driver.

Rumen

polakovic, 2020-04-16:
Manual filterwheel is definitely not the reason to downgrade to INDI :) You can use "CCD Simulator (wheel)" device with the same effect.

tlkw79, 2020-04-17:
> Hi you should start indigo_server from the command line like this: Indigo_server -i indi_driver Just replace indi_driver with the real driver name and make sure no other Indigo server is running. But it is better to wait a bit and I will make a manual filter wheel driver. Rumen
> […](#)

Thank you for that.

tlkw79, 2020-04-17:
> Manual filterwheel is definitely not the reason to downgrade to INDI :) You can use "CCD Simulator (wheel)" device with the same effect.

I Will test that later. I hope it works too. :)

19ben81, 2020-04-26:
Hello there,
not only that I'm new to Indigo, I'm also completely new to linux. I've just started working with (l)ubuntu.
So far I have installed Indigo and now I'm trying to make the drivers work. I have a ASI1600mm camera. Therefor I have tried to install the ccd_asi driver from the command line via the command
"Indigo_server -i ccd_asi" and also via 
"Indigo_server -i indigo_ccd_asi"
just typed in from my home folder, but I always get the message, that the command has not been found. Do I have to save the files on a local drive first (and if so, how?) or will this command get them directly from github?
Pardon me for my stupid questions and thanks for helping!
Ben

polakovic, 2020-04-26:
Hi, it seems, that you used upper case "I" at the beginning. Just start indigo_server (without any arguments), connect to http://localhost:7624 and enable ZWO CCD driver (replace localhost with host name running the server). Peter

closed 2020-06-20 by rumengb