indigo-astronomy/indigo #322: ZWO EFW not detected properly

docbobo, 2020-04-17:
I was just trying out indigo as alternative to indi, and ran into an issue with the ZWO EFW that came with my ASI1600MM - it's not detected until it's being replugged into the system. 

The issue is very similar to what's been described in https://github.com/indigo-astronomy/indigo/tree/master/indigo_drivers/focuser_asi - just with the difference that everything's deployed on an RPi4.

docbobo, 2020-04-17:
BTW: It does work without any problems via INDI. Not sure this is a permission problem as stated in the mentioned README.

docbobo, 2020-04-17:
Just noticed https://github.com/indigo-astronomy/indigo/blob/master/indigo_drivers/wheel_asi/bin_externals/libEFWFilter/lib/99-efw.rules as part of the driver subtree. This file is not present on  my machine, even though I did a fresh install of indigo yesterday. Will ad that manually to see how the system behaves.

docbobo, 2020-04-17:
Adding that rule to the system improves things quite a bit. I also noticed that the similar file for the EAF is also missing on my system. I suspect there might be something wrong with the build, resulting in those files not to be included in the packages?

rumengb, 2020-04-17:
What platform are you using? Raspberry pi?
what linux... what version of indigo?

rumengb, 2020-04-17:
I checked my RPI4 with indigosky... it works fine... and the files are there. Do you install it from deb?

[image]

docbobo, 2020-04-17:
Hmm.

> pi@AstroPi4:~ $ cat /etc/apt/sources.list.d/indigo.list
> deb [trusted=yes] https://indigo-astronomy.github.io/indigo_ppa/ppa indigo main

> pi@AstroPi4:~ $ apt list --installed | grep indigo
> 
> WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
> 
> indigo-control-panel/indigo,now 0.7-0 armhf [installed]
> indigo/indigo,now 2.0-120 armhf [installed]

Did I take the wrong turn somewhere?

rumengb, 2020-04-17:
Seems ok...
Can you please verify that the files i showed above are there as on the screenshot?
Also did you load the drivers for those devices?

And also do you have indi installed?

docbobo, 2020-04-17:
Yes, I do have indi installed on the same machine right now. Having some compatibility problems (just for reference: https://bugs.kde.org/show_bug.cgi?id=420187).

Here's the list of files:

> total 104
>  1336  4 drwxr-xr-x 2 root root  4096 Apr 17 08:26 .
>   161  4 drwxr-xr-x 4 root root  4096 Apr  6 12:41 ..
>  6873  4 -rw-r--r-- 1 root root  1001 Sep 17  2019 51-sbig-debian.rules
> 21587  4 -rw-r--r-- 1 root root  3623 Aug 27  2019 80-theimagingsource-cameras.rules
>  6188 20 -rw-r--r-- 1 root root 16647 Apr  6 14:41 85-qhyccd.rules
>  7173  4 -rw-r--r-- 1 root root   538 Apr  6 15:28 95-pentax.rules
>  7175  4 -rw-r--r-- 1 root root   470 Apr  6 15:28 95-samsung.rules
>  7809  4 -rw-r--r-- 1 root root   373 Sep 17  2019 99-altaircam.rules
>  3465  4 -rw-r--r-- 1 root root   187 Sep 17  2019 99-apogee.rules
>  3788  4 -rw-r--r-- 1 root root   213 Apr  6 12:10 99-asi.rules
>  3812  4 -rw-r--r-- 1 root root   891 Apr  6 14:41 99-atik.rules
> 17928  4 -rw-r--r-- 1 root root  1127 Jul 30  2019 99-com.rules
>  2929  4 -rw-r--r-- 1 root root   109 Apr 17 08:26 99-efw.rules
>  3832  4 -rw-r--r-- 1 root root   164 Sep 17  2019 99-fishcamp.rules
>  4196  4 -rw-r--r-- 1 root root   214 Apr  6 14:41 99-fli.rules
>  6185  4 -rw-r--r-- 1 root root   305 Sep 17  2019 99-inovaplx.rules
>  3749  4 -rw-r--r-- 1 root root   649 Sep 17  2019 99-meadedsi.rules
>  6161  4 -rw-r--r-- 1 root root   194 Apr  6 14:41 99-miccd.rules
>  7814  4 -rw-r--r-- 1 root root   405 Sep 17  2019 99-nncam.rules
>  7167  4 -rw-r--r-- 1 root root   212 Apr  6 14:41 99-pentax.rules
>  7811  4 -rw-r--r-- 1 root root   405 Sep 17  2019 99-starshootg.rules
>  7804  4 -rw-r--r-- 1 root root   198 Sep 17  2019 99-toupcam.rules

Judging from the timestamps, none of them were installed by INDIGO. The ```99-efw.rules```I created myself by copying from git.

docbobo, 2020-04-17:
And here's the output from apt back when I installed:

> Log started: 2020-04-16  21:46:39
> Selecting previously unselected package libavahi-compat-libdnssd1:armhf.
> (Reading database ... 266638 files and directories currently installed.)
> Preparing to unpack .../libavahi-compat-libdnssd1_0.7-4+b1_armhf.deb ...
> Unpacking libavahi-compat-libdnssd1:armhf (0.7-4+b1) ...
> Selecting previously unselected package indigo.
> Preparing to unpack .../indigo_2.0-120_armhf.deb ...
> Remove pre-2.0-76 build if any
> Unpacking indigo (2.0-120) ...
> Replacing files in old package indi-dsi (0.2-2) ...
> Replacing files in old package libqhy (20.2.19) ...
> Replacing files in old package libsbig (4.9.9-2) ...
> Setting up libavahi-compat-libdnssd1:armhf (0.7-4+b1) ...
> Setting up indigo (2.0-120) ...
> Created symlink /etc/systemd/system/multi-user.target.wants/indigo-environment.service → /lib/systemd/system/indigo-environment.service.
> no patch applied because no INDIGO Sky installation
> Processing triggers for libc-bin (2.28-10+rpi1) ...
> ldconfig: file /usr/lib/arm-linux-gnueabihf/libsbig.so.4.9.9 is truncated
> 
> ldconfig: file /usr/lib/arm-linux-gnueabihf/libsbig.so.4 is truncated
> 
> ldconfig: file /usr/lib/arm-linux-gnueabihf/libsbig.so is truncated
> 
> Log ended: 2020-04-16  21:46:57
> 
> Log started: 2020-04-16  21:47:05
> Selecting previously unselected package indigo-control-panel.
> (Reading database ... 267263 files and directories currently installed.)
> Preparing to unpack .../indigo-control-panel_0.7-0_armhf.deb ...
> Unpacking indigo-control-panel (0.7-0) ...
> Setting up indigo-control-panel (0.7-0) ...
> Processing triggers for mime-support (3.62) ...
> Processing triggers for hicolor-icon-theme (0.17-2) ...
> Processing triggers for gnome-menus (3.31.4-3) ...
> Processing triggers for desktop-file-utils (0.23-4) ...
> Log ended: 2020-04-16  21:47:07

rumengb, 2020-04-17:
Can you try to install indigo on a clean installation I mean without INDI,
They have some conflicting files... and this may lead to the problems you
are facing...

rumengb, 2020-09-03:
No response. Considering it resolved.

closed 2020-09-03 by rumengb