indigo-astronomy/indigo #325: ccd_asi: ZWO ASI SDK hangup

tail-feather, 2020-04-28:
1. Connect two ASI cameras (I tested with ASI294MC and ASI120MM-mini)
2. Start shooting with ASI120MM-mini in streaming mode
3. Start 1 second exposure on the ASI294MC
4. Call AbortExposure because it is still in BUSY state after few seconds
    * Note: If another camera is working in streaming mode, it will take longer than usual to finish shooting. (`ASIGetExpStatus` keeps returning `ASI_EXP_WORKING` for a long time.)
5. ASI driver hangs up.
    * In the log below, INDIGO hung for 5 minutes and then started flowing again.

log:
```
21:46:37.752442 indigo_server: INDIGO Bus: Change request - Device 'ZWO ASI294MC Pro #0' token 0x0, Proprerty 'CCD_ABORT_EXPOSURE' token 0x0
21:46:37.752463 indigo_server: indigo_ccd_asi[asi_abort_exposure:363]: asi_abort_exposure(939650176)
21:46:37.752468 indigo_server: indigo_ccd_asi[asi_abort_exposure:365]: calling ASIStopExposure(0)
21:46:38.613203 indigo_server: ASIGetVideoData(1): 0.002864s
21:46:38.613224 indigo_server: indigo_ccd_asi[streaming_timer_callback:530]: ASIGetVideoData((1) = 0
21:46:38.613235 indigo_server: RAW to FITS conversion in 7e-06s
21:51:53.667472 indigo_server: indigo_ccd_asi[asi_abort_exposure:367]: done! ASIStopExposure(0): 0
21:51:53.667494 indigo_server: indigo_ccd_asi[asi_abort_exposure:373]: ASIStopExposure(0) = 0
21:51:53.667911 indigo_server: Client upload in 0.535454s
```

polakovic, 2020-04-28:
Did you reduced bandwidth in Advanced/Advanced property?

tail-feather, 2020-04-28:
I didn't know that property.
How should I use it?

rumengb, 2020-04-28:
Actually nobody know. but if sometimes some exposure crash or can not be
downloaded the value should be either increased or decreased...
U usually use valies from 80-90 for USB2 and about 45-60 on USB3

tail-feather, 2020-04-28:
Oh, really?
The ASI120MM is a USB 2.0 camera, but it is 40, and the ASI294MC is a USB3.0 camera, but it is 80.
This seems to be the opposite of each other.

tail-feather, 2020-04-28:
Wow! It's working!!
Thank you so much!!

closed 2020-04-28 by tail-feather