indigo-astronomy/indigo #326: Question: How to user sequnce in the imager_agent

Weruminger, 2020-04-29:
I have an INDIGO SERVER running on a Raspberry PI, what realy works fine.
Now i wnt to use the Indigo Server controlled by an Apple IPad using the sequence functionality from the imager agent.
**ISSUE:**
 I can not found any documentation or sample , how to use it. i dit try some thing like
exposure=180; wait=2; filter=2, gain=10, offset=0
and many other trys but No Trigger and no Error logs. if I do start the sequence der agent just run the default Batch.
**QUESTION:**
Is there any how to or some samples of sequence and batch strings, so I can use this feature ?

polakovic, 2020-04-29:
You have to put a string as you did, e.g.

exposure=3.0;delay=0.0;count=2.0;name=TESTSEQ1;mode=RAW 1600x1200;filter=Filter #1 

to items Batch #1 .... and the numbers in which the batches should be executed to Sequence item, e.g.

1;2;3;4

Actually, this low level API is not very user friendly, it is supposed to be used from some client which will configure the agent for you.

Peter

Weruminger, 2020-04-29:
Wonder full. Thx for the weryy quick response.
I just needed the Syntax. This helps me so much.
I also thought tha ther could be filenames for som JSON files.
But indeed. This easy small interface is fine for me.

Weruminger, 2020-04-29:
Just a short additional queastion:
I do think it is also as easy to create and use profiles.
**ISSUE:**
I've no idea what is the workflow and usage of it. And I also found no information or samples.
**QUESTION:**
Is there any How to or Sample?

rumengb, 2020-04-29:
imager agent is Peter's baby, I asked him to create a simple document to
describe it... Maybe you can make him do it :)

Weruminger, 2020-04-29:
@rumengb : It is easy. Because the syntax is generic, it was not easy to find the information in code but now i do see how it works. I will play with it and write an article, how to use it what may be added to the Project wiki.

rumengb, 2020-04-29:
That would be great. I recently added some documents to the repo but we
have so many ongoing tasks on Indigo that documentation is never a
priority...

Weruminger, 2020-04-29:
I know this issue from my projects. But if i‘ve got such a quick a solving support, it is on me to revenge a little by documenting. 😸

Weruminger, 2020-05-11:
A resume:
Sequence worked fine with sequences on CCD Simulation.
Sequence worked fine with sequences on GPhoto2 with my Olymus OMD.
Sequence on ZWO ASI533C Pro let the server complete hangs up. I tried batches down to most simple bulp only. The server needs to be restarted. No Log was written.

For now i just use a Jenkins CI server to make sequences. 
the base workflow seems to be fine, 
Next i'll try to communicate with the Indigo Server (on my first try i'll use groovy, if this leaks i'll change to Python) wish me luck.

polakovic, 2020-05-11:
Pls. can you set log level to "trace" on server side (or to use -vvv switch on command line) and try it once again with ASI? You can find the log on http://indigosky.local:7624/log. Thank you...

Weruminger, 2020-05-17:
### Resume:
after my last resume i did made some changes.

1. I changed the system to astroberry ( reason, it comes wit CCDciel, what i wanted to try against APT because it runs on the Raspi and i can control it via IPad. ( works fine but CCDciel is not the most stable tool.
2. because astroberry use INDI, I took the INDIGO sources from GitHub Build and install them on the astroberry, disable the INDI services and started INDIGO Server.

Result: 👍 
Imager Sequence works as expected (i'm very lucky about it) 
Also as expected, the installed Astronomical SW works fine (i'd say better, becaus more performant) with INDIGO than with INDI 

NEXT STEPS:
I'll revert to INDIGOSKY system on my spare Raspberry and check again. and store the log file
If the issue raise again, I'll also build and install from the fresh repository and chek if it works.

P.S. sorry for the time gaps. but i'm very busy by some business projects.

polakovic, 2020-05-17:
👍

closed 2020-08-24 by rumengb