indigo-astronomy/indigo #329: Nexstar communication error

Maxattax97, 2020-05-06:
It seems like the webpage is updating with some semblance of information, but it's EXTREMELY slow to update, and I see a lot of read failures in here. It's almost as if some of these commands are not supported?

This could be user error... It took me awhile to discover that I needed to use the RS-232 setting nested in the menus. It wasn't documented anywhere I could find.

I have a Celestron 80 GT, I'm fairly certain my handset is running version 101.0 based on the information [here](https://www.nexstarsite.com/Firmware/GTFirmware.htm).

Attached is a Trace-level log from Indigo.
```
00:31:13.643246 indigo_server: 26 → {"newSwitchVector":{"device":"Mount Nexstar","name":"CONNECTION","items":[{"name":"CONNECTED","value":true}]}}
00:31:13.665374 indigo_server: INDIGO Bus: property change request
00:31:13.665483 indigo_server: 'Mount Nexstar'.'CONNECTION' SWITCH UNDEFINED Idle 2.0 0 (null) {
00:31:13.665550 indigo_server:   'CONNECTED' = On 
00:31:13.665610 indigo_server: }
00:31:13.665677 indigo_server: INDIGO Bus: Change request - Device 'Mount Nexstar' token 0x0, Proprerty '(null)' token 0xade289e8
00:31:13.665781 indigo_server: INDIGO Bus: property update
00:31:13.665853 indigo_server: 'Mount Nexstar'.'CONNECTION' SWITCH rw Busy 2.0 0 (null) {
00:31:13.665915 indigo_server:   'CONNECTED' = On 
00:31:13.665979 indigo_server:   'DISCONNECTED' = Off 
00:31:13.666041 indigo_server: }
00:31:13.666206 indigo_server: 26 ← { "setSwitchVector": { "device": "Mount Nexstar", "name": "CONNECTION", "state": "Busy", "items": [  { "name": "CONNECTED", "value": true }, { "name": "DISCONNECTED", "value": false } ] } }
00:31:13.670257 indigo_server: libnexstar: write 56
00:31:13.683105 indigo_server: libnexstar: read 65 00 23
00:31:13.683225 indigo_server: libnexstar: write 56
00:31:13.699051 indigo_server: libnexstar: read 65 00 23
00:31:13.699176 indigo_server: libnexstar: write 56
00:31:13.715054 indigo_server: libnexstar: read 65 00 23
00:31:13.715161 indigo_server: libnexstar: write 76
00:31:18.769949 indigo_server: libnexstar: read FAILED
00:31:18.770113 indigo_server: libnexstar: write 6d
00:31:23.809950 indigo_server: libnexstar: read FAILED
00:31:23.810089 indigo_server: indigo_mount_nexstar[mount_change_property:680, 0xade823d0]: get_mount_capabilities(9) = -1
00:31:23.811270 indigo_server: libnexstar: write 6d
00:31:28.849938 indigo_server: libnexstar: read FAILED
00:31:28.850097 indigo_server: indigo_mount_nexstar[mount_change_property:685, 0xade823d0]: tc_get_model(9) = -1 (File exists)
00:31:28.852015 indigo_server: libnexstar: write 56
00:31:28.860459 indigo_server: libnexstar: read 65 00 23
00:31:28.860584 indigo_server: libnexstar: write 76
00:31:33.889942 indigo_server: libnexstar: read FAILED
00:31:33.890135 indigo_server: libnexstar: write 74
00:31:38.929906 indigo_server: libnexstar: read FAILED
00:31:38.930067 indigo_server: indigo_mount_nexstar[mount_change_property:734, 0xade823d0]: tc_get_tracking_mode(9) = -1 (File exists)
00:31:38.931910 indigo_server: libnexstar: write 50 01 b0 fe 00 00 00 02
00:31:43.969907 indigo_server: libnexstar: read FAILED
00:31:43.970046 indigo_server: INDIGO Bus: property definition
00:31:43.975653 indigo_server: 'Mount Nexstar'.'TRACKING_MODE' SWITCH rw Ok 2.0 0 (null) { // OneOfMany
00:31:43.975741 indigo_server:   'EQ' = Off // EQ mode
00:31:43.975810 indigo_server:   'AA' = Off // Alt/Az mode
00:31:43.975880 indigo_server:   'AUTO' = On // Automatic mode
00:31:43.975946 indigo_server: }
00:31:43.976124 indigo_server: 26 ← { "defSwitchVector": { "version": 512, "device": "Mount Nexstar", "name": "TRACKING_MODE", "group": "Mount", "label": "Tracking mode", "perm": "rw", "state": "Ok", "rule": "OneOfMany", "items": [  { "name": "EQ", "label": "EQ mode", "value": false }, { "name": "AA", "label": "Alt/Az mode", "value": false }, { "name": "AUTO", "label": "Automatic mode", "value": true } ] } }
00:31:43.976271 indigo_server: timer #0 (of 2) used for 0s
00:31:43.976403 indigo_server: 21 → 0
00:31:43.976567 indigo_server: libnexstar: write 4c
00:31:43.976851 indigo_server: INDIGO Bus: property definition
00:31:43.976937 indigo_server: 'Mount Nexstar'.'MOUNT_INFO' TEXT ro Ok 2.0 0 (null) { // 
00:31:43.977003 indigo_server:   'VENDOR' = 'Celestron' // Vendor
00:31:43.977069 indigo_server:   'MODEL' = 'Unknown' // Model
00:31:43.977135 indigo_server:   'FIRMWARE_VERSION' = 'NexStar 101.00' // Firmware
00:31:43.977198 indigo_server: }
00:31:43.977381 indigo_server: 26 ← { "defTextVector": { "version": 512, "device": "Mount Nexstar", "name": "MOUNT_INFO", "group": "Mount", "label": "Info", "perm": "ro", "state": "Ok", "items": [  { "name": "VENDOR", "label": "Vendor", "value": "Celestron" }, { "name": "MODEL", "label": "Model", "value": "Unknown" }, { "name": "FIRMWARE_VERSION", "label": "Firmware", "value": "NexStar 101.00" } ] } }
00:31:43.977464 indigo_server: INDIGO Bus: property definition
00:31:43.977536 indigo_server: 'Mount Nexstar'.'GEOGRAPHIC_COORDINATES' NUMBER rw Ok 2.0 0 (null) { // 
00:31:43.977629 indigo_server:   'LATITUDE' = -4.64373e+216 (-90, 90, 0) // Latitude (-90 to +90° +N)
00:31:43.977707 indigo_server:   'LONGITUDE' = 360 (-180, 360, 0) // Longitude (0 to 360° +E)
00:31:43.977772 indigo_server: }
00:31:43.977988 indigo_server: 26 ← { "defNumberVector": { "version": 512, "device": "Mount Nexstar", "name": "GEOGRAPHIC_COORDINATES", "group": "Site", "label": "Location", "perm": "rw", "state": "Ok", "items": [  { "name": "LATITUDE", "label": "Latitude (-90 to +90° +N)", "min": -90, "max": 90, "step": 0, "format": "%12.9m", "target": 0, "value": -4.64373e+216 }, { "name": "LONGITUDE", "label": "Longitude (0 to 360° +E)", "min": -180, "max": 360, "step": 0, "format": "%12.9m", "target": 0, "value": 360 } ] } }
00:31:43.978071 indigo_server: INDIGO Bus: property definition
00:31:43.978140 indigo_server: 'Mount Nexstar'.'MOUNT_LST_TIME' NUMBER ro Ok 2.0 0 (null) { // 
00:31:43.978214 indigo_server:   'TIME' = 5.45116 (0, 24, 0) // LST Time
00:31:43.978278 indigo_server: }
[... 330 lines cut ...]
00:33:06.215035 indigo_server:   'TIME' = '2009-11-23T08:39:36' 
00:33:06.215096 indigo_server:   'OFFSET' = '859451440' 
00:33:06.215157 indigo_server: }
00:33:06.215379 indigo_server: 26 ← { "setTextVector": { "device": "Mount Nexstar", "name": "UTC_TIME", "state": "Alert", "items": [  { "name": "TIME", "value": "2009-11-23T08:39:36" }, { "name": "OFFSET", "value": "859451440" } ] } }
00:33:06.215457 indigo_server: INDIGO Bus: property update
00:33:06.215523 indigo_server: 'Mount Nexstar'.'MOUNT_TRACKING' SWITCH rw Alert 2.0 0 (null) {
00:33:06.215585 indigo_server:   'ON' = Off 
00:33:06.215645 indigo_server:   'OFF' = On 
00:33:06.215705 indigo_server: }
00:33:06.215797 indigo_server: 26 ← { "setSwitchVector": { "device": "Mount Nexstar", "name": "MOUNT_TRACKING", "state": "Alert", "items": [  { "name": "ON", "value": false }, { "name": "OFF", "value": true } ] } }
00:33:06.215865 indigo_server: INDIGO Bus: property update
00:33:06.215929 indigo_server: 'Mount Nexstar'.'TRACKING_MODE' SWITCH rw Ok 2.0 0 (null) {
00:33:06.215991 indigo_server:   'EQ' = Off 
00:33:06.216051 indigo_server:   'AA' = Off 
00:33:06.216111 indigo_server:   'AUTO' = On 
00:33:06.216169 indigo_server: }
00:33:06.216260 indigo_server: 26 ← { "setSwitchVector": { "device": "Mount Nexstar", "name": "TRACKING_MODE", "state": "Ok", "items": [  { "name": "EQ", "value": false }, { "name": "AA", "value": false }, { "name": "AUTO", "value": true } ] } }
00:33:06.216336 indigo_server: timer #0 (of 2) used for 0.5s
00:33:06.716524 indigo_server: libnexstar: write 4c
00:33:06.725748 indigo_server: libnexstar: read 30 23
00:33:06.725856 indigo_server: libnexstar: write 65
00:33:11.729933 indigo_server: libnexstar: read FAILED
00:33:11.732929 indigo_server: indigo_mount_nexstar[position_timer_callback:481, 0xad4ff3d0]: tc_get_rade_p(9) = -1 (Success)
00:33:11.733049 indigo_server: libnexstar: write 77
```

Maxattax97, 2020-05-06:
I've determined this is because the firmware on my hand control is out of date. The board inside is labeled with `NXW210` and `2003`. Holding the Info and Undo buttons on boot does not lead to the `Transmitting...` message used to load new firmware, and I need v4.03 in order to use these commands.

I'll order a new one online and report back.

Maxattax97, 2020-05-10:
I have acquired a controller with v4.04 which actually _is_ compatible, despite documentation I've read.

This is no longer an issue. If you have problems like me, it's probably your controller.

closed 2020-05-10 by Maxattax97

rumengb, 2020-05-10:
Thank you for your report! It is useful when people report how the problem
was solved.