indigo-astronomy/indigo #331: Mount Iexos 100

ettoreribaudo, 2020-05-14:
Good morning is possibile to install drivers INDI to Indigosky for this mount?
Regards
Ettore

polakovic, 2020-05-14:
What OS is it?

On linux you have to install both INDI and INDIGO and add the INDI driver with -i switch on indigo_server command line. Nevertheless, I don't recommend it, installing INDI and INDIGO on the same machine can have unwanted side effects :(

On macOS you it is very hard do it.

ettoreribaudo, 2020-05-14:
The sistem is raspbian for Pi 4b. I have driver for INDI is perfect but i
try INDIGOSky and this driver is not present.
I try tomorrow to install.
Thank for all.
Ettore

AstroAde, 2020-05-26:
May I second a request for an Explore Scientific iEXOS-100 (PMC-8) mount driver for INDIGO.

I'm also using a Raspberry Pi 4b running INDIGO Sky; the current INDI driver works just fine.

I really would love to have the iEXOS-100 mount running under INDIGO Sky!

Many thanks in advance, Ade

polakovic, 2020-05-26:
It is on TODO list, but we have no such mount yet.

ettoreribaudo, 2020-05-27:
OK thanks.

Ettore Guido Basiglio Ribaudo
Area Manager Udine
Mob.+39 334 6274558
ettore.ribaudo@gmail.com

S4 ENERGIA SRL
Via Torino, 105 
30172 Mestre (VE)
S4 Energia

> Il giorno 26 mag 2020, alle ore 20:55, Peter Polakovic <notifications@github.com> ha scritto:
> 
> It is on TODO list, but we have no such mount yet.

polakovic, 2020-08-10:
PCM8 support added: 1947b06619836e7519fb5f20bf5a19b6e78fed2a

closed 2020-08-10 by polakovic