indigo-astronomy/indigo #333: QHY294C camera is not  detected

alemk96, 2020-05-19:
I use a  ccd QHY294C connect via USB 3.0 with a raspberry 4 but loading ccd_qhy driver the camera isn't detected from INDIGO webserver. There is a chance for support this camera?

polakovic, 2020-05-19:
Try ccd_qhy2 driver.

alemk96, 2020-05-19:
I already use ccd_qhy2 but camere still not detected

polakovic, 2020-05-19:
You can try to turn more verbose logging in a control panel (Server > Main > Log level > Trace), repeat the test and to check the log at http://indigosky.local/log and maybe you will find something what will help to diagnose the problem. Nevertheless, the only chance how to get any particular camera supported is to make QHY to finally fix their SDK :(

alemk96, 2020-06-07:
Currently, we use a custom version of indigo because we have written a driver for our dome. We compiled this version following the instructions reported here (http://indigo-astronomy.org/for-developers.html) and adding our `dome_driver` folder into the `indigo_drivers` folder. Both with this custom setup and the standard version installed with `sudo apt install indigo`, our QHY294c is not detected. following your advice, we have enabled a verbose logging. In both cases, the output is:

```
QHYCCD|QHYCCD.CPP|InitQHYCCDResource| START
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|InitQHYCCDResourceInside   START
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|    InitQHYCCDResourceInside   END
QHYCCD|QHYCCD.CPP|ScanQHYCCD|START
QHYCCD|QHYCCD.CPP|ScanQHYCCD|ScanQHYCCD numdev=0
QHYCCD|QHYCCD.CPP|ScanQHYCCD|Scan  finished. Return nid=0
QHYCCD|QHYCCD.CPP|ScanQHYCCD|numdev =  0
QHYCCD|QHYCCD.CPP|InitQHYCCDResource| END
```

With some other tests, we have discovered that using IndigoSky on our Raspberry Pi 4, the qhy294C works with qhy2_driver. However, after the installation of the prerequisites reported in your web site for the source build (`sudo apt-get install build-essential autoconf autotools-dev libtool cmake libudev-dev libavahi-compat-libdnssd-dev libusb-1.0-0-dev fxload libcurl4-gnutls-dev libgphoto2-dev libz-dev git curl`), our qhy294C stop working (and being detected).

I don't have any idea about the reason.

polakovic, 2020-06-07:
It just means, that the camera was not detected by used SDK. It is quite outdated, but we hopefully have a new one. I'll try to update it today.

rumengb, 2020-06-07:
Can you please show here the output of fxload -h. It might be that the
firmware is not loaded. And maybe fxload is overwritten at some point with
this installation of packages. And the raspbian stock fxload does not work
for all cameras.

polakovic, 2020-06-07:
Pls. can you also try branch New_QHY_SDK?

alemk96, 2020-06-21:
Ok, this week-end we fixed QHY error, we have used on Raspberry  Ubuntu server OS and indigo github's version and QHY294C works correctly. Thanks for yours support.

closed 2020-06-21 by alemk96