indigo-astronomy/indigo #334: ASIAIR aux device ?

astrorafael, 2020-05-20:
ASIAIR is an interesting device. It is a bit more than a RPi4 in disguise. It has a HAT to control four 12V power outputs that are independently switchable. Are there any plans to include an `aux_asiair` device in your indigosky distribution?
Best wishes
Rafael

rumengb, 2020-05-20:
I doubt ASI will be happy with this. ASIAIR is a proprietary product which
sells fith their own image and they will not be happy to share the
specification of this HAT. But we can try...

astrorafael, 2020-05-20:
Thanks ! let's cross fingers ...

rumengb, 2020-05-22:
Can you contact me in personal at rumen@skyarchive.org

astrorafael, 2020-05-22:
Sure! mail sent

rumengb, 2020-06-22:
Discussed with ZWO and they prefer not to make such integration. So this can not be done!

closed 2020-06-22 by rumengb