indigo-astronomy/indigo #336: Driver UPBv2: does not connect in first attempt

Ruediger3, 2020-05-23:
Hello Rumen and Peter,
when connecting UPBv2 the first connect attempt fails. It needs a second try to establish connection.
It it was connected and disconnected you can connect immediately. 

Version 120 installed via apt.

rumengb, 2020-05-23:
Peter, please see what I did in mgpbox driver. Maybe the problem is the same. The device fails to respond to commands when freshly booted and needs several attempts with some delay in between...

Ruediger3, 2020-05-23:
The UPBv2 was not booted in between. Just the INDIGO server was restarting.

polakovic, 2020-11-17:
Pls. can you check if it works now? Thank you!

Ruediger3, 2020-11-17:
Hi Peter,sure I will test asap. Only need some good weather. The equipment is in the house because of very bad weather here. 
I will update this issue.

Ruediger3, 2020-11-20:
Hi Peter,
 just tested and INDIGO connected on the first attempt. 
Thanks for fixing! Enjoy your weekend!
Cheers
Rüdiger

closed 2020-11-20 by Ruediger3