indigo-astronomy/indigo #337: Driver MGBox: Connect causes restart of server

Ruediger3, 2020-05-23:
Hello Rumen and Peter,

I have loaded driver indigo_aux_mgbox.so in V120. When I try to connect MGBox the server restarts.

Logfile:
/usr/bin/indigo_worker: symbol lookup error: /usr/bin/../lib/../lib/indigo_aux_mgbox.so: undefined symbol: indigo_cancel_timer_sync

Thank you!

rumengb, 2020-05-23:
This is behaviour is expected and it is not a bug. This driver uses a call introduced in 2.0.121 framework. Seems like you are trying to load it in Indigo 2.0.120. Please use the driver with indigo 2.0.121...

closed 2020-05-23 by rumengb

Ruediger3, 2020-05-23:
😊 but not available via apt-get 😝

 

Cheers

Rüdiger

rumengb, 2020-05-23:
You can use it with the version of Indigo you  compiled it. It will fail to compile against 2.0.120 anyway...

Ruediger3, 2020-05-23:
Not possible. DEV System is not in productive environment. P-System is installed via APT only.