indigo-astronomy/indigo #340: INDIGOSKY aux_driver

astrorafael, 2020-05-28:
Given the fact that INDIGOSKY runs on a RaspberryPi and this little computer has a GPIO interface, 
I think it would be great to have an aux_gpio driver to offer I/O lines available and exposed though INDIGO properties.

The user should be able to define the pin direction INPUT and OUTPUT and the level in case of OUTPUT pins.

Possible uses:
- Sensing limits through micro-switches (INPUT lines)
- activating things on output and HAT boards (i.e. relay boards, for home made dew heaters, etc)

And maybe an agent which could define actions on other devices baed on GPIO inputs.

Greetings
Rafael

polakovic, 2020-05-28:
Yep, Rumen works on it already :)

astrorafael, 2020-05-28:
Excellent news!

closed 2020-08-24 by rumengb