indigo-astronomy/indigo #351: While using ccd_qhy/ccd_qhy2 driver, indigo_server does not detect the QHY 5L-II in verbose mode.

dmlir, 2020-06-24:
I used RPi4 4GB RAM with Raspbian. Were here attempts with that, please?
For the same conditions, I was trying ZWO with suitable driver and Indigo detected it (the camera was attached).
Change my mind that it is working sometimes.

polakovic, 2020-06-25:
Pls. can you share a trace log (start indigo_server with -vvv switch)? Is it 2.0-122 debian package or 2.0-123 built from the sources (we updated QHY SDK in the meantime).

dmlir, 2020-06-25:
[indigo.log](https://github.com/indigo-astronomy/indigo/files/4830301/indigo.log)

polakovic, 2020-06-25:
The log shows that only SSAG/QHY5 driver was active. For 5L-II you need QHY2 driver.

dmlir, 2020-06-25:
I see, sorry. Here is it 
[indigo.log](https://github.com/indigo-astronomy/indigo/files/4833671/indigo.log)

polakovic, 2020-11-17:
Pls. can you retest with 2.0-134? I tried with 5L-II and QHY2 driver and it seems to work correctly. But pls. note that this driver doesn't support hot-plug anymore (it is not reliable, so we disabled it). The camera should be connected before you power up RPi.

polakovic, 2026-10-02:
No feedback since the retest request. The camera works with the QHY2 driver here, provided it is connected before the server starts. Please reopen with a trace log if the problem persists.

closed 2026-10-02 by polakovic