indigo-astronomy/indigo #353: Canon 1300D and Zwo 120mm

ettoreribaudo, 2020-07-16:
I tried indigo, not indigosky, for raspberry but i have problem with Canon and guider.
[image]
[image]

ettoreribaudo, 2020-07-16:
16:14:40.357053 indigo_server: indigo_ccd_gphoto2[thread_capture:1534, 0xa7ef63d0]: [rc:-110,camera:0x18e39b0,context:0x1933900] gphoto2_set_key_val_char 'eosremoterelease' 'Press Full'
16:14:45.450981 indigo_server: indigo_ccd_gphoto2: 'Canon EOS 1300D' detached
16:14:45.452060 indigo_server: indigo_ccd_gphoto2: detach device 'Canon EOS 1300D 04a9:32b4 (bus 001, device 004)' from slot '0'
16:14:47.880182 indigo_server: indigo_ccd_gphoto2: auto-detected device 'Canon EOS 1300D 04a9:32b4 (bus 001, device 009)'
16:14:48.040214 indigo_server: indigo_ccd_gphoto2: attach device 'Canon EOS 1300D 04a9:32b4 (bus 001, device 009)' in slot '0'
16:14:48.121485 indigo_server: indigo_ccd_gphoto2: widget 'Bulb Mode' is not available for camera 'CANON EOS 1300D'
16:14:48.124616 indigo_server: indigo_ccd_gphoto2: widget 'Canon EOS Remote Release' is available for camera 'CANON EOS 1300D'
16:14:48.127656 indigo_server: indigo_ccd_gphoto2: widget 'Capture Target' is available for camera 'CANON EOS 1300D'
16:14:48.135469 indigo_server: indigo_ccd_gphoto2: widget 'Camera Date and Time' is available for camera 'CANON EOS 1300D'
16:14:48.226442 indigo_server: indigo_ccd_gphoto2: 'Mirror lockup (customfuncex)' is available for camera 'CANON EOS 1300D'
16:14:48.244597 indigo_server: indigo_ccd_gphoto2: CCD_IMAGE_FORMAT_PROPERTY JPEG uses compression format 'Large Fine JPEG'
16:14:48.244649 indigo_server: indigo_ccd_gphoto2: CCD_IMAGE_FORMAT_PROPERTY FITS/RAW uses compression format 'RAW'
16:14:48.251273 indigo_server: indigo_ccd_gphoto2: 'Canon EOS 1300D' attached
16:17:39.486380 indigo_server: 6 clients
16:17:39.486478 indigo_server: Worker thread finished
16:17:39.486726 indigo_server: 5 clients
16:17:39.486798 indigo_server: Worker thread finished
16:17:39.487154 indigo_server: 4 clients
16:17:39.487225 indigo_server: Worker thread finished
16:17:39.487316 indigo_server: 3 clients
16:17:39.487384 indigo_server: Worker thread finished
16:17:39.487455 indigo_server: 2 clients
16:17:39.487553 indigo_server: Worker thread finished
16:17:39.488198 indigo_server: 1 clients
16:17:39.488425 indigo_server: Worker thread finished
16:20:20.495460 indigo_server: Worker thread started socket = 33
16:20:20.495578 indigo_server: 2 clients
16:20:20.498685 indigo_server: 1 clients
16:20:20.498763 indigo_server: Worker thread finished
16:20:20.529445 indigo_server: Worker thread started socket = 18
16:20:20.529548 indigo_server: 2 clients
16:20:20.531017 indigo_server: GET /mng.html -> OK (1989 bytes)
16:20:20.654978 indigo_server: GET /bootstrap.min.css -> OK (21089 bytes)
16:20:20.678439 indigo_server: Worker thread started socket = 27
16:20:20.678541 indigo_server: 3 clients
16:20:20.680709 indigo_server: GET /indigo.css -> OK (791 bytes)
16:20:20.700089 indigo_server: GET /mng.png -> OK (7867 bytes)
16:20:20.716132 indigo_server: GET /jquery.min.js -> OK (23995 bytes)
16:20:20.749598 indigo_server: Worker thread started socket = 28
16:20:20.749677 indigo_server: 4 clients
16:20:20.750297 indigo_server: Worker thread started socket = 29
16:20:20.750381 indigo_server: 5 clients
16:20:20.751218 indigo_server: Worker thread started socket = 30
16:20:20.751326 indigo_server: 6 clients
16:20:20.753101 indigo_server: GET /popper.min.js -> OK (7266 bytes)
16:20:20.753317 indigo_server: GET /vue.min.js -> OK (31573 bytes)
16:20:20.753510 indigo_server: GET /components.js -> OK (5449 bytes)
16:20:20.753586 indigo_server: GET /bootstrap.min.js -> OK (14002 bytes)
16:20:20.754465 indigo_server: GET /indigo.js -> OK (1957 bytes)
16:20:22.131434 indigo_server: Worker thread started socket = 31
16:20:22.131518 indigo_server: 7 clients
16:20:22.133526 indigo_server: Protocol switched to JSON-over-WebSockets
16:20:22.152012 indigo_server: GET /favicon.ico -> Failed
16:20:22.152231 indigo_server: 6 clients
16:20:22.152276 indigo_server: Worker thread finished
16:20:22.437153 indigo_server: GET /ctrl.png -> OK (10561 bytes)
16:20:22.437873 indigo_server: GET /imager.png -> OK (18162 bytes)
16:20:22.437970 indigo_server: GET /mount.png -> OK (13978 bytes)
16:20:22.459634 indigo_server: GET /guider.png -> OK (13849 bytes)
16:20:27.904265 indigo_server: GET /imager.html -> OK (5151 bytes)
16:20:27.953099 indigo_server: JSON Parser: parser finished
16:20:27.953236 indigo_server: 5 clients
16:20:27.953269 indigo_server: Worker thread finished
16:20:28.023449 indigo_server: GET /bootstrap.min.css -> OK (21089 bytes)
16:20:28.036265 indigo_server: GET /glyphicons.css -> OK (7107 bytes)
16:20:28.056221 indigo_server: GET /indigo.css -> OK (791 bytes)
16:20:28.057266 indigo_server: GET /imager.png -> OK (18162 bytes)
16:20:28.064675 indigo_server: GET /jquery.min.js -> OK (23995 bytes)
16:20:28.104656 indigo_server: GET /popper.min.js -> OK (7266 bytes)
16:20:28.118695 indigo_server: Worker thread started socket = 33
[... 55 lines cut ...]
16:21:25.945597 indigo_server: GET /bootstrap.min.js -> OK (14002 bytes)
16:21:25.945884 indigo_server: GET /indigo.js -> OK (1957 bytes)
16:21:25.946834 indigo_server: GET /vue.min.js -> OK (31573 bytes)
16:21:25.974500 indigo_server: GET /components.js -> OK (5449 bytes)
16:21:26.165976 indigo_server: Worker thread started socket = 28
16:21:26.166045 indigo_server: 7 clients
16:21:26.168048 indigo_server: Protocol switched to JSON-over-WebSockets
16:21:26.250136 indigo_server: GET /ctrl.png -> OK (10561 bytes)
16:21:26.251357 indigo_server: GET /mng.png -> OK (7867 bytes)
[... cut ...]
16:21:33.856890 indigo_server: indigo_ccd_gphoto2[gphoto2_set_key_val:897, 0xad8ff3d0]: [camera:0xb4900730,context:0x1933900] camera get config failed
16:21:36.548251 indigo_server: indigo_ccd_asi[asi_open:239, 0xa86f73d0]: indigo_try_global_lock(): failed to get lock.
16:21:46.768637 indigo_server: indigo_ccd_asi[asi_open:239, 0xa86f73d0]: indigo_try_global_lock(): failed to get lock.
16:21:51.685306 indigo_server: indigo_ccd_asi[asi_open:239, 0xa86f73d0]: indigo_try_global_lock(): failed to get lock.

polakovic, 2020-07-16:
gphoto2 driver is obsoleted, pls. use indigo_ccd_ptp instead. Also you can set exposure time for DSLR only if shutter is set to BULB. Otherwise use 0.

ASI driver shows that there is lock file created for the camera. I don't know why, but try to stop the server and remove all /tmp/indigo_lock_* files.

ettoreribaudo, 2020-07-16:
I tried and the asi is ok but Canon not download image.
[image]

ettoreribaudo, 2020-07-16:
Guider ok
[image]

ettoreribaudo, 2020-07-16:
Canon
[image]

polakovic, 2020-07-17:
Yes, webGUI doesn’t support all RAW formats, but if you will use some JPEG + RAW mode, raw will be saved and JPEG used for preview.

But generally, webGUi is intended for astro cameras, DSLR is substantially different and hard to fit into the same set of controls so you need to configure many things in the control panel.

ettoreribaudo, 2020-07-17:
I tried but nothing canon shoot but nothing image and block all.

ettoreribaudo, 2020-07-17:
Mount and Asi perfect.

polakovic, 2020-07-18:
Try to set Server > log level to "trace" in control panel, restart the server and check the log once again. Remember to reset it back to "error" afterwards.

ettoreribaudo, 2020-07-18:
Sorry for my problems but where is file log?
Ettore

ettoreribaudo, 2020-07-18:
12:07:26.342425 indigo_server: indigo_ccd_ptp[ptp_transaction:943]: request GetEvent_Canon (9116) 00000906 []
12:07:26.342848 indigo_server: indigo_ccd_ptp[ptp_transaction:945]: libusb_bulk_transfer(12) -> OK
12:07:26.344132 indigo_server: indigo_ccd_ptp[ptp_transaction:985]: libusb_bulk_transfer() -> OK, 20
12:07:26.344304 indigo_server: indigo_ccd_ptp[ptp_transaction:995]: data 9116 00000906 +8 bytes
12:07:26.345031 indigo_server: indigo_ccd_ptp[ptp_transaction:1023]: libusb_bulk_transfer() -> OK, 12
12:07:26.345183 indigo_server: indigo_ccd_ptp[ptp_transaction:1033]: response OK (2001) 00000906 []
12:07:27.345379 indigo_server: indigo_ccd_ptp[ptp_transaction:943]: request GetEvent_Canon (9116) 00000907 []
12:07:27.345685 indigo_server: indigo_ccd_ptp[ptp_transaction:945]: libusb_bulk_transfer(12) -> OK
12:07:27.346716 indigo_server: indigo_ccd_ptp[ptp_transaction:985]: libusb_bulk_transfer() -> OK, 20
12:07:27.346767 indigo_server: indigo_ccd_ptp[ptp_transaction:995]: data 9116 00000907 +8 bytes
12:07:27.347177 indigo_server: indigo_ccd_ptp[ptp_transaction:1023]: libusb_bulk_transfer() -> OK, 12
12:07:27.347226 indigo_server: indigo_ccd_ptp[ptp_transaction:1033]: response OK (2001) 00000907 []
12:07:27.470379 indigo_server: INDIGO Bus: Change request - Device 'Server' token 0x0, Proprerty '(null)' token 0xadfa39e8

rumengb, 2020-07-18:
Hi,
unfortunately this log is not complete. Can you please run indigo_server
like this:
indigo_server -vvv indigo_ccd_ptp >indigo.log 2>&1

run exposure with your camera then stop the server with ctrl+c and send us
the recorded log. It is in indigo.log file.

Thank you!
Rumen

ettoreribaudo, 2020-07-18:
Hi Rumen this is my log.
[indigo.log](https://github.com/indigo-astronomy/indigo/files/4942356/indigo.log)

ettoreribaudo, 2020-07-19:
Hi have news for me?

rumengb, 2020-07-19:
Please be patient :) I am on vacation and Peter is not available several
days too... I looked at the log on my mobile and it is complete. This will
help us to debug it...

ettoreribaudo, 2020-07-19:
Ok I wait no problem.
Ettore

polakovic, 2020-07-22:
It looks that there is something wrong with DSLR_SHUTTER property. The camera reports original value 12, which is not listed in the list of available values (that's why no option is selected). You subsequently requested change to BULB, but the change was not accepted by the camera. Can you try to select BULB manually on the camera body and try again?

Also, if you want to test it with webGUI, you can't use RAW mode yet.

ettoreribaudo, 2020-07-22:
I tried but nothing. this is a log.
[indigo.log](https://github.com/indigo-astronomy/indigo/files/4961630/indigo.log)

ettoreribaudo, 2020-07-22:
I tried 5 second but nothing this is a log.
20:10:15.097337 indigo_server: INDIGO server 2.0-122 built on Jun  4 2020 11:57:51
20:10:15.107182 indigo_server: Driver indigo_ccd_ptp 2.0.0.11 loaded
20:10:15.107302 indigo_server: indigo_ccd_ptp[hotplug_callback:507]: libusb_get_device_descriptor ->  OK
20:10:15.107337 indigo_server: indigo_ccd_ptp[hotplug_callback:507]: libusb_get_device_descriptor ->  OK
20:10:15.107369 indigo_server: indigo_ccd_ptp[hotplug_callback:507]: libusb_get_device_descriptor ->  OK
20:10:15.107583 indigo_server: indigo_ccd_ptp[hotplug_callback:507]: libusb_get_device_descriptor ->  OK
20:10:15.107622 indigo_server: indigo_ccd_ptp[hotplug_callback:507]: libusb_get_device_descriptor ->  OK
20:10:15.107679 indigo_server: indigo_ccd_ptp[hotplug_callback:507]: libusb_get_device_descriptor ->  OK
20:10:15.107714 indigo_server: indigo_ccd_ptp[indigo_ccd_ptp:671]: libusb_hotplug_register_callback ->  OK
20:10:15.107752 indigo_server: Resource /mng.html (1989, text/html) added
20:10:15.107782 indigo_server: Resource /mng.png (7867, image/png) added
20:10:15.107810 indigo_server: Resource /ctrl.html (1448, text/html) added
20:10:15.107839 indigo_server: Resource /ctrl.png (10561, image/png) added
20:10:15.107867 indigo_server: Resource /indigo.js (1957, text/javascript) added
20:10:15.107895 indigo_server: Resource /components.js (5449, text/javascript) added
20:10:15.107924 indigo_server: Resource /indigo.css (791, text/css) added
20:10:15.107954 indigo_server: Resource /bootstrap.min.css (21089, text/css) added
20:10:15.107984 indigo_server: Resource /bootstrap.min.js (14002, text/javascript) added
20:10:15.108014 indigo_server: Resource /popper.min.js (7266, text/javascript) added
20:10:15.108049 indigo_server: Resource /glyphicons.css (7107, text/css) added
20:10:15.108078 indigo_server: Resource /glyphicons-regular.ttf (125747, font/ttf) added
20:10:15.108108 indigo_server: Resource /jquery.min.js (23995, text/javascript) added
20:10:15.108141 indigo_server: Resource /vue.min.js (31573, text/javascript) added
20:10:15.108171 indigo_server: Resource /imager.html (5151, text/html) added
20:10:15.108199 indigo_server: Resource /imager.png (18162, image/png) added
20:10:15.108228 indigo_server: Resource /mount.html (6854, text/html) added
20:10:15.108256 indigo_server: Resource /mount.png (13978, image/png) added
20:10:15.108286 indigo_server: Resource /celestial.min.js (37255, text/javascript) added
20:10:15.108315 indigo_server: Resource /d3.min.js (53548, text/javascript) added
20:10:15.108344 indigo_server: Resource /celestial.css (1748, text/css) added
20:10:15.108374 indigo_server: Resource /data/constellations.json (4436, application/json; charset=utf-8) added
20:10:15.108404 indigo_server: Resource /data/constellations.bounds.json (9007, application/json; charset=utf-8) added
20:10:15.108433 indigo_server: Resource /data/mw.json (178122, application/json; charset=utf-8) added
20:10:15.108463 indigo_server: Resource /data/planets.json (1514, application/json; charset=utf-8) added
20:10:15.109120 indigo_server: port_type(): path = /dev/serial1, type = 32, res = 0
20:10:15.109203 indigo_server: indigo_enumerate_serial_ports(): path = /dev/serial1, IS SERIAL (type = 32)
20:10:15.110294 indigo_server: port_type(): path = /dev/ttyAMA0, type = 32, res = 0
20:10:15.110339 indigo_server: indigo_enumerate_serial_ports(): path = /dev/ttyAMA0, IS SERIAL (type = 32)
20:10:15.112522 indigo_server: port_type(): path = /dev/tty1, type = 0, res = -1 error = 'Inappropriate ioctl for device'
20:10:15.112725 indigo_server: port_type(): path = /dev/tty, type = 0, res = -1 error = 'Inappropriate ioctl for device'
20:10:15.114156 indigo_server: indigo_ccd_ptp: 'Canon EOS 1300D #010102' attached
20:10:15.114212 indigo_server: INDIGO Bus: property definition
20:10:15.114247 indigo_server: 'Canon EOS 1300D #010102'.'INFO' TEXT ro Ok 2.0 0 (null) { // 
20:10:15.114277 indigo_server:   'DEVICE_NAME' = 'Canon EOS 1300D #010102' // Device name
20:10:15.114306 indigo_server:   'DEVICE_VERSION' = '2.0.0.11' // Driver version
20:10:15.114335 indigo_server:   'DEVICE_INTERFACE' = '2' // Interface
20:10:15.114399 indigo_server: }
20:10:15.114432 indigo_server: INDIGO Bus: property definition
20:10:15.114463 indigo_server: 'Canon EOS 1300D #010102'.'CONFIG' SWITCH rw Ok 2.0 0 (null) { // AtMostOne
20:10:15.114491 indigo_server:   'LOAD' = Off // Load
20:10:15.114519 indigo_server:   'SAVE' = Off // Save
20:10:15.114547 indigo_server:   'REMOVE' = Off // Remove
20:10:15.114574 indigo_server: }
20:10:15.114603 indigo_server: INDIGO Bus: property definition
20:10:15.114633 indigo_server: 'Canon EOS 1300D #010102'.'PROFILE' SWITCH rw Ok 2.0 0 (null) { // OneOfMany
20:10:15.114662 indigo_server:   'PROFILE_0' = On // Profile #0
20:10:15.114691 indigo_server:   'PROFILE_1' = Off // Profile #1
20:10:15.114723 indigo_server:   'PROFILE_2' = Off // Profile #2
20:10:15.114753 indigo_server:   'PROFILE_3' = Off // Profile #3
20:10:15.114783 indigo_server:   'PROFILE_4' = Off // Profile #4
20:10:15.114816 indigo_server: }
20:10:15.114848 indigo_server: INDIGO Bus: property definition
20:10:15.114878 indigo_server: 'Canon EOS 1300D #010102'.'CONNECTION' SWITCH rw Ok 2.0 0 (null) { // OneOfMany
20:10:15.114906 indigo_server:   'CONNECTED' = Off // Connected
20:10:15.114940 indigo_server:   'DISCONNECTED' = On // Disconnected
20:10:15.114982 indigo_server: }
[... cut ...]
20:10:49.732467 indigo_server: INDIGO Bus: property change request
20:10:49.732515 indigo_server: 'Server'.'LOG_LEVEL' SWITCH UNDEFINED Idle 0.0 0 (null) {
20:10:49.732553 indigo_server:   'ERROR' = On 
20:10:49.732591 indigo_server:   'INFO' = Off 
20:10:49.732628 indigo_server:   'DEBUG' = Off 
20:10:49.732665 indigo_server:   'TRACE' = Off 
20:10:49.732700 indigo_server: }
20:10:49.732742 indigo_server: INDIGO Bus: Change request - Device 'Server' token 0x0, Proprerty '(null)' token 0xb375efb8

polakovic, 2020-07-22:
It is slightly better now, BULB mode is recognised:

...
19:47:50.492979 indigo_server: INDIGO Bus: property definition
19:47:50.493003 indigo_server: 'Canon EOS 1300D #010102'.'DSLR_SHUTTER' SWITCH rw Ok 2.0 0 (null) { // OneOfMany
19:47:50.493021 indigo_server:   'c' = On // Bulb
...

What I don't see is any ObjectAddedEx_Canon or ObjectAddedEx2_Canon event coming. Is there properly formatted card inside?

ettoreribaudo, 2020-07-22:
No, is not present inside card.

polakovic, 2020-07-22:
Yes, that's the problem. For each particular camera there is some way how to use it without the card, but the most common working with any camera is to let it write the image to the card and wait for object added event.

ettoreribaudo, 2020-07-22:
Sorry my error i try.
Ettore

ettoreribaudo, 2020-07-22:
Perfect.
Thank Peter.

closed 2020-07-22 by ettoreribaudo

rumengb, 2020-07-22:
Peter,
maybe it is a good idea to list such issues in the driver README, in the
troubleshooting section...