indigo-astronomy/indigo #354: FITS-Tag: ROWORDER

Ruediger3, 2020-07-17:
Hello Peter and Rumen,

please consider to add Fits keyword "ROWORDER" with its possible values
1. "BOTTOM-UP"
2. "TOP-DOWN"

Reason:
Ensure to make data acquired with INDIGO reliable correct orientated with other shared data or older versions e.g. with dark libraries, bias and flats or even lights collected from different facilities.

Example:
I had a lot of issues when trying to calibrate lights with bias until I found out, that PI orientated the calibration files 180 flipped, because they were taken with vendor's native tool.

Note:
Identical request was placed at APT forum.

Further information:
https://free-astro.org/index.php?title=Siril:FITS_orientation
INDIGO should also get listed there.

Thank you!

rumengb, 2020-07-17:
It is an easy fix. I will add it...

Ruediger3, 2020-07-17:
Great! Thanks!

rumengb, 2020-07-18:
Done.

closed 2020-07-18 by rumengb