indigo-astronomy/indigo #359: Imager web app issues with QSI 683 camera

ozbandit, 2020-07-26:
First issue is with cooler. The cooler can be turned on or off to a particular set point. But the camera will output the sensor temperature all the time. When cooler is on, the web app reports the current temperature, but when it is off, it just says "Off". I'd like It to continue reporting the sensor temperature. Use case for this is when I shut down, and I turn off the cooler. I want to let the sensor reach ambient temperature before I power it down. But I am forced to control panel to see the temperature when web app could continue showing it.

Second issue is that drop down for exposure durations allows durations that the device won't support. Anything less than 1 second gives an error. If there is a way to filter those ones out by asking camera what it can do, that would be better.

Third issue is with filter wheel. This is working perfectly from web control panel. I can select filters. But on first connect it does not know which is the active filter slot. Says 0. Maybe there is a way to read the current slot from camera? In the Imager web app however, when you select the filter wheel driver, a row appears underneath that says "MOVING..." and does not go away. The filter wheel is not moving and there is no way to choose a filter.

I have not tested naming the filter slots yet. SO it is possible this doesn't work. I will test later.

polakovic, 2020-07-30:
> First issue is with cooler. The cooler can be turned on or off to a particular set point. But the camera will output the sensor temperature all the time. When cooler is on, the web app reports the [...]

Copy & Paste error fixed here: eb920b6d02e09356114d8d8f5917fe88b0f73ebd

> Second issue is that drop down for exposure durations allows durations that the device won't support. Anything less than 1 second gives an error. If there is a way to filter those ones out by asking camera what it can do, that would be better.

Not so easy, but I'll look at it.

> Third issue is with filter wheel. This is working perfectly from web control panel. I can select filters. But on first connect it does not know which is the active filter slot. Says 0. Maybe there [...]

Fixed here: bf688c10b4c8c5108c1eb3b99f1bf7e6bd223a8a

> I have not tested naming the filter slots yet. SO it is possible this doesn't work. I will test later.

It is implemented in a base code so it probably works.

polakovic, 2020-07-30:
> Second issue is that drop down for exposure durations allows durations that the device won't support. Anything less than 1 second gives an error. If there is a way to filter those ones out by asking camera what it can do, that would be better.

Partially fixed here: 8c0a39a4f10f02d8dd81f4e0b1b25dd698f01bdc, min and max value is set on exposure time now, so if min is 1s, time shorter than 1s will be changed to 1s and shouldn't end with error.

rumengb, 2020-09-03:
I believe all these issues are fixed with the latest driver updates.