indigo-astronomy/indigo #370: latest master branch will not build on my archliux machine

akadata, 2020-08-17:
Symptoms are build errors, first error related to x64 not sure why an x85_64-linux-gnu would want to be x86 unless they dont use x64

Patch below allows a clean build. 

diff --git a/Makefile b/Makefile
index 02e19af8..6f200876 100644
--- a/Makefile
+++ b/Makefile
@@ -97,8 +97,8 @@ else
                endif
                ifeq ($(ARCH_DETECTED),x86_64)
                        ifeq ($(wildcard /lib/x86_64-linux-gnu/),)
-                               ARCH_DETECTED = x86
-                               DEBIAN_ARCH = i386
+                               ARCH_DETECTED = x64
+                               DEBIAN_ARCH = amd64
                        else
                                ARCH_DETECTED = x64
                                DEBIAN_ARCH = amd64
diff --git a/indigo_drivers/ccd_qhy/indigo_ccd_qhy.cpp b/indigo_drivers/ccd_qhy/indigo_ccd_qhy.cpp
index 77c2d432..db6d6aa7 100644
--- a/indigo_drivers/ccd_qhy/indigo_ccd_qhy.cpp
+++ b/indigo_drivers/ccd_qhy/indigo_ccd_qhy.cpp
@@ -28,7 +28,8 @@
 #define DRIVER_VERSION 0x000B
 
 #include <stdlib.h>
-#include <string.h>
+#include <string>
+#include <cstring>
 #include <unistd.h>
 #include <math.h>
 #include <assert.h>

polakovic, 2020-08-18:
The point of this hack is, that if $ARCH_DETECTED is "x86_64" but /lib/x86_64-linux-gnu/ doesn't exist, it is actually build running in docker with x86 image on x64 host. Your path will break it.

What exactly fails on archlinux?

akadata, 2020-08-19:
The first issue is the arch detection, it detects arch x86 not x64 

indigo_aux_dsusb ---------------------------------------

Sources:    indigo_aux_dsusb.c indigo_aux_dsusb_main.c
Archive:    indigo_aux_dsusb.a
SO library: indigo_aux_dsusb.so
indigo_aux_dsusb.o 
Executable: indigo_aux_dsusb
Rules:      dsusb.rules
SDK:        bin_externals/libdsusb/lib/Linux/x86/libdsusb.a -Ibin_externals/libdsusb/include -Lbin_externals/libdsusb/lib/Linux/x86/ -ldsusb

gcc -g -fPIC -O3 -isystem/usr/src/indigo/indigo_libs -I/usr/src/indigo/indigo_drivers -I/usr/src/indigo/indigo_linux_drivers -I/usr/src/indigo/build/include -std=gnu11 -pthread -DINDIGO_LINUX -Ibin_externals/libdsusb/include   -c -o indigo_aux_dsusb.o indigo_aux_dsusb.c
ar -rv /usr/src/indigo/build/drivers/indigo_aux_dsusb.a indigo_aux_dsusb.o
ar: creating /usr/src/indigo/build/drivers/indigo_aux_dsusb.a
a - indigo_aux_dsusb.o
gcc -shared -o /usr/src/indigo/build/drivers/indigo_aux_dsusb.so -Wl,--whole-archive /usr/src/indigo/build/drivers/indigo_aux_dsusb.a -Wl,--no-whole-archive -Lbin_externals/libdsusb/lib/Linux/x86/ -ldsusb -lindigo -lm -lrt -lusb-1.0 -pthread -L/usr/src/indigo/build/lib -Wl,-rpath=\$ORIGIN/../lib,-rpath=\$ORIGIN/../drivers,-rpath=. -lusb-1.0
/usr/bin/ld: skipping incompatible bin_externals/libdsusb/lib/Linux/x86//libdsusb.a when searching for -ldsusb
/usr/bin/ld: cannot find -ldsusb
collect2: error: ld returned 1 exit status
make[2]: *** [../../Makefile.drv:121: /usr/src/indigo/build/drivers/indigo_aux_dsusb.so] Error 1
make[2]: Leaving directory '/usr/src/indigo/indigo_drivers/aux_dsusb'
make[1]: *** [../Makefile.drvs:66: aux_dsusb/] Error 2
make[1]: Leaving directory '/usr/src/indigo/indigo_drivers'
make: *** [Makefile:127: all] Error 2

polakovic, 2020-08-19:
Maybe the easiest workaround is to create /lib/x86_64-linux-gnu folder.

polakovic, 2020-08-24:
Because Makefile tests the existence of standard location /lib/x86_64-linux-gnu (while on your system they are on a nonstandard location /lib/gcc/x86_64-pc-linux-gnu) to detect if it is really a 64 bit system.

"wildcard" is a command, not a pathname.

closed 2020-11-04 by rumengb