indigo-astronomy/indigo #374: Adafruit MotorHat Focuser

supritsingh, 2020-09-08:
Hi,

Any possibility to integrate a Focuser driver based on Adafruit Motor Hat for Raspberry Pi. The same can be incorporated in indigosky much like https://github.com/rkaczorek/astroberry-amh. Let me know if this already works. I plan to drive a 27:1 planetary stepper for focusing. 

Best,
Suprit

rumengb, 2020-09-08:
Hi,
The trouble is that this is INDI driver. Not INDIGO. In general you can use
INDI drivers with INDIGO server, but we never do this in our distribution
for many reasons. One of which is that we develop and support INDIGO. We
can not support PHD, kstars, CCdciel etc... therefore we provide indigosky.
which is essentially preconfigured INDIGO server. The good news is that you
can install everything as this is a normal raspbian based linux under the
hood. So you can install this driver (but in order to work you will need
the whole indi framework) and load it in indigo server. The other
(better) approach is to ask the author of this driver to write an INDIGO
driver for that hardware. And the third option is that we can write one but
we are too busy with order tasks on indigo and I have no idea when we will
be able to do it.

Rumen

supritsingh, 2020-09-08:
Thanks for clarification, Rumen. I will try to tinker with it [although thinking and writing drivers is an alien thing to me, having never done so]. Any pointers for writing drivers for indigo will be appreciated.

polakovic, 2020-09-08:
https://github.com/indigo-astronomy/indigo/blob/master/indigo_docs/DRIVER_DEVELOPMENT_BASICS.md

closed 2023-11-18 by rumengb