indigo-astronomy/indigo #383: ccd_simultor: coolding does not work

rumengb, 2020-10-30:
[image]
turn on cooling set target temperate and nothing happens.

polakovic, 2020-10-30:
It does work :) Temperature is decreased or increased 1 degree every 5 seconds/

rumengb, 2020-10-30:
It just does not work sometimes. I waited for several minutes. Then
reloaded the diver.

rumengb, 2020-10-30:
It is not only the cooler. When i start guiding with guider imager, start exposures with imager and turn the temperature control on. very often it never starts, sometimes guider freezes like in the example below:

rumen@vega:~/work/indigo_deadlock_detector.git$ ./indigo_deadlock_detector.sh indigo_server
Inspecting 'indigo_server' (pid = 363067) for deadlocks
[New LWP 363068]
[New LWP 363069]
[New LWP 363070]
[New LWP 363071]
[New LWP 363072]
[New LWP 363076]
[New LWP 363078]
[New LWP 363079]
[New LWP 363080]
[New LWP 363081]
[New LWP 363103]
[New LWP 363117]
[New LWP 363143]
[New LWP 363286]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x00007f2861e8849f in __libc_accept (fd=9, addr=..., addr@entry=..., len=len@entry=0x7ffe95a4b3e4) at ../sysdeps/unix/sysv/linux/accept.c:26
None

Thread 15 (Thread 0x7f283f7fe700 (LWP 363286)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7f2830001874) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7f2830001878, cond=0x7f2830001848) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=cond@entry=0x7f2830001848, mutex=mutex@entry=0x7f2830001878) at pthread_cond_wait.c:638
#3  0x00007f2861ebe99b in timer_func (timer=0x7f2830001820) at indigo_timer.c:123
#4  0x00007f2861e7d609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#5  0x00007f2861da4293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 14 (Thread 0x7f283ffff700 (LWP 363143)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7f2838005994) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7f2838005998, cond=0x7f2838005968) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=cond@entry=0x7f2838005968, mutex=mutex@entry=0x7f2838005998) at pthread_cond_wait.c:638
#3  0x00007f2861ebe99b in timer_func (timer=0x7f2838005940) at indigo_timer.c:123
#4  0x00007f2861e7d609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#5  0x00007f2861da4293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 13 (Thread 0x7f2851ffb700 (LWP 363117)):
#0  futex_abstimed_wait_cancelable (private=<optimized out>, abstime=0x7f2851ffae20, clockid=<optimized out>, expected=0, futex_word=0x7f2844000bb0) at ../sysdeps/nptl/futex-internal.h:320
#1  __pthread_cond_wait_common (abstime=0x7f2851ffae20, clockid=<optimized out>, mutex=0x7f2844000bb8, cond=0x7f2844000b88) at pthread_cond_wait.c:520
#2  __pthread_cond_timedwait (cond=cond@entry=0x7f2844000b88, mutex=mutex@entry=0x7f2844000bb8, abstime=abstime@entry=0x7f2851ffae20) at pthread_cond_wait.c:656
#3  0x00007f2861ebea76 in timer_func (timer=0x7f2844000b60) at indigo_timer.c:74
#4  0x00007f2861e7d609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#5  0x00007f2861da4293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 12 (Thread 0x7f28517fa700 (LWP 363103)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7f2848023f64) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7f2848023f68, cond=0x7f2848023f38) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=cond@entry=0x7f2848023f38, mutex=mutex@entry=0x7f2848023f68) at pthread_cond_wait.c:638
#3  0x00007f2861ebe99b in timer_func (timer=0x7f2848023f10) at indigo_timer.c:123
#4  0x00007f2861e7d609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#5  0x00007f2861da4293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 11 (Thread 0x7f28527fc700 (LWP 363081)):
#0  0x00007f2861d623bf in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=req@entry=0x7f28527fbd40, rem=rem@entry=0x0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
#1  0x00007f2861d68047 in __GI___nanosleep (requested_time=requested_time@entry=0x7f28527fbd40, remaining=remaining@entry=0x0) at nanosleep.c:27
#2  0x00007f2861d9a9bf in usleep (useconds=<optimized out>) at ../sysdeps/posix/usleep.c:32
#3  0x00007f2861eca77e in indigo_usleep (delay=<optimized out>) at indigo_bus.c:1314
#4  0x00007f2862d6025f in exposure_batch (device=device@entry=0x7f285413ab30) at indigo_agent_imager.c:426
#5  0x00007f2862d606e1 in exposure_batch_process (device=0x7f285413ab30) at indigo_agent_imager.c:525
#6  0x00007f2861ebe8a8 in timer_func (timer=0x7f2848024de0) at indigo_timer.c:86
#7  0x00007f2861e7d609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007f2861da4293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 10 (Thread 0x7f2852ffd700 (LWP 363080)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7f2848024c20) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7f2848024c28, cond=0x7f2848024bf8) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=cond@entry=0x7f2848024bf8, mutex=mutex@entry=0x7f2848024c28) at pthread_cond_wait.c:638
[... 63 lines cut ...]
#3  0x00007f2861aff6b3 in libusb_handle_events () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#4  0x00007f2861ecba17 in hotplug_thread (arg=<optimized out>) at indigo_driver.c:613
#5  0x00007f2861e7d609 in start_thread (arg=<optimized out>) at pthread_create.c:477
[... cut ...]
#1  0x00007f2861ebb1f1 in indigo_server_start (callback=<optimized out>) at indigo_server_tcp.c:378
#2  0x000056388562cd4d in server_main () at indigo_server.c:1154
#3  0x000056388562aea5 in main (argc=<optimized out>, argv=<optimized out>) at indigo_server.c:1296

Blocked threads:
*****************************************************
*****************************************************

[Inferior 1 (process 363067) detached]

And the exposure becomes busy... and stays like that forever:
[image]

rumengb, 2020-10-31:
it is not new.... it affects 2.0-128 too on both rpi and intel

polakovic, 2020-10-31:
fixed by 6850e5cbd88f076acde4440724a5d5869df1e057

closed 2020-10-31 by polakovic