indigo-astronomy/indigo #386: Hotspot will not automatically connect.

tlkw79, 2020-11-03:
Hi all,

I found out hotspot (AP) will not automatically connect when home wifi drop. Not very convenient at all. 

Is this feature not included in indigo yet?

Thanks.

polakovic, 2020-11-03:
It is not INDIGO feature, but Debian WiFi configuration issue. We need to investigate it...

rumengb, 2020-11-03:
I have checked the issue as I understand it and it works as expected.
There are 2 use cases: 
1. configure indigosky in AP mode - default mode, it accepts connections and it does not connect to anything other devices connect to it.

2. configure indigosky as a WiFi client - indigosky connects to an existing WIFI network.

I assume you are talking about the second use case.  You configure indigosky to connect to an existing network.
I did so, it connected and it works, then I turned off the AP, indigosky lost connection. Then I turned the AP on again and in seconds indigosky connected again. I did his several times and it always worked.

Can you please be more specific and explain in greater details the steps to reproduce it and what exactly happened?

tlkw79, 2020-11-04:
Hi,

What I mean is If I configure indigo as a Wifi client, will AP mode become standby but not totally turn off? 

example:

I connected indigo as a Wifi Client and suddenly my home Wifi router having an issue, for sure indigo will disconnect. So, will the AP mode wake up itself and let me connect it as hotspot and continue the job?

Thanks.

rumengb, 2020-11-04:
Ah, I understand now :) Let me explain a bit. There is no such WIFI feature, however there is something similar to what you want. To run the WiFi interface in both AP and STA (AP client mode) at the same time, but not all wireless interfaces support it.
Raspberry PI is all about being affordable and its onboard WIFI is not an exception, they did not sacrifice this feature, but at another cost. However it claims it supports this feature the reality is a bit different. The truth is that it can barely maintain a stable operation in AP mode, let alone in both AP and STA at the same time. I experimented with this in the past but I gave it up as I was unable to make it work reliably. And for what I have seen on the net nobody really did. By the way many advanced users are used to stick an external WiFi dongles that sometimes cost about the praise of the RPI itself and use it instead of the built in one because they do not find the AP mode stable enough. I am sorry but in this case: "You get what you pay for" is 100% valid.
There is not much we can do about this.

closed 2020-11-04 by rumengb