indigo-astronomy/indigo #387: Add DARK FLATS option to frame type

polakovic, 2020-11-06:


rumengb, 2020-11-07:
I do not think we need it. It will just clutter the property and it is
basically a dark frame for the flat frame. There is no technical difference
between dark frame and dark flat frame. It just have to match the flat
exposure time. So there is no reason to do this.

polakovic, 2020-11-07:
The reason is, that it is copies to FITS metadata and/or file name. It is repeated user request...

polakovic, 2020-11-17:
https://pixinsight.com/forum/index.php?threads/understand-darkflat-image-type-in-wbpp-bpp.14545/

polakovic, 2020-11-17:
825b1d95c91b1225d58b3d72ab8bc6ed155af09d

closed 2020-11-17 by polakovic

jconejero, 2020-11-17:
FWIW, we have had some passionate discussions about the terms 'dark 
flat' and 'flat dark'. See for example:

https://pixinsight.com/forum/index.php?threads/understand-darkflat-image-type-in-wbpp-bpp.14545/#post-88690

https://pixinsight.com/forum/index.php?threads/understand-darkflat-image-type-in-wbpp-bpp.14545/#post-88696

I personally prefer 'flat dark', but our preprocessing pipelines support 
both the FLATDARK and DARKFLAT FITS keywords. I would generate FLATDARK 
keywords because I think it's more correct semantically, but I'm not 
sure about compatibility with other applications.

Juan Conejero
PixInsight Development Team
Pleiades Astrophoto, S.L.
https://pixinsight.com/

polakovic, 2020-11-17:
Yes, I saw you support both. I googled a little bit and situation is unclear :)

jconejero, 2020-11-17:
Absolutely unclear, of course. We're speaking of FITS, what did you 
expect? :D