indigo-astronomy/indigo #389: Support for large >2GB files on Indigo Sky?

aaronfreimark, 2020-11-30:
Now that AVI and SER support has been added to the drivers, it is easy to make large files. However, my files could reach a maximum of 2GB only. I saw this only on indigosky. When I compiled on Mac, I could create large files.

The error I see is:
08:29:38.749009 indigo_server: indigo_write(): File too large

Googling, it looks like we may need the flag `-D_FILE_OFFSET_BITS=64`.

aaronfreimark, 2020-11-30:
I was able to verify that this issue occurs when the system is built on Raspberry Pi (not just Indigo Sky). During make, the following lines appear:

Mac:
`checking for _FILE_OFFSET_BITS value needed for large files... no`

RPi:
`checking for _FILE_OFFSET_BITS value needed for large files... 64`

I've submitted a patch to add the above flag during compilation. I've verified that this patch allows files over 2GB on the Raspberry Pi. Not sure I put the patch in the correct place, or if this has negative side effects somewhere else though!

polakovic, 2026-10-02:
32-bit ARM builds are compiled with -D_FILE_OFFSET_BITS=64, so files larger than 2 GB can be written. Closing.

closed 2026-10-02 by polakovic